armbian/build #550: Dockerfile for building on ARM with DistCC

jadr2ddude, 2016-11-20:
Now that I have a docker swarm cluster running Armbian, I would like to be able to use it to automatically build beta armbian images.

zador-blood-stained, 2016-11-20:
I can't even start to imagine how you can support different toolchains with distcc. I don't think this will ever be implemented in this build script.

jadr2ddude, 2016-11-20:
distccd can be run in docker containers with the same toolchain.

I am willing to do this and send a PR

zador-blood-stained, 2016-11-20:
> distccd can be run in docker containers with the same toolchain.

Yes, I know that. But in our case we need to switch different Linaro toolchains during the build (i.e. if u-boot and kernel require different versions), and if this isn't somehow implemented, then this feature cannot be considered "fully supported", and we probably don't want to support partially working code.

jadr2ddude, 2016-11-20:
Yeah I should be able to do that

jadr2ddude, 2016-11-20:
I can either:
1. Install multiple versions of gcc into the same image 
2. Have multiple images with different versions of gcc

probbably will do 1

jadr2ddude, 2016-11-21:
Anyway, I am starting with an ARM dockerfile for now

zador-blood-stained, 2016-12-28:
Please open a PR for further duscussion when/if you are done with this.

closed 2016-12-28 by zador-blood-stained