armbian/build #552: gpm doesn't work

rrohbeck, 2016-11-20:
Still the case after upgrading to stretch. See https://forum.armbian.com/index.php/topic/423-banana-pi-console-mouse-gpm-not-working/.

zador-blood-stained, 2016-11-21:
`/dev/input/mice` may be missing due to kernel configuration. I'll try to fix this.

zador-blood-stained, 2016-11-22:
@rrohbeck 
sunxi-next kernel in http://beta.armbian.com/ should now have necessary kernel options. Please test it.

zador-blood-stained, 2016-11-29:
Tested new kernel after update of configuration
```
➜  ~  % ls -la /dev/input
итого 0
drwxr-xr-x  4 root root     260 ноя 29 20:53 .
drwxr-xr-x 15 root root   13480 ноя 29 20:53 ..
drwxr-xr-x  2 root root      80 ноя 29 20:53 by-id
drwxr-xr-x  2 root root     120 ноя 29 20:53 by-path
crw-rw----  1 root input 13, 64 ноя 29 20:53 event0
crw-rw----  1 root input 13, 65 ноя 29 20:53 event1
crw-rw----  1 root input 13, 66 ноя 29 20:53 event2
crw-rw----  1 root input 13, 67 ноя 29 20:53 event3
crw-rw----  1 root input 13, 68 ноя 29 20:53 event4
crw-rw----  1 root input 13, 63 ноя 29 20:53 mice
crw-rw----  1 root input 13, 32 ноя 29 20:53 mouse0
crw-rw----  1 root input 13, 33 ноя 29 20:53 mouse1
crw-rw----  1 root input 13, 34 ноя 29 20:53 mouse2
➜  ~  %
```
If there are any other issues, they are probably caused by default GPM configuration.

closed 2016-11-29 by zador-blood-stained