armbian/build #553: consoleblank=1

rrohbeck, 2016-11-20:
After dist-upgrading to stretch, consoleblank was set to 1 in boot.cmd. While it looked cool to apparently boot directly to the display manager is makes the console unusable.

igorpecovnik, 2016-11-21:
We are not supporting stretch yet.

zador-blood-stained, 2016-11-21:
@rrohbeck 
What board and OS release are you talking about? I can't find a single script in this repository which has `consoleblank=1, and unset value is equivalent to `consoleblank=600`. In addition I am not sure that dist-upgrade operation would touch boot script in any way.

rrohbeck, 2016-11-26:
> What board and OS release are you talking about?

It's the original Banana Pi.

rrohbeck, 2016-11-26:
P.S. I dist-upgraded to stretch.
root@bpi:~# apt-cache policy linux-image-next-sunxi
linux-image-next-sunxi:                              
  Installed: 5.23                         
  Candidate: 5.23
  Version table:
 *** 5.23 500
        500 http://apt.armbian.com jessie/main armhf Packages
        100 /var/lib/dpkg/status

zador-blood-stained, 2016-12-28:
Still we don't touch `consoleblank` in any way, and once Stretch is officially released, it will be tested. So unless somebody finds what exactly touches the kernel command line during a dist-upgrade, it can't be dealt with.

closed 2016-12-28 by zador-blood-stained