armbian/build #558: WiFi issues

ThomasKaiser, 2016-11-24:
We started to support more and more devices that have onboard Wi-Fi and experience the following problems:

- people think Wi-Fi is broken since they run in troubles when they connect Wi-Fi and Ethernet to the 'same' network and access to the device through Wi-Fi seems to stop when Ethernet gets disconnected (where a router or Access point is acting as a 'bridge' and may be part of or the problem). This can only be solved by documenting the stuff (we have the infrastructure since months, we just don't use it -- instead we're answering the same questions again and again in the forum)
- Wi-Fi power management: At the moment defaults to whatever the driver does. IMO the correct way is to deal with it through `iwconfig` since then it's driver agnostic and we can adjust behaviour through `network-manager` hooks. Current implementation is ugly/hack-ish (`armhwinfo` and `h3consumption` fiddling around with files) and should be replaced with something that's part of board support package. Default should IMO be off (since users never read documentation and then simply think 'it's broken' and blame Armbian for 'bad drivers' and stuff like that) and documentation should be added how to enable powermanagement for IoT usage (it's basically just providing two simple scripts below `/etc/NetworkManager/dispatcher.d/` and adding/removing the executable bit to get the desired behabiour)
- Setting regulatory domain: Can be done in some driver dependant or static way but based on my tests I would prefer to completely rely on `network-manager` here since a good AP/router announces everything and then `CRDA` stuff will be set correctly in the background (doing the `iw reg set` call himself). In case that does not work a manual hook thrown in `/etc/NetworkManager/dispatcher.d/pre-up.d` eg. can be used (see below please for an example). Problem: Some drivers seem to not support that (eg. `dhd`) and it also needs documentation
- Again documentation: We need a page with the different Wi-Fi chips we support and status for each, regarding client, AP and monitor mode and which features which driver supports
- 3rd party dongles: I really don't care about those since crappy anyway but our users do (especially the 'buy cheap, buy twice' kind of). Based on forum threads (I only skim through when `8192` or `8188` appear) there's also room for improvement: Documentation and maybe ensuring that the right drivers are included by default and not the wrong ones (seems to be a common problem with 8192?). All the information is already there but scattered through different places (forum threads).

I doubt that anything will improve here soon since we totally fail to get new contributors on board.

    cat 99set-reg-domain.sh
    #!/bin/bash
    
    status=`cat /sys/class/net/wlan0/operstate`
    /bin/bash -n /etc/default/crda && source /etc/default/crda
    
    if [[ $status == 'up' ]]
    then
            iw reg set $REGDOMAIN
    fi

zador-blood-stained, 2016-11-25:
> Again documentation: We need a page with the different Wi-Fi chips we support and status for each, regarding client, AP and monitor mode and which features which driver supports

This will be a nightmare since it depends on kernel branch and SoC family (mostly for vendor kernels)

Also I want to add that firmware package needs to be cleaned up and instructions for installing firmware should be added to the documentation.

ThomasKaiser, 2016-11-26:
> Also I want to add that firmware package needs to be cleaned up

Might this be the reason for Wi-Fi failing when upgrading to latest Armbian on BPi M2+?

zador-blood-stained, 2016-11-26:
> Might this be the reason for Wi-Fi failing when upgrading to latest Armbian on BPi M2+?

I have no idea about wireless on M2 apart from #484, it needs to be tested by someone who owns this board.

ThomasKaiser, 2016-11-30:
Based an amount of interest here I would propose to disable power management ASAP using the 2 network-manager hooks but in a more sane way by adding those two files to board support packages with `disable` script execution bit set. Then the hack-ish parts in `armhwinfo` and `h3consumption` can be removed.

Slightly related to Wi-Fi since some OPi Zero owners are just interested to do an initial Wi-Fi setup on their boards: why not enabling `g_serial` there too by default to prevent stuff like this https://forum.armbian.com/index.php/topic/2933-do-i-need-a-display/#entry20589 ?

zador-blood-stained, 2016-11-30:
> Slightly related to Wi-Fi since some OPi Zero owners are just interested to do an initial Wi-Fi setup on their boards: why not enabling g_serial there too by default to prevent stuff like this https://forum.armbian.com/index.php/topic/2933-do-i-need-a-display/#entry20589 ?

Added: 0ca301701388b6f942a8c66f144adadb96af1032

ThomasKaiser, 2016-12-28:
Closing since zero progress/interest and our Github issues are only [misused](http://dietpi.com/phpbb/viewtopic.php?f=11&t=1038#p4482).

This will be fun when DietPi starts providing 4.9 based images soon since everything that doesn't work from then on he'll blame Armbian (devs) for -- good luck with this approach!

closed 2016-12-28 by ThomasKaiser

jpascher, 2017-04-20:
I don't think the problem is relay solved please look up my way to get around the problem. https://forum.armbian.com/index.php?/topic/4055-orange-pi-pc-wifi-ap-is-not-responding/#comment-29782