armbian/build #559: GPIO Support for H2/H3 boards with a unified WiringPI library in a neat little package

lanefu, 2016-11-24:
Time to wrangle all the ad-hoc forks of wiringPI and make a mature release that can easily be bundled into Armbian images to support most H2/H3 SBCs.

Inspired from [this post](https://forum.armbian.com/index.php/topic/2808-orange-pi-zero-went-to-the-market/page-4#entry20302)

lanefu, 2016-11-24:
Please keep all discusion on the armbian forums on [this topic](https://forum.armbian.com/index.php/topic/2956-559-gpio-support-for-h2h3-boards-with-a-unified-wiringpi-library-in-a-neat-little-package/)

zador-blood-stained, 2017-11-25:
I think we can celebrate the 1 year anniversary of this task by closing it. Even though there were some recent advancements in the direction of creating a universal GPIO library, it's better to use the forum and Collab for task tracking and keep this as a build script issue tracker.

closed 2017-11-25 by zador-blood-stained