armbian/build #561: H3 DVFS

jernejsk, 2016-11-24:
While I was comparing mine DVFS settings to Armbian, I found out that following two entries have "0" at the beginning, which, if I'm not mistaken, means octal representation.

https://github.com/igorpecovnik/lib/blob/master/config/fex/orangepi2.fex#L1102
https://github.com/igorpecovnik/lib/blob/master/config/fex/orangepiplus.fex#L1118

zador-blood-stained, 2016-11-24:
AFAIK these entries are used only with "dumb" GPIO regulator and are ignored with I2C one

jernejsk, 2016-11-24:
Source doesn't tell much, because it is just forwarded to ARISC. Maybe @ThomasKaiser knows for sure?

ThomasKaiser, 2016-11-24:
Also don't know for sure but I believe Mikhail's right and this value only matters with GPIO driven voltage regulation and ignored if I2C is used (PMU type 2). Should be easy to test but I'm afraid Plus and 2 (and especially the combination of both: 'Plus 2') are the most uninteresting H3 devices ever ;)

jernejsk, 2016-11-24:
Ok, then feel free to close it.

ThomasKaiser, 2016-11-25:
Adjusted with https://github.com/igorpecovnik/lib/commit/d65b3ed84cb6226ad21a21b3b024caa08db9ca7c to prevent 3rd parties using the fex files for differing H2+/H3 devices making mistakes. Thank you!

closed 2016-11-25 by ThomasKaiser