armbian/build #564: M2+ doesn't boot with >5.20

Grabber, 2016-11-26:
I tried to compile and boot from sdcard on bump versions >5.20, none are booting / working. Latest commit (#3bcc3c2) is recognizing M2+ as Xunlong Orange Pi PC, giving CRC errors...

[image]

Any insights?

jadr2ddude, 2016-11-26:
Are you sure you have the right image?

Otherwise, there might be some debug output through serial port (maybe?)

Grabber, 2016-11-26:
@jadr2ddude, yes.

I compiled it from git source: "Armbian_5.24_Bananapimp2plus_Ubuntu_trusty_3.4.112.img"

jadr2ddude, 2016-11-26:
Have you tried the version on the website or compiling with the dev kernel?

Grabber, 2016-11-26:
@jadr2ddude, as I told on the first post... I found the issues directly from cloning and compiling the sources (sudo ./compile.sh)

Check the strange error: https://www.youtube.com/watch?v=9Tg_U7ZUtQ8&feature=youtu.be

jadr2ddude, 2016-11-26:
Wait. . . what's the error???

Grabber, 2016-11-26:
@jadr2ddude, it is not booting from SD. I have 50 M2+ boards in my hands now... this one that I recored the video had a previous / old Armbian installed on eMMC. So it start booting from sdcard and switches to eMMC, never seen this behavior before. Look at the error on framebuffer... look at the wrong detection of the board... there is something wrong with newer Armbian commits. On other M2+ without anything on eMMC, screen go black and board keep rebooting after some minutes.

jadr2ddude, 2016-11-26:
Ok, so it just hangs at booting kernel. . .

Here is an image that I have with the dev kernel built for an Orange Pi that may or may not somewhat work (both use H3): https://transfer.sh/sRi7O/cryptov2.img.xz

Grabber, 2016-11-26:
Can you provide more details about your image?

jadr2ddude, 2016-11-26:
I have been working on merging together some of the major WIP branches of the 4.9 kernel (not the legacy one). It has somewhat functional support for the crypto engine (driver is there but openssl doesn't support it). It is Xenial with BTRFS. Non-accelerated graphics are untested (this is not a desktop image). It is XZ compressed.

Kernel branch here: https://github.com/jadr2ddude/linux/tree/sun8i-crypto-megous-4.9

kad, 2016-11-26:
Out of curiosity, how boards are powered ? I observed similar hang if board powered over barrel jack. If power provided via USB, board boots correctly.

zador-blood-stained, 2016-11-26:
> Latest commit (#3bcc3c2) is recognizing M2+ as Xunlong Orange Pi PC, giving CRC errors...

This is not an issue. There is no M2+ configuration in u-boot, so we are using one for OPi PC.

Overall booting issue was confirmed some time ago (https://forum.armbian.com/index.php/topic/2882-banana-pi-m2-does-not-boot-after-power-cycle/), no solution yet, but I would guess that something in FEX file causes this problem.

ThomasKaiser, 2016-11-26:
@jadr2ddude please try to stay on-topic (legacy kernel issue).

@Grabber I second @zador-blood-stained's idea to check fex settings: https://github.com/igorpecovnik/lib/commits/master/config/fex/bananapim2plus.fex (simply start with latest version known to work and iterate through until the problem occurs). I can confirm the booting problem but can't do much about it since my BPi M2+ sits unaccessible at a remote location now (stuck at booting).

igorpecovnik, 2016-11-26:
@Grabber Perhaps this is related to broken sunxi tools in our repository? Check if script.bin o.k. in first place.

Grabber, 2016-11-26:
@kad, only powered over barrel plug.

Grabber, 2016-11-26:
@igorpecovnik If I boot from sdcard, write to nand with 5.20 (both prebuilt or compiled from source) and last update to 5.23 with apt-get upgrade it works.

Grabber, 2016-12-02:
Anyone with new idea? Not advancing here...

haitao52198, 2016-12-02:
I find there is nothing in /boot, and I can see many errors when building image
/output/cache/mount-next-bananapipro-jessie/boot/.next.BQdgBa" failed: No space left on device (28)
/output/cache/mount-next-bananapipro-jessie/boot/.verbose.4EZ7uZ" failed: No space left on device (28)
/output/cache/mount-next-bananapipro-jessie/boot/.System.map-4.8.12-sunxi.TCYZoO" failed: No space left on device (28)
/output/cache/mount-next-bananapipro-jessie/boot/.armbianEnv.txt.E8ydoD" failed: No space left on device (28)
/output/cache/mount-next-bananapipro-jessie/boot/.armbian_first_run.txt.VT4rns" failed: No space left on device (28)
/output/cache/mount-next-bananapipro-jessie/boot/.boot.bmp.ChYGmh" failed: No space left on device (28)
/output/cache/mount-next-bananapipro-jessie/boot/.boot.cmd.ZUJWl6" failed: No space left on device (28)
/output/cache/mount-next-bananapipro-jessie/boot/.boot.scr.YV9clV" failed: No space left on device (28)
/output/cache/mount-next-bananapipro-jessie/boot/.config-4.8.12-sunxi.HXXtkK" failed: No space left on device (28)

Grabber, 2016-12-02:
This SinoVoip image is booting perfectly, not presenting the same bug as bound on Armbian. 

```
Linux bpi-iot-ros 3.4.112-sun8i #2 SMP PREEMPT....
```

dmesg:
https://gist.github.com/Grabber/80758fc9f65afc144c961af58ffec861

image download:
http://forum.banana-pi.org/t/bpi-m2-new-image-ubuntu-16-04-xenial-minimal-preview-bpi-m2p-img-2016-07-10/1989

I'm investigating...

zador-blood-stained, 2016-12-02:
If this script.fex is from this image, then they used another board as a reference:
```
[product]
version = "100"
machine = "orangepi-plus"
```

Grabber, 2016-12-02:
@zador-blood-stained 

**Armbian 5.12**
```
[product]
version = "100"
machine = "Banana Pi BPI-M2-Plus"
```

**SinoVoip / 2016-07-10-ubuntu-16.04-xenial-minimal-preview-bpi-m2p.img**
```
[product]
version = "100"
machine = "orangepi-plus"
```

Are you saying that by just changing machine param value it can work?

zador-blood-stained, 2016-12-02:
No, I'm saying that we need to test current Armbian images with this script.fex, and if it boots, check the differences between fex files.

Grabber, 2016-12-02:
@zador-blood-stained, it did this test now. I compiled Armbian 5.20 from sources, burnt to sdcard and the problem is happen. Last I replaced the script.fex from Armbian with SinoVoip version found on the .img I mentioned on previous link. It is now working...

Grabber, 2016-12-02:
@zador-blood-stained, please check both script.fex for comparison. 

https://gist.github.com/Grabber/67214110d1aec17fb749a2e06c4e9e96

Grabber, 2016-12-02:
Replacing Armbian fex with Sinovoip fex is resulting in "gmac0 gmac0: eth0: Initialize hardware error" message.

[image]

After a while waiting to boot... it boots, without networking.

Grabber, 2016-12-03:
It seems that if you not plug HDMI cable and power up fist, the problem is gone.

Grabber, 2016-12-03:
It seems that if you boot the M2+ without HDMI cable connected it always work with Armbian 5.20 and fex from lib repo. Same behavior with 5.20 upgrade to 5.23 via apt upgrade.

zador-blood-stained, 2016-12-03:
> Replacing Armbian fex with Sinovoip fex is resulting in "gmac0 gmac0: eth0: Initialize hardware error" message.

`gmac_phy_power_en` in `[gmac_phy_power]` section GPIO values are different

> It seems that if you boot the M2+ without HDMI cable connected it always work with Armbian 5.20 and fex from lib repo. Same behavior with 5.20 upgrade to 5.23 via apt upgrade.

Apart from different display resolution (720p60 vs 1080p60) display settings look identical.

Grabber, 2016-12-08:
@zador-blood-stained, power up the M2+ without HDMI connected is working, no more issues. Tried to find the root cause, no success, no more ideas.

igorpecovnik, 2016-12-09:
> Apart from different display resolution (720p60 vs 1080p60) display settings look identical.

@Grabber 
I know it's stupid but since it's related to HDMI somehow ... if you change screen resolution to match stock Sinovoip?

Grabber, 2016-12-09:
It is not stupid, but in my logic it would make sense if Armbian > SinoVoip resolution.

Grabber, 2016-12-13:
@igorpecovnik and @zador-blood-stained, do you know about any bug on ethernet? It appears that when ethernet cable is connected it works.

igorpecovnik, 2016-12-13:
Working fine on latest nighty build so we can close this.

Grabber, 2016-12-24:
It is not solved.

Grabber, 2016-12-24:
@igorpecovnik, please reopen it. If you unplug boards and plug power after some hours... the boot is stuck after Armbian logo.

Grabber, 2016-12-24:
Images from Sinovoip are booting normally. It is an Armbian issue.

igorpecovnik, 2016-12-25:
This is also the only board which has such weird problems and it aslo causes kernel crash on current mainline kernel on shutdown. All other H3 boards act fine. What is fundamental different is U-boot and I would start findind problems there.

Grabber, 2016-12-25:
Is M2+ using different u-boot from other boards?

zador-blood-stained, 2016-12-25:
Sinovoip images are using legacy u-boot (most likely) and different partitioning scheme.
Mainline u-boot works fine on all other H3 devices, so even in case boot troubles are related to the  mainline u-boot, it is **not** Armbian problem. It is most likely related to kernel and u-boot configuration, which is based on documentation provided by vendor.

Grabber, 2016-12-25:
1. don't know if it helps anyway: BPI-M2_Plus_Android_V2.zip (provided by Sinovoip) is working;
2. kernel 4 is not booting, it is freezing on Armbian logo;

zador-blood-stained, 2017-06-13:
@igorpecovnik 
Is this still an issue with current builds?

igorpecovnik, 2017-06-13:
You mean showing artefacts? No, this is gone.

closed 2017-06-13 by igorpecovnik

igorpecovnik, 2017-06-13:
Tested with BPi M2 v1.1 No problems found, wireless and Bluetooth works out of the box, ...