armbian/build #565: Armbian keep rebooting after trying to change default password

Grabber, 2016-11-26:
This is a long-term bug I saw in Armbian. Both when booting from sdcard or emmc, on the first login it is required to change the default password.

The issue is that sometimes after login root/1234 the system just reboot automatically, somethings it prompts for changing the password and as soon you type the password name it reboots.

https://www.youtube.com/watch?v=r1NnNkr9Q-c

Since I'm testing it on M2+ and only 5.20 is working (+5.20 has other problems related in other issue)... tried on Debian, trusty & xenial.

zador-blood-stained, 2016-11-26:
Is this the first boot or first boot after automatic FS resize? On images with legacy kernel reboot will be triggered automatically and timing depends on speed of SD/eMMC.

Also it doesn't help that you can't see full error messages in the video.

zador-blood-stained, 2016-11-26:
Since I see `Last login ... on tty1` I can assume that automatic FS resize already happened. Did you try to add user before? If process of adding a new user was interrupted by automatic reboot then it would cause the issue you are experiencing.

Grabber, 2016-11-26:
@zador-blood-stained 

I agree with your latest post. How can we avoid it?

zador-blood-stained, 2016-11-26:
Easy solution - just wait for automatic reboot to happen, not touching anything. Newer Armbian images display an extra warning about automatic reboot
Not so easy solution - move FS resizing to initrd. Probably won't happen anytime soon.

Grabber, 2016-11-26:
@zador-blood-stained, if I do it it keeps rebooting on loop.

zador-blood-stained, 2016-11-26:
This should not happen unless you got a filesystem corruption (bad SD cards, power supply issues, ...)

Grabber, 2016-11-26:
@zador-blood-stained, I burned the image to sdcard again and after first boot I waited for first automatic reboot... now the problem is gone. But it is strange to this reboot depends on something "final user" don't know.

closed 2016-12-13 by Grabber

twhtanghk, 2021-08-12:
USB dongle for keyboard confused the system to identify usb device. My orange pi pc plus keeps rebooting with message "Failed to shut down host controller". Once unplugged the usb dongle, everything resumed normal.

igorpecovnik, 2021-08-12:
> USB dongle for keyboard confused the system to identify usb device. My orange pi pc plus keeps rebooting with message "Failed to shut down host controller". Once unplugged the usb dongle, everything resumed normal.

This has nothing to do with this issue. Please use https://forum.armbian.com/