armbian/build #569: Suggestion: have all of the dev kernels in one repo

jadr2ddude, 2016-11-28:
For adding crypto engine support, I had to merge branches from megous and montjoie. I am thinking: why don't we just keep a repository with different branches for different versions of the kernel, and we can pull in changes as we need features. This might also make it more stable, since we would be able to easily test everything in a seperate branch before merging. Like we could have a repo with branches:

sun8i
pine64
etc.

And then we could just have WIP branches named differently like wip-crypto or whatever. We can also modify these branches instead of patching for things that most people would want (like support for olinuxino-a33)

jadr2ddude, 2016-11-28:
Like this: https://github.com/jadr2ddude/linux/tree/armbian-sun8i
(for sun8i)

zador-blood-stained, 2016-11-28:
Because somebody would have to maintain all these repositories, pull changes, resolve conflicts, etc.
At the same time we don't make any images with dev kernels for end users, so essentially this would be just a waste of time.

closed 2016-11-28 by jadr2ddude