armbian/build #571: usb temper MOTD issue

calhemp, 2016-11-29:
Hi, I have a usb temper device, and works well in my cubieboard 1 4.8.4 kernel.

but I tryed to see in MOTD the value of this usb, and I think script there are a issue and but it shows an error.

I don't know about code, but I think that temper outs a string and motd not check this and when calculates the value, throws error, because divides a string and not a value with bc command.

this is my output of my temper:

`temper -c`
output
2016/11/29 21:38:44 Temperature 24.81C

but in 
lib/scripts/update-motd.d/30-sysinfo
line
143

`if [[ -n $(which temper) && $(dpkg --print-architecture) == armhf ]]; then
   amb_temp=$(temper -c) --> output value 2016/11/29 21:38:44 Temperature 24.81C
   if echo $amb_temp | egrep -qv "Couldn't find the USB device"; then
--> error     amb_temp=$(echo "scale=1;${amb_temp}/1" | bc)
  else
      amb_temp="" 
  fi
fi`

if I run only this part in a bash script
shows in console:
(standard_in) 1: syntax error
(standard_in) 1: illegal character: :
(standard_in) 1: illegal character: :
(standard_in) 1: illegal character: T

I think that this string has to split only have the last part to work reliable in motd

Is correct?

thanks.
Peter

igorpecovnik, 2016-11-29:
https://github.com/igorpecovnik/lib/commit/7805982efb9cd30fd043375fc575cd8c26d4aced
This should fix the problem.

closed 2016-11-29 by igorpecovnik

calhemp, 2016-11-29:
thanks, Igor, test and working.