armbian/build #575: image is fail

oleg-krv, 2016-12-06:
Build image Lamobo dev with https://github.com/igorpecovnik/lib/pull/255#issuecomment-205045273 - image is fail:
```
#LANG=us_UK.UTF-8 fdisk -lu Armbian_5.24_Lamobo-r1_Debian_jessie_4.9.0.img
Disk Armbian_5.24_Lamobo-r1_Debian_jessie_4.9.0.img: 1.2 GiB, 1334837248 bytes, 2607104 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x135bc6e3

Device                                          Boot Start     End Sectors  Size Id Type
Armbian_5.24_Lamobo-r1_Debian_jessie_4.9.0.img1       2048 2607103 2605056  1.2G 83 Linux

# LANG=us_UK.UTF-8 mount -t ext3 -o loop,offset=$((2048*512)) Armbian_5.24_Lamobo-r1_Debian_jessie_4.9.0.img /mnt/tmp
mount: wrong fs type, bad option, bad superblock on /dev/loop5,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.

```
[debug_files.zip](http://olegkrv.pp.ua/Armbian/debug_files.zip)
[Armbian_5.24_Lamobo-r1_Debian_jessie_4.9.0.img.7z](http://olegkrv.pp.ua/Armbian_5.24_Lamobo-r1_Debian_jessie_4.9.0.img.7zl)

zador-blood-stained, 2016-12-06:
- Image link is broken

- Default file system is ext4

- Using `losetup` and `partprobe` would be a better way to test filesystem presence

oleg-krv, 2016-12-06:
Correct image link
[Armbian_5.24_Lamobo-r1_Debian_jessie_4.9.0.img.7z](http://olegkrv.pp.ua/Armbian_5.24_Lamobo-r1_Debian_jessie_4.9.0.img.7z)

Also, if you move the image on the sd-card (using dd): unknown partition

zador-blood-stained, 2016-12-06:
Well, if 1GB+ image was compressed to 300KB, then something definitely went wrong. Please repeat building procedure and provide console output from build script

oleg-krv, 2016-12-06:
Thank you, I did not pay attention to the size of the archive. deb-files were gathered together perfectly.
Question for the assembly: it is possible to add to the building of the image of my files (new or replacement files with git) without editing the build scripts? As a custom patch

zador-blood-stained, 2016-12-06:
> Question for the assembly: it is possible to add to the building of the image of my files (new or replacement files with git) without editing the build scripts?

Depends on what you want to add. Are you talking about a custom kernel patch from [here](https://github.com/igorpecovnik/lib/issues/511#issuecomment-264846295), some static files (i.e. configuration) or something else? Some of these cases are described somewhere in documentation, so make sure to read it first.

oleg-krv, 2016-12-06:
```
[ o.k. ] Calling image customization script [ customize-image.sh ]
[ o.k. ] Preparing image file for rootfs [ lamobo-r1 jessie ]
[ o.k. ] Current rootfs size [ 818 MiB ]
[ o.k. ] Creating blank image for rootfs [ 1274 MiB ]
1.24GiB [  60MiB/s] [========================================================================================================================================================>] 100%
[ o.k. ] Creating partitions [ root: ext4 ]
[ .... ] Creating rootfs [ ext4 ]
/dev/loop6p1: Operation not permitted while setting up superblock
tune2fs: Operation not permitted while trying to open /dev/loop6p1
Couldn't find valid filesystem superblock.
mount: /dev/loop6p1 is write-protected, mounting read-only
mount: wrong fs type, bad option, bad superblock on /dev/loop6p1,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.
[ o.k. ] Copying files to image [ sdcard-dev-lamobo-r1-jessie.raw ]
        724.15M  99%   20.37MB/s    0:00:33 (xfr#45499, to-chk=0/56811)

sent 727.47M bytes  received 913.29K bytes  20.52M bytes/sec
total size is 728.03M  speedup is 1.00
[ o.k. ] Copying files to /boot partition [ sdcard-dev-lamobo-r1-jessie.raw ]
         22.68M  99%  166.13MB/s    0:00:00 (xfr#153, to-chk=0/160)

sent 22.70M bytes  received 2.95K bytes  45.40M bytes/sec
total size is 22.68M  speedup is 1.00
[ o.k. ] Free space: [ SD card ]
tmpfs           1.5G  818M  683M  55% /root/output/cache/sdcard-dev-lamobo-r1-jessie
[ o.k. ] Writing U-boot bootloader [ /dev/loop6 ]
umount: /root/output/cache/mount-dev-lamobo-r1-jessie: not mounted
[ o.k. ] Done building [ /root/output/images/Armbian_5.24_Lamobo-r1_Debian_jessie_4.9.0.img ]
[ o.k. ] Runtime [ 13 min ]
```

The problem is found.
It would be good-that-be at such problem was issued an error message. But this is my carelessness.
Patches for the kernel I realized how do make. On other issues - I will look the documentation.
Thank you

zador-blood-stained, 2016-12-07:
> `/dev/loop6p1: Operation not permitted while setting up superblock`

Are you sure Docker is running in privileged mode? What host OS do you use (what is Docker running on?)

oleg-krv, 2016-12-07:
Ubuntu 16.04 -

> Linux oleg 4.4.0-53-generic #74-Ubuntu SMP Fri Dec 2 15:59:10 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

start docker

> sudo docker run -it -P --privileged=true --name armbian lib /bin/bash

id docker:

> root @ fc68330c38a7:/dev# ls -lh /dev/loo*
> crw-rw---- 1 root disk  10, 237 Dec  6 14:36 /dev/loop-control
> brw-rw---- 1 root disk   7,   0 Dec  6 14:36 /dev/loop0
> brw-rw---- 1 root disk   7,   1 Dec  6 14:36 /dev/loop1
> brw-rw---- 1 root disk   7,   2 Dec  6 14:36 /dev/loop2
> brw-rw---- 1 root disk   7,   3 Dec  6 14:36 /dev/loop3
> brw-rw-rw- 1 root disk   7,   4 Dec  7 07:57 /dev/loop4
> brw-rw-rw- 1 root root 259,   1 Dec  6 15:48 /dev/loop4p1
> brw-rw-rw- 1 root disk   7,   5 Dec  6 14:36 /dev/loop5
> brw-rw-rw- 1 root disk   7,   6 Dec  6 14:36 /dev/loop6
> brw-rw-rw- 1 root disk   7,   7 Dec  6 14:36 /dev/loop7

zador-blood-stained, 2016-12-29:
Project README was updated to state that using Docker may not work for building images, so closing this.

closed 2016-12-29 by zador-blood-stained