armbian/build #576: Login and Password

Nill2016, 2016-12-06:
Friend, I downloaded the version of armbian for orange pi plus 2 and installed it on the micro sd card boots it does the whole boot process and when it finishes it asks for login and password. It does not enter the system. Where do I locate this login and password?

jadr2ddude, 2016-12-07:
The initial username is root and password is 1234. It will then ask you to change the password

zador-blood-stained, 2016-12-07:
https://docs.armbian.com/User-Guide_Getting-Started/#how-to-login

closed 2016-12-07 by zador-blood-stained

Nill2016, 2016-12-07:
Friends are sending pictures of the screen where to and pressing enter prompts login and password. I have already done the procedure that is in the https://docs.armbian.com/User-Guide_Getting-Started/#how-to-login tab but when I put what you ask appears incorrect.

[image]
[image]

zador-blood-stained, 2016-12-07:
I don't see any error messages here

Nill2016, 2016-12-07:
When this message appeared I entered my name and password 1234 but an incorrect message appeared. Replaces with orangepiplus2e and the same password and also gave incorrect. This screen is only so because I re-formatted the card thinking that it could be some recording error.

Nill2016, 2016-12-07:
In this photo I made a test putting only pi in the login and the password 1234 and appears as incorrect.
[image]

zador-blood-stained, 2016-12-07:
Please read the documentation again, it lists what user name should be used for first login, and then look at what you will be prompted to do.
You are suposed to enter
login: **root**
password: **1234**

old password: **1234**
and then new password for root user (2 times)

Nill2016, 2016-12-07:
OK thank you. I'll test then return with the result.

Nill2016, 2016-12-08:
Dear friends, armbian is working properly now. Thank you for your help.