armbian/build #579: strange artifacts on armbian boot

Grabber, 2016-12-13:
image built from latest commit (today), already seen this issue on previous commits, tested on M2+. @igorpecovnik reported that in his 4k monitor it is showing a pink screen.

[image]

igorpecovnik, 2016-12-14:
This is present in all Allwinner boars in this combination: mainline u-boot + legacy kernel

jernejsk, 2016-12-24:
If this is only for a second or two during Linux boot, issue is known and fixed with:
https://github.com/jernejsk/OpenELEC-OPi2/blob/openelec-7.0/projects/H3/patches/u-boot/u-boot-06-fix-screen-switch.patch

But this doesn't apply to mainline kernel, so fix will not be included in mainline U-Boot. If anyone needs detailed explanation why this happens, just ask...

Grabber, 2016-12-24:
@jernejsk, sometimes it freezes on the artifact screen and board (M2+) doesn't boot. Can you provide us more information about the patch and the "whys" behind?

jernejsk, 2016-12-24:
I'm pretty sure that this artifact screen is harmless otherwise. So, U-Boot driver currently places framebuffer (32MB) at the end, e.g. if system has 512MB of RAM, then framebuffer is located at 480MB. This last chunk (256MB) of memory is somewhat special to the system. Some HW units can only use memory located in this chunk, so Linux kernel reserve this memory and use it for those units. But because it can't check that part of this memory is already in use, it reuses it at the boot and place some data into it. This data gets displayed then and you can see it as the artifact.

So the solution is to place framebuffer lower, not at the complete end of the RAM. Patch was tested only on 1080p screen, so it might still not be enough for 4K, but this can be solved by hardcoding lower address. According to my test, it doesn't even need to be in the last 256MB of RAM, but this still needs to be more thorougly tested.

My opinion is that this artifact actually is not harmful to the system. If anything weird happens to the kernel during boot, you will almost always see this artifact, because HDMI is one of the very last unit which gets initialized.

Grabber, 2016-12-24:
@jernejsk, 
/drivers/video/sun8i_display.c doesn't exist. We have both sunxi_display.c and sunxi_display2.c

jernejsk, 2016-12-24:
Well, I'm using a bit older driver in OpenELEC. Just edit the patch and rename filename to sunxi_display2.c

Grabber, 2016-12-25:
@igorpecovnik, I did a new patch with @jernejsk code
[u-boot-06-fix-screen-switch.patch.txt](https://github.com/igorpecovnik/lib/files/672000/u-boot-06-fix-screen-switch.patch.txt)

Why am I getting [u][c] and not [l][c]?

```
[ o.k. ] ... [l][c] h3-enable-power-led.patch [ succeeded ]
[ o.k. ] ... [l][c] h3-set-safe-axi_apb-clock-dividers.patch [ succeeded ]
[ o.k. ] ... [l][c] sunxi-boot-splash.patch [ succeeded ]
[ o.k. ] ... [u][c] u-boot-06-fix-screen-switch.patch [ succeeded ]
```

jernejsk, 2016-12-25:
@Grabber, does it help in any way?

Grabber, 2016-12-25:
@jernejsk, I'm trying to patch it now. If you can I'm on-line on Skype (grabberbr) or Hangouts (grabber@gmail.com)

Grabber, 2016-12-25:
@jernejsk 

Sometimes it is booting, sometimes it is not. It is showing less artifacts... but we still have artifacts.
https://youtu.be/Ns3_yw13DzU

jernejsk, 2016-12-25:
Well, you can put any number at gd->fb_base = (unsigned long)... line. The only issue with it is how to select appropriate address. I don't really know. It should be a number or algorithm which would work on all boards. RAM starts at address 0x40000000 and depending on amount of it could go as high as 0xBFFFFFFF (2GB). Additional limitation is that last 3 numbers must be 0.

You can try this algo:
RAM top - 256MB - 32MB, where RAM top is one variable found in U-Boot and tells max RAM address

Grabber, 2016-12-25:
Got your idea, but I think we need some advise from someone who has experienced most board with Armbian.

Isn't memalign() deprecated? Maybe we should yse posix_meanalign()?

Grabber, 2016-12-27:
If we disable HDMI / GPU with **h3consumption -g off**, after u-boot we will see the artifacts. The interesting part is that after Armbian up to login, the artifact screen show some bits changing... my guessing is that something is still writing data to the wrong buffer.

jernejsk, 2016-12-28:
We are talking about U-Boot, so there is no "posix_memalign()" function.

This issue is due to a combination of U-Boot frambebuffer and Linux kernel using same memory region for something else, so thing described in your last post is completely expected.

zador-blood-stained, 2017-01-03:
Fixed in 1d38c5b thanks to @jernejsk patch.

closed 2017-01-03 by zador-blood-stained