armbian/build #583: firmware-realtek

hellfi, 2016-12-20:
Hi!

On current Linux orangepi 4.8.4-sunxi apt-get install _firmware-realtek_ errors out:
Unpacking firmware-realtek (0.43) ...
dpkg: error processing archive /var/cache/apt/archives/firmware-realtek_0.43_all.deb (--unpack):
 trying to overwrite '/lib/firmware/RTL8192SU/rtl8192sfw.bin', which is also in package armbian-firmware 5.22

Which somewhat limits the easy use of the 'buy cheap, buy twice' kind of adapters, manually extracting /rtlwifi/ from firmware-realtek_0.43_all.deb seems to work fine. 
somewhat related to https://github.com/igorpecovnik/lib/issues/424 and others

BR
M

igorpecovnik, 2016-12-20:
We pack all firmware in armbian-firmware-full ... that's the proper sollution.

hellfi, 2016-12-20:
thanks for the answer!
I see your point, but ~ 100kB vs ~ 100MB

jadr2ddude, 2016-12-20:
Also, would it make sense to mark armbian-firmware-full as conflicting with firmware-realtek (and maybe provides too)?

zador-blood-stained, 2016-12-20:
Ideally if firmware if present both in Debian and Ubuntu repositories it should not be included in non-full Armbian package, otherwise it should have at least "Replaces" field.

zador-blood-stained, 2017-01-03:
Fixed with ea5e97c, even though it's not the ideal solution.

closed 2017-01-03 by zador-blood-stained