armbian/build #584: Orange Pi Zero debootstrap failed

nvl1109, 2016-12-21:
I was tried to build OS image (master branch) for my Orange Pi Zero and got following issue:
`dpkg: dependency problems prevent configuration of armbian-tools-xenial:
 armbian-tools-xenial depends on libusb-1.0-0 (>= 2:1.0.8); however:
  Package libusb-1.0-0 is not installed.

dpkg: error processing package armbian-tools-xenial (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 armbian-tools-xenial`
I was tried to build with mainline (kernel 4.9) or stable (kernel 3.4), jessie or xenial. Same error on all cases.

How can I resolve this? Thank you.

igorpecovnik, 2016-12-21:
I just cleaned the root cache and recreate and Jessie image for OPi0 with kernel 4.9.x without a problem. Try removing rootfs cache and repeat tje process.

nvl1109, 2016-12-21:
Hi @igorpecovnik ,
Thank you for your suggestion. It is absolutely true. After delete the cache, then rebuild, the image now works fine.
Thank you again.

closed 2016-12-21 by nvl1109