armbian/build #586: USB HID Gadget driver segmentation fault in latest dev kernel (4.9.0-sun8i)

liftoff, 2016-12-23:
For troubleshooting this I unloaded all kernel modules (crashes regardless of loaded kernel modules though):
```
root@orangepizero:~/keyboard-gadget# lsmod
Module                  Size  Used by
```
Here's the crash:
```
root@orangepizero:~/keyboard-gadget# ./gadget-setup.sh
UDC: musb-hdrc.1.auto
[   70.740719] Unable to handle kernel NULL pointer dereference at virtual address 00000002
[   70.749926] pgd = ce164000
[   70.752790] [00000002] *pgd=4dae8835, *pte=00000000, *ppte=00000000
[   70.759208] Internal error: Oops: 17 [#1] SMP THUMB2
[   70.764204] Modules linked in: usb_f_hid libcomposite [last unloaded: uio]
[   70.771216] CPU: 1 PID: 1143 Comm: gadget-setup.sh Not tainted 4.9.0-sun8i #7
[   70.778383] Hardware name: Allwinner sun8i Family
[   70.783119] task: ce271c80 task.stack: c9c80000
[   70.787825] PC is at alloc_ep_req+0x1b/0x50 [libcomposite]
[   70.793370] LR is at kmem_cache_alloc+0xe5/0xec
[   70.797940] pc : [<bf840ddc>]    lr : [<c01c2939>]    psr: 20000033
[   70.797940] sp : c9c81de8  ip : 00000000  fp : cd2a1e98
[   70.809465] r10: ce0c68c4  r9 : cdacacec  r8 : 00000100
[   70.814723] r7 : ce0c68a8  r6 : cd20e418  r5 : cea35100  r4 : 00000008
[   70.821286] r3 : 00000000  r2 : ffffffff  r1 : 02080020  r0 : cea35100
[   70.827853] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA Thumb  Segment none
[   70.835200] Control: 50c5387d  Table: 4e16406a  DAC: 00000051
[   70.840982] Process gadget-setup.sh (pid: 1143, stack limit = 0xc9c80210)
[   70.847805] Stack: (0xc9c81de8 to 0xc9c82000)
[   70.852210] 1de0:                   bf84d0c0 cdacac94 00000000 bf84c47d ce0c6924 c0b48ba8
[   70.860443] 1e00: cd2a1dec bf83d8df cdacac94 ce0c68a8 00000200 00000000 00000100 bf83d743
[   70.868677] 1e20: ce0c691c cdacac94 ce0c68a8 00000200 ce0c691c bf840511 cd2a1c00 cd2a1e60
[   70.876978] 1e40: 00000000 cd20f148 bf842180 cd2a1dec 00008060 cd195400 00000000 cd2a1dec
[   70.885273] 1e60: cdaf5900 cd2a1dec c0b48ba8 c9cee000 00000051 c0545913 cd195400 00000000
[   70.893567] 1e80: c0b48bc0 c0545bed 00000000 cdaf5900 cd2a1c00 cd2a1d90 fffffff0 00000011
[   70.901861] 1ea0: ce1d0e40 bf840c0b 000ccc08 00000000 00000011 cdaf52c0 000ccc08 cdaf52d8
[   70.910148] 1ec0: c9c81f80 c021e735 c9c81f80 ce1d0e40 c021e6a5 c0712138 c9c81f80 00000000
[   70.918441] 1ee0: 00000011 00000000 000c475c c01d01d7 cc751b50 00000301 00000000 00000428
[   70.926735] 1f00: 00000000 00000000 00000000 c9c81f10 c9c81efc 00000001 cd35fb40 00000000
[   70.935028] 1f20: 0000000a 00000100 ce184200 ce258b40 0000000a c01e57f9 ce1d0e40 00000011
[   70.943323] 1f40: 000ccc08 c9c81f80 00000000 00000000 000c475c c01d0c3d cdaeba80 00000000
[   70.951616] 1f60: 00000002 ce1d0e40 ce1d0e40 00000000 00000000 000ccc08 00000011 c01d17df
[   70.959910] 1f80: 00000000 00000000 ce271c80 00000011 000ccc08 b6f5bd58 00000004 c0106024
[   70.968204] 1fa0: c9c80000 c0105e61 00000011 000ccc08 00000001 000ccc08 00000011 00000000
[   70.976498] 1fc0: 00000011 000ccc08 b6f5bd58 00000004 00000000 000aa538 000c715c 000c475c
[   70.984791] 1fe0: 00000000 befca7dc b6ebf79b b6efaf96 00000030 00000001 00000000 00000000
[   70.993208] [<bf840ddc>] (alloc_ep_req [libcomposite]) from [<bf84c47d>] (hidg_bind+0x68/0x1c8 [usb_f_hid])
[   71.003158] [<bf84c47d>] (hidg_bind [usb_f_hid]) from [<bf83d743>] (usb_add_function+0x3a/0xcc [libcomposite])
[   71.013414] [<bf83d743>] (usb_add_function [libcomposite]) from [<bf840511>] (configfs_composite_bind+0x1cc/0x2f0 [libcomposite])
[   71.025267] [<bf840511>] (configfs_composite_bind [libcomposite]) from [<c0545913>] (udc_bind_to_driver+0x1b/0x74)
[   71.035751] [<c0545913>] (udc_bind_to_driver) from [<c0545bed>] (usb_gadget_probe_driver+0xa1/0xdc)
[   71.044993] [<c0545bed>] (usb_gadget_probe_driver) from [<bf840c0b>] (gadget_dev_desc_UDC_store+0x82/0x98 [libcomposite])
[   71.056153] [<bf840c0b>] (gadget_dev_desc_UDC_store [libcomposite]) from [<c021e735>] (configfs_write_file+0x91/0xf8)
[   71.066898] [<c021e735>] (configfs_write_file) from [<c01d01d7>] (__vfs_write+0xf/0xb4)
[   71.075025] [<c01d01d7>] (__vfs_write) from [<c01d0c3d>] (vfs_write+0x71/0x13c)
[   71.082455] [<c01d0c3d>] (vfs_write) from [<c01d17df>] (SyS_write+0x2b/0x64)
[   71.089602] [<c01d17df>] (SyS_write) from [<c0105e61>] (ret_fast_syscall+0x1/0x4c)
[   71.097290] Code: 6a73 2120 f2c0 2108 (f993) 2002
[   71.102403] ---[ end trace 80d83374e23cb8d0 ]---
Segmentation fault
```
I'm running the latest kernel compiled from the Docker image as of today (20161223).  I made sure to enable CONFIG_USB_G_HID before compiling: 
```
root@orangepizero:~/keyboard-gadget# zcat /proc/config.gz | grep USB_G_HID
CONFIG_USB_G_HID=m
```
I have no idea how to troubleshoot this kind of issue.  Any assistance in that regard would be helpful.  Of course, if you know how to fix it let me know that too!

Here's the (mostly boilerplate) shell script that attempts to setup the USB HID device:
```
#!/bin/bash

# This script will setup a simple HID keyboard gadget via ConfigFS.
# In order to use it, you must have kernel >= 3.19 and configfs enabled
# when the kernel was compiled (it usually is).

# variables and strings
MANUFACTURER="Riskable"                                       #  manufacturer attribute
SERIAL="FFFFFF"                                           #  device serial number
IDPRODUCT="0xa4ac"                                                #  hex product ID, issued by USB Group
IDVENDOR="0x0525"                                                 #  hex vendor ID, assigned by USB Group
PRODUCT="Emulated HID Keyboard"                                   #  cleartext product description
CONFIG_NAME="Configuration 1"                                     #  name of this configuration
MAX_POWER_MA=120                                                  #  max power this configuration can consume in mA
PROTOCOL=1                                                        #  1 for keyboard. see usb spec
SUBCLASS=1                                                        #  it seems either 1 or 0 works, dunno why
REPORT_LENGTH=8                                                   #  number of bytes per report
#DESCRIPTOR=/config/usb_gadget/keyboardgadget/kybd-descriptor.bin  #  binary blob of report descriptor, see HID class spec
DESCRIPTOR=/root/keyboard-gadget/kybd-descriptor.bin  #  binary blob of report descriptor, see HID class spec
UDC=`ls /sys/class/udc/`                                           #  name of the UDC driver to use (found in /sys/class/udc/)
#UDC=ci_hdrc.0                                                     #  name of the UDC driver to use (found in /sys/class/udc/)

# gadget configuration
modprobe libcomposite                                             #  load configfs
mkdir -p /config
mount none /config -t configfs                                    #  mount configfs as type configfs at /config
mkdir -p /config/usb_gadget/keyboardgadget                           #  make a new gadget skeleton
cd /config/usb_gadget/keyboardgadget                              #  cd to gadget dir
mkdir configs/c.1                                                 #  make the skeleton for a config for this gadget
mkdir functions/hid.usb0                                          #  add hid function (will fail if kernel < 3.19, which hid was added in)
echo $PROTOCOL > functions/hid.usb0/protocol                      #  set the HID protocol
echo $SUBCLASS > functions/hid.usb0/subclass                      #  set the device subclass
echo $REPORT_LENGTH > functions/hid.usb0/report_length            #  set the byte length of HID reports
cat $DESCRIPTOR > functions/hid.usb0/report_desc                  #  write the binary blob of the report descriptor to report_desc; see HID class spec
mkdir strings/0x409                                               #  setup standard device attribute strings
mkdir configs/c.1/strings/0x409
echo $IDPRODUCT > idProduct
echo $IDVENDOR > idVendor
echo $SERIAL > strings/0x409/serialnumber
echo $MANUFACTURER > strings/0x409/manufacturer
echo $PRODUCT > strings/0x409/product
echo $CONFIG_NAME > configs/c.1/strings/0x409/configuration
echo $MAX_POWER_MA > configs/c.1/MaxPower
ln -s functions/hid.usb0 configs/c.1                              #  put the function into the configuration by creating a symlink

# binding
echo "UDC: $UDC"
echo $UDC > UDC                                                   #  bind gadget to UDC driver (brings gadget online). This will only
                                                                  #  succeed if there are no gadgets already bound to the driver. Do
                                                                  #  lsmod and if there's anything in there like g_*, you'll need to
                                                                  #  rmmod it before bringing this gadget online. Otherwise you'll get
                                                                  #  "device or resource busy."
```

zador-blood-stained, 2016-12-23:
This kernel is called "dev" for a reason, USB OTG support for H3 is still experimental. According to https://linux-sunxi.org/Linux_mainlining_effort @megous is working on it, but issues are disabled in the kernel repository (https://github.com/megous/linux)

liftoff, 2016-12-23:
I guess I just have to be patient then.

closed 2016-12-23 by liftoff

RoganDawes, 2017-01-31:
It looks like there might be some development on this front?

https://www.spinics.net/lists/arm-kernel/msg554766.html

liftoff, 2017-02-02:
Awesome.  I already gave away my Orange Pi Zero though because it lacked this feature =(