armbian/build #590: H2+/H3 | kernel 4.9 missing interactive CPU gov

Fourdee, 2016-12-26:
As title. 
```interactive``` does not exist in the default kernel config, any reasons for this?: https://github.com/igorpecovnik/lib/blob/master/config/kernel/linux-sun8i-dev.config#L602-L621

OPi-Zero/NanoPi Neo:
```
root@DietPi:~# cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors
conservative ondemand userspace powersave performance schedutil
```

If ```interactive``` needs testing with 4.9, let me know and i'll re-build to run some tests?

zador-blood-stained, 2016-12-26:
Mainline kernel in general doesn't have interactive governor - it is specific to Android kernels. Ondemand or schedutil should be used as default on mainline.

Fourdee, 2016-12-26:
@zador-blood-stained 

> Mainline kernel in general doesn't have interactive governor - it is specific to Android kernels.

Interesting, thanks.

closed 2016-12-26 by Fourdee