armbian/build #591: OrangePi Zero kernel 4.9 | Missing WiFi modules

Fourdee, 2016-12-26:
Freshly built with ARMbian build tools today. Appears no modules are being built for this device (including the onboard ```xradio_wlan```)?
```
root@DietPi:~# ls -lha /lib/modules/4.9.0-sun8i/kernel/net/wireless/
total 240K
drwxr-xr-x  2 root root 4.0K Dec 26 14:57 .
drwxr-xr-x 38 root root 4.0K Dec 26 14:57 ..
-rw-r--r--  1 root root 232K Dec 26 14:56 cfg80211.ko
```

Firmware appears to be there:
```
root@DietPi:~# ls -lha /lib/firmware
total 660K
drwxr-xr-x 12 root root 4.0K Dec 26 15:02 .
drwxr-xr-x 17 root root 4.0K Dec 26 15:34 ..
drwxr-xr-x  2 root root 4.0K Dec 26 15:02 ap6210
drwxr-xr-x  2 root root 4.0K Dec 26 15:02 ap6212
drwxr-xr-x  2 root root 4.0K Dec 26 15:02 brcm
-rw-r--r--  1 root root 2.0K Dec 26 14:33 dvb-demod-mn88472-02.fw
-rw-r--r--  1 root root 8.0K Dec 26 14:33 dvb-usb-tbs5520.fw
drwxr-xr-x  4 root root 4.0K Dec 26 15:01 imx
-rw-r--r--  1 root root  45K Dec 26 14:33 mt7601.bin
-rw-r--r--  1 root root  45K Dec 26 14:33 mt7601u.bin
-rw-r--r--  1 root root 8.0K Dec 26 14:33 rt2870.bin
drwxr-xr-x  2 root root 4.0K Dec 26 15:02 RTL8192SU
drwxr-xr-x  2 root root 4.0K Dec 26 15:02 rtlwifi
drwxr-xr-x  2 root root 4.0K Dec 26 15:02 sdma
drwxr-xr-x  2 root root 4.0K Dec 26 15:02 ti-connectivity
-rw-r--r--  1 root root 248K Dec 26 14:33 v4l-coda960-imx6dl.bin
-rw-r--r--  1 root root 248K Dec 26 14:33 v4l-coda960-imx6q.bin
drwxr-xr-x  2 root root 4.0K Dec 26 15:02 vpu
drwxr-xr-x  2 root root 4.0K Dec 26 15:02 xr819
```

zador-blood-stained, 2016-12-26:
Xradio driver is not present in the branch that is used currently for sun8i-dev. There is out of tree driver [here](https://github.com/fifteenhex/xradio/tree/original), so it needs to be converted to a patch, or rather 2 patches - one for the code and one for the DT bindings.

zador-blood-stained, 2016-12-29:
@Fourdee 
Xradio driver was added to the Orange Pi Zero dev configuration.

Please keep in mind that this driver is still experimental and hardware chip lacks necessary documentation, so we can't provide any support for it apart from compilation related fixes.

Fourdee, 2016-12-29:
@zador-blood-stained 
Legend, thanks 👍 

> Please keep in mind that this driver is still experimental and hardware chip lacks necessary documentation, so we can't provide any support for it apart from compilation related fixes.

Aye, no worries. I'll build it and test in the next few days.

Fourdee, 2016-12-30:
Ok, so report of testing:
- 🈯️ Module appears to be fully functional
- 🈯️ ```iwlist wlan0 scan``` appears to be fully functional
```
wlan0     Scan completed :
          Cell 01 - Address: E8:DE:27:4A:D0:66
                    Channel:8
                    Frequency:2.447 GHz (Channel 8)
                    Quality=52/70  Signal level=-58 dBm
                    Encryption key:on
```
- 🈴 Fails to obtain DHCP offer. No connection.

```
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 6
No DHCPOFFERS received.
No working leases in persistent database - sleeping.
```

```
root@orangepizero:~# dmesg | grep xradio
[  809.145941] xradio_wlan mmc1:0001:1 wlan0: disabling HT/VHT due to WEP/TKIP use
[  908.350301] xradio_wlan mmc1:0001:1 wlan0: disabling HT/VHT due to WEP/TKIP use
[  997.973991] xradio_wlan mmc1:0001:1 wlan0: disabling HT/VHT due to WEP/TKIP use
```

```
[  994.885338] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  997.430259] wlan0: authenticate with e8:de:27:4a:d0:66
[  997.958586] wlan0: send auth to e8:de:27:4a:d0:66 (try 1/3)
[  997.973840] wlan0: authenticated
[  997.973991] xradio_wlan mmc1:0001:1 wlan0: disabling HT/VHT due to WEP/TKIP use
[  997.991633] wlan0: associate with e8:de:27:4a:d0:66 (try 1/3)
[  997.994079] wlan0: RX AssocResp from e8:de:27:4a:d0:66 (capab=0x411 status=12 aid=5)
[  997.994092] wlan0: e8:de:27:4a:d0:66 denied association (code=12)
```

Even when disabling power management:
```
iwconfig wlan0 power off

root@orangepizero:~# iwconfig wlan0 | grep "Power Management"
          Power Management:off

```

module contains no params we can set:
```
root@orangepizero:~# systool -vm xradio_wlan
Module = "xradio_wlan"

  Attributes:
    coresize            = "92375"
    initsize            = "0"
    initstate           = "live"
    refcnt              = "1"
    taint               = ""
    uevent              = <store method only>

  Sections:
    .ARM.exidx          = "0xbf923574"
    .ARM.exidx.exit.text= "0xbf923548"
    .ARM.exidx.init.text= "0xbf923540"
    .ARM.extab          = "0xbf923550"
    .alt.smp.init       = "0xbf9232ce"
    .bss                = "0xbf924b40"
    .data.unlikely      = "0xbf924920"
    .data               = "0xbf923bf4"
    .exit.text          = "0xbf9217f0"
    .gnu.linkonce.this_module= "0xbf924940"
    .init.text          = "0xbf92b000"
    .note.gnu.build-id  = "0xbf9217f4"
    .rodata.str         = "0xbf921c1c"
    .rodata             = "0xbf921818"
    .rodata.str1.4      = "0xbf9223f0"
    .strtab             = "0xbf92e984"
    .symtab             = "0xbf92b004"
    .text               = "0xbf913000"
    __bug_table         = "0xbf9221ec"
```

So yep, looks like experimental driver is not functional at the moment. With cause most likely:
```
[  997.994079] wlan0: RX AssocResp from e8:de:27:4a:d0:66 (capab=0x411 status=12 aid=5)
```

zador-blood-stained, 2016-12-30:
I tested it at least 3 times - 2 times with a Jessie image and 1 time with a Xenial image, using NetworkManager (nmtui). 

With Jessie I had association issues similar to yours the first time I tried, but the second time I tried it worked fine for me, and I just booted this image again and it associated with my AP fine:
```
root@orangepizero:~# iwconfig wlan0
wlan0     IEEE 802.11  ESSID:"<censored>"
          Mode:Managed  Frequency:2.472 GHz  Access Point: 12:34:56:78:90:ab
          Bit Rate=39 Mb/s   Tx-Power=20 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=70/70  Signal level=-32 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:1  Invalid misc:0   Missed beacon:0
```

```
root@orangepizero:~# dmesg | grep 'wlan'
[   13.838909] xradio_wlan: unknown parameter 'macaddr' ignored
[   25.584870] wlan0: authenticate with 12:34:56:78:90:ab
[   25.585378] wlan0: send auth to 12:34:56:78:90:ab (try 1/3)
[   25.719619] wlan0: authenticated
[   25.731009] wlan0: associate with 12:34:56:78:90:ab (try 1/3)
[   25.734804] wlan0: RX AssocResp from 12:34:56:78:90:ab (capab=0x431 status=0 aid=1)
[   25.739065] wlan0: associated
```

With Xenial I had to run `iwlist wlan0 scan` before it displayed my network in `nmtui`, but it is probably caused by my wireless setup (wireless channel 13 HT40-), apart from this it worked without any issues.

Please note that this driver is based on legacy kernel driver which has known problems in AP mode: https://forum.armbian.com/index.php/topic/3046-opi-zero-and-hostapd/#entry21921

zador-blood-stained, 2016-12-30:
Also you may try uncommenting this line in the patch
```
#ccflags-$(CONFIG_WLAN_VENDOR_XRADIO) += -DXRADIO_DISABLE_HW_CRYPTO
```
and compile the kernel again since it seems that you have some enctyption related issues.

Also please check your AP/router logs if you can since you are getting `denied association (code=12)` and logs on the other side may contain more useful info.

Fourdee, 2016-12-30:
@zador-blood-stained 
Thanks 👍 , i'll run further tests based on your notes.

> Edit: My AP is set up as WPA2 only PSK/AES, without WEP, TKIP or WPS.

Same here WPA2-PSK + AES

fifteenhex, 2016-12-31:
I added the -DXRADIO_DISABLE_HW_CRYPTO option when trying to figure out why hostap with WPA wasn't working (works now). You shouldn't enable it.

If anything enable the line that defines DEBUG. That will give you a load of output and show you the sorts of messages going to and from the chip.

And some issues with wpa_supplicant seem to be due to there not being enough entropy available at boot for key generation etc.

ThomasKaiser, 2017-01-02:
> And some issues with wpa_supplicant seem to be due to there not being enough entropy available at boot for key generation etc.

As expected since [DietPi removes `haveged` intentionally](https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh#L103).

Fourdee, 2017-01-02:
@ThomasKaiser 
Pretty sure @fifteenhex wasn't using DietPi for those tests.

fifteenhex, 2017-01-04:
@Fourdee 

I'm using a buildroot based system. wpa_supplicant is pretty much useless there unless I tell it to use /dev/urandom instead of /dev/random with the -e switch.

zador-blood-stained, 2017-01-09:
Since we have more reports about working wireless on the mainline (i.e. [here](https://forum.armbian.com/index.php/topic/2808-orange-pi-zero-went-to-the-market/page-10#entry22735) and [here](https://forum.armbian.com/index.php/topic/3243-orange-pi-zero-xradio-st-cw1200/#entry22860)) and only one report of non-working wireless that looks like a hardware quality issue, this can be considered solved, and I'll try to see if the legacy driver can be improved based on the mainline one.

closed 2017-01-09 by zador-blood-stained