armbian/build #592: NanoPi Neo Air, kernel 4.9 | No cpufreq/gov

Fourdee, 2016-12-26:
No CPUfreq/gov information available: 
https://github.com/Fourdee/DietPi/issues/640#issuecomment-269232618

```
cat: /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors: No such file or directory
```

Tested with NanoPi Neo (not air) and 4.9, thats fine:
```
root@DietPi:~# cat  /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors
conservative ondemand userspace powersave performance schedutil
```

zador-blood-stained, 2016-12-26:
If you started providing DietPi for H3 boards based on mainline kernel, you'll encounter tons of small inconsistencies like this. And this is exactly the reason why there are only nightly Armbian images with mainline kernel for H3 boards.

```
[    8.464810] of: dev_pm_opp_of_cpumask_add_table: couldn't find opp table for cpu:0, -19
[    8.464859] cpu cpu0: dev_pm_opp_get_opp_count: OPP table not found (-19)
```
suggests that DT for NanoPi Neo is missing CPU operating points.

For the record, NanoPi Neo is currently using Device tree from OrangePi One, and NanoPi Air is using DT from NanoPi Neo 😄

Fourdee, 2016-12-26:
@zador-blood-stained 

> If you started providing DietPi for H3 boards based on mainline kernel, you'll encounter tons of small inconsistencies like this. And this is exactly the reason why there are only nightly Armbian images with mainline kernel for H3 boards (and only for several of them).

Yep, i'am considering it for some devices, but each device will need to be thoroughly tested before we consider replacing 3.x images.

> For the record, NanoPi Neo is currently using Device Tree for the OrangePi One, and NanoPi Air is using DT for the NanoPi Neo

The strange thing is, the NanoPi Neo with 4.9 seems alot more stable than 3.x at the moment lol. USB WiFi was non-functional on 3.x: https://github.com/Fourdee/DietPi/issues/667. Seems to be working great on 4.9.

The other reason for 4.9 is for working ```brcm``` module (as per FriendlyARM's image) instead of ```dhd``` for NanoPi Neo Air: https://github.com/Fourdee/DietPi/issues/640
Has this also been changed for 3.x? https://github.com/igorpecovnik/lib/commit/8f4e8e25e425169c90d12c67d21e5b84f5cdb231

zador-blood-stained, 2016-12-26:
> USB WiFi was non-functional on 3.x: Fourdee/DietPi#667. Seems to be working great on 4.9.

Not a surprise. 8192cu is out of tree for 3.x kernels, ~~and it's possible that firmware is needed~~ (not in the case you linked). Also kernel upgrade will break it since DKMS is not supported at the moment. While in mainline a good enough driver is present in the kernel source to start with and it will work out of the box if you have correct firmware installed.

> The other reason for 4.9 is for working brcm module (as per FriendlyARM's image) instead of dhd for NanoPi Neo Air: Fourdee/DietPi#640
Has this also been changed for 3.x? 8f4e8e2

Sorry, don't have Neo or Neo Air, so can't test or change anything (that requires testing) related to these boards.

Fourdee, 2016-12-27:
> For the record, NanoPi Neo is currently using Device Tree for the OrangePi One, and NanoPi Air is using DT for the NanoPi Neo 

Ah yes, air ```.dtb``` isnt available yet, or shared with Neo?:
```
/boot/dtb-4.9.0-sun8i/sun8i-h3-nanopi-neo.dtb
/boot/dtb/sun8i-h3-nanopi-neo.dtb
```

Was worth a shot, but as expected, no WiFi :
```
cp /boot/dtb/sun8i-h3-orangepi-one.dtb /boot/dtb/sun8i-h3-nanopi-neo.dtb
cp /boot/dtb-4.9.0-sun8i/sun8i-h3-orangepi-one.dtb /boot/dtb-4.9.0-sun8i/sun8i-h3-nanopi-neo.dtb
```

Fourdee, 2016-12-27:
@zador-blood-stained 

> suggests that DT for NanoPi Neo is missing CPU operating points.

```.dtb``` are binary, where are they generated in ARMbian source code (if at all)? Thinking we could simply copy the ```orangepi-one.dtb``` cpu vars into the ```nanopi-neo.dtb``` ?

zador-blood-stained, 2016-12-27:
> Ah yes, air .dtb isnt available yet, or shared with Neo?

Don't think Air DTB is avaliable, but [this](https://github.com/igorpecovnik/lib/blob/master/patch/kernel/sun8i-dev/neo-air-wireless.patch) patch adds wireless to Neo DT.

> .dtb are binary, where are they generated in ARMbian source code (if at all)? Thinking we could simply copy the orangepi-one.dtb cpu vars into the nanopi-neo.dtb ?

They are generated at kernel compilation time, so you need to create a kernel patch to add DVFS specific bits and modify u-boot patch to load respective DT for Neo and Air.

zador-blood-stained, 2016-12-28:
@Fourdee 

I updated the patches in d5b51f6a. Please check if DVFS works for Neo and Neo Air.

Fourdee, 2016-12-30:
@zador-blood-stained 

> I updated the patches in d5b51f6. Please check if DVFS works for Neo and Neo Air.

Legend 👍  Thanks again 😃 

Will test and report back.

Fourdee, 2016-12-30:
@zador-blood-stained 

Checked Air. Works a treat, thanks 👍 
```
root@nanopiair:~# cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
schedutil

root@nanopiair:~# cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors
conservative ondemand userspace powersave performance schedutil

root@nanopiair:~# cat /sys/devices/system/cpu/cpu0/cpufreq/stats/time_in_state
120000 0
240000 4732
312000 1276
480000 850
624000 850
816000 867
1008000 15405

```

Ondemand:
```
root@DietPi:~# cat /sys/devices/system/cpu/cpu0/cpufreq/stats/time_in_state
120000 450
240000 521
312000 179
480000 80
624000 0
816000 0
1008000 10880
```

closed 2016-12-30 by zador-blood-stained