armbian/build #593: OrangePi Zero legacy kernel - g_hid (and g_multi) doesn't work as a module

arcaine2, 2016-12-28:
Those 2 doesn't load as a module despite being present in /lib/modules:

```
root@orangepizero:/lib/modules/3.4.113-sun8i/kernel/drivers/usb/gadget# strace modprobe g_hid
execve("/sbin/modprobe", ["modprobe", "g_hid"], [/* 20 vars */]) = 0
brk(0)                                  = 0xb8ee7000
uname({sys="Linux", node="orangepizero", ...}) = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6ff9000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=22214, ...}) = 0
mmap2(NULL, 22214, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6ff3000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\3\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0Mw\1\0004\0\0\0"..., 512) = 512
lseek(3, 908188, SEEK_SET)              = 908188
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 2880) = 2880
lseek(3, 904740, SEEK_SET)              = 904740
read(3, "A4\0\0\0aeabi\0\1*\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\3\f"..., 53) = 53
fstat64(3, {st_mode=S_IFREG|0755, st_size=911068, ...}) = 0
mmap2(NULL, 980392, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6eba000
mprotect(0xb6f95000, 61440, PROT_NONE)  = 0
mmap2(0xb6fa4000, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xda000) = 0xb6fa4000
mmap2(0xb6fa7000, 9640, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6fa7000
close(3)                                = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6ff2000
set_tls(0xb6ff2850, 0xb6fd2050, 0xb6ff2f38, 0xb6ff2850, 0xb6fd2050) = 0
mprotect(0xb6fa4000, 8192, PROT_READ)   = 0
mprotect(0xb6ffb000, 4096, PROT_READ)   = 0
mprotect(0xb6fd1000, 4096, PROT_READ)   = 0
munmap(0xb6ff3000, 22214)               = 0
brk(0)                                  = 0xb8ee7000
brk(0xb8f08000)                         = 0xb8f08000
uname({sys="Linux", node="orangepizero", ...}) = 0
stat64("/etc/modprobe.d", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
openat(AT_FDCWD, "/etc/modprobe.d", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_DIRECTORY|O_CLOEXEC) = 3
fcntl64(3, F_GETFD)                     = 0x1 (flags FD_CLOEXEC)
getdents64(3, /* 5 entries */, 32768)   = 168
newfstatat(3, "orangepizero.conf", {st_mode=S_IFREG|0644, st_size=14, ...}, 0) = 0
newfstatat(3, "fbdev-blacklist.conf", {st_mode=S_IFREG|0644, st_size=390, ...}, 0) = 0
newfstatat(3, "xradio_wlan.conf", {st_mode=S_IFREG|0644, st_size=46, ...}, 0) = 0
getdents64(3, /* 0 entries */, 32768)   = 0
close(3)                                = 0
stat64("/run/modprobe.d", 0xbe8e6870)   = -1 ENOENT (No such file or directory)
stat64("/lib/modprobe.d", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
openat(AT_FDCWD, "/lib/modprobe.d", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_DIRECTORY|O_CLOEXEC) = 3
getdents64(3, /* 3 entries */, 32768)   = 80
newfstatat(3, "aliases.conf", {st_mode=S_IFREG|0644, st_size=655, ...}, 0) = 0
getdents64(3, /* 0 entries */, 32768)   = 0
close(3)                                = 0
open("/lib/modprobe.d/aliases.conf", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_GETFL)                     = 0xa0000 (flags O_RDONLY|O_LARGEFILE|O_CLOEXEC)
fstat64(3, {st_mode=S_IFREG|0644, st_size=655, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6ff8000
_llseek(3, 0, [0], SEEK_CUR)            = 0
read(3, "# These are the standard aliases"..., 4096) = 655
read(3, "", 4096)                       = 0
close(3)                                = 0
munmap(0xb6ff8000, 4096)                = 0
open("/etc/modprobe.d/fbdev-blacklist.conf", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_GETFL)                     = 0xa0000 (flags O_RDONLY|O_LARGEFILE|O_CLOEXEC)
fstat64(3, {st_mode=S_IFREG|0644, st_size=390, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6ff8000
_llseek(3, 0, [0], SEEK_CUR)            = 0
read(3, "# This file blacklists most old-"..., 4096) = 390
read(3, "", 4096)                       = 0
close(3)                                = 0
munmap(0xb6ff8000, 4096)                = 0
[... 86 lines cut ...]
modprobe: DEBUG: ../libkmod/libkmod-module.c:1736 kmod_module_get_initstate() could not open '/sys/module/g_hid': No such file or directory
modprobe: DEBUG: ../libkmod/libkmod-module.c:1374 kmod_module_get_options() modname=snd_pcsp mod->name=g_hid mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1374 kmod_module_get_options() modname=snd_usb_audio mod->name=g_hid mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1374 kmod_module_get_options() modname=cx88_alsa mod->name=g_hid mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1374 kmod_module_get_options() modname=snd_atiixp_modem mod->name=g_hid mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1374 kmod_module_get_options() modname=snd_intel8x0m mod->name=g_hid mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1374 kmod_module_get_options() modname=snd_via82xx_modem mod->name=g_hid mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1374 kmod_module_get_options() modname=xradio_wlan mod->name=g_hid mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1374 kmod_module_get_options() modname=hdmi mod->name=g_hid mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1374 kmod_module_get_options() modname=disp mod->name=g_hid mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1726 kmod_module_get_initstate() could not open '/sys/module/g_hid/initstate': No such file or directory

modprobe: DEBUG: ../libkmod/libkmod-module.c:1736 kmod_module_get_initstate() could not open '/sys/module/g_hid': No such file or directory
modprobe: DEBUG: ../libkmod/libkmod-module.c:728 kmod_module_get_path() name='g_hid' path='/lib/modules/3.4.113-sun8i/kernel/drivers/usb/gadget/g_hid.ko'
modprobe: DEBUG: ../libkmod/libkmod-module.c:728 kmod_module_get_path() name='g_hid' path='/lib/modules/3.4.113-sun8i/kernel/drivers/usb/gadget/g_hid.ko'
insmod /lib/modules/3.4.113-sun8i/kernel/drivers/usb/gadget/g_hid.ko
modprobe: DEBUG: ../libkmod/libkmod-module.c:728 kmod_module_get_path() name='g_hid' path='/lib/modules/3.4.113-sun8i/kernel/drivers/usb/gadget/g_hid.ko'
modprobe: INFO: ../libkmod/libkmod-module.c:868 kmod_module_insert_module() Failed to insert module '/lib/modules/3.4.113-sun8i/kernel/drivers/usb/gadget/g_hid.ko': No such device
modprobe: ERROR: could not insert 'g_hid': No such device
modprobe: DEBUG: ../libkmod/libkmod-module.c:452 kmod_module_unref() kmod_module 0xb80763d0 released
modprobe: DEBUG: ../libkmod/libkmod.c:408 kmod_pool_del_module() del 0xb80763d0 key='g_hid'
modprobe: INFO: ../libkmod/libkmod.c:321 kmod_unref() context 0xb8076128 released
```

My goal is to have both keyboard and mouse appear and working on the host with g_hid. There's currently patched hid.c for 3.4 kernel on armbian [forums](https://forum.armbian.com/index.php/topic/1487-orange-pi-one-otg-hid/) but it requires g_hid to be compiled directly into the kernel, ruling out the other gadget feature and it emulates keyboard only.

zador-blood-stained, 2016-12-29:
Legacy kernel is broken somewhere, and unless somebody finds and fixed what is wrong, your best option is waiting for the mainline support for Orange Pi Zero.

arcaine2, 2016-12-29:
I see. To be honest, it's fine for me to have g_hid compiled directly into the kernel, but i still need it to act as both keyboard and mouse and current "patched" hid.c allows to emulate keyboard only. So far i tried adding another desciptor and structs for mouse and registering them in hidg_init the same way as keyboard but it crashes kernel during boot. Any pointers on solving that would be great.

zador-blood-stained, 2016-12-29:
According to [gadget_hid](https://git.kernel.org/cgit/linux/kernel/git/stable/linux-stable.git/tree/Documentation/usb/gadget_hid.txt) documentation

> 	You can add as many HID functions as you want, only limited by
	the amount of interrupt endpoints your gadget driver supports.

I'm pretty sure that this is the main limitation for supporting more than one device and using composite gadget drivers.

arcaine2, 2016-12-29:
Yes, i saw that, but it's not clear enough for me on how to implement that and i couldn't find any example, sadly. There's a [patch ](https://github.com/pelya/android-keyboard-gadget/blob/master/kernel-3.4.patch) for 3.4 kernel used in Android phones, but it's still a bit different here.

zador-blood-stained, 2017-05-28:
Mainline H3 OTG support will be present starting with kernel 4.12, and unfortunately looks like nobody is still interested in fixing the legacy kernel.
https://linux-sunxi.org/Linux_mainlining_effort#Status_Matrix

closed 2017-05-28 by zador-blood-stained

SeldaBagcan, 2017-10-14:
Armbian has builds with 4.13 kernel version on their site but still no g_hid support. g_ether and g_serial works. I know some programming but have little experience with linux. Should i try to implement g_hid module or there is something wrong with something else so its not that easy to add g_hid?