armbian/build #594: para_tab ptbl data

simoncheen, 2016-12-30:
hello,

https://github.com/igorpecovnik/lib/blob/master/patch/kernel/sun8i-default/z-0003-add-additional-video-modes.patch

I have saw your patch , but can not unstand what's ' dclk/int/mx/vst/my/mhbw/vfp/mhfp/+x', can you please tell me what's that para? and where I could get it , if I wanna to connect to a 2048*1536 LCD via HDMI ?

note: I was saw a link: 
https://forum.armbian.com/index.php/topic/752-tutorial-h3disp-change-display-settings-on-h3-devices/page-2#entry17751

you said "you need to make a calculation"
http://k.kramerav.com/support/bwcalculator.asp
but you haven't said , how to get those para from this bwcalculator.

thanks

jernejsk, 2017-01-06:
I guess the easiest way for adding custom resolutions would be to make special U-Boot binary, which would be loaded through FEL and it would do calculation based on your connected HDMI monitor. It would simply print out two arrays which are usually the hardest to calculate.

jernejsk, 2017-01-17:
@simoncheen 
I just managed to modify a u-boot which reads out preffered values and calculates array for you. You can build it by yourself from https://github.com/jernejsk/u-boot/tree/h3_video_helper or if you trust me, just use prebuilt version from https://down.nu/temp/u-boot-h3-video-helper.bin

You can write this U-Boot to SD card or better yet, use fel mode to run it. For exact modifications, please check this patch and make analogous changes: https://github.com/igorpecovnik/lib/blob/master/patch/kernel/sun8i-default/z-0003-add-additional-video-modes.patch

simoncheen, 2017-01-17:
@jernejsk 
It done, I have do that successfully via EDID, 

thanks for helping

jernejsk, 2017-01-17:
Just for clarification: using my U-Boot or using something else? I just want to know if my method worked for you.

simoncheen, 2017-01-17:
I use this https://github.com/igorpecovnik/lib and compiled a new kernel , intall the new debs , and reboot it . it go success. the currently version has EDID , my old armbian version have no.

jernejsk, 2017-01-17:
Can you please describe what did you do? Last kernel change was done before you opened the issue.

If you have time, can you please confirm that values for your mode are the same as reported in my modified U-Boot?

simoncheen, 2017-01-17:
thing is that, my armbian was installed since 6 months ago, and I never upgrade it,  I tried to complied a new kernel yesterday , and found the LCD is working !

jernejsk, 2017-01-17:
Ah, you wanted a resolution which is already included in aforementioned patch. Well, you could just update the system, no need for kernel recompilation.

simoncheen, 2017-05-09:
hello,

long time no chat,  I fright out the 2048x1536 , the resolution that I told to you was wrong, because I forgot I was re-cmpile the kernel. now , I pasted the patch file. 

build/patch/kernel/sun8i-default/z-004-add-addonal-2048x1536.patch

=========================
diff -ur a/drivers/video/sunxi/disp/de/disp_display.c b/drivers/video/sunxi/disp/de/disp_display.c
--- a/drivers/video/sunxi/disp/de/disp_display.c	2017-01-10 00:58:30.819762504 +0800
+++ b/drivers/video/sunxi/disp/de/disp_display.c	2017-01-10 00:59:53.043760236 +0800
@@ -764,6 +764,10 @@
 			width = 1680;
 			height = 1050;			
 			break;
+		case DISP_TV_MOD_2048_1536P:
+			width = 2048;
+			height = 1536;
+			break;
 		default:
 			printk("err:function:%s, line:%d,output_mode=%d\n", __func__, __LINE__, output_mode);
 		}
@@ -842,6 +846,10 @@
 			width = 1680;
 			height = 1050;			
 			break;
+                case DISP_TV_MOD_2048_1536P:
+                        width = 2048;
+                        height = 1536;
+                        break;
 		default:
 			printk("err:function:%s, line:%d,output_mode=%d\n", __func__, __LINE__, output_mode);
 		}
diff -ur a/drivers/video/sunxi/disp2/disp/de/disp_display.c b/drivers/video/sunxi/disp2/disp/de/disp_display.c
--- a/drivers/video/sunxi/disp2/disp/de/disp_display.c	2017-01-10 00:58:30.843762503 +0800
+++ b/drivers/video/sunxi/disp2/disp/de/disp_display.c	2017-01-10 01:02:09.595756469 +0800
@@ -775,6 +775,10 @@
 			width = 1680;
 			height = 1050;			
 			break;
+		case DISP_TV_MOD_2048_1536P:
+			width = 2048;
+			height = 1536;
+			break;
 		}
 	}
 	/* FIXME: add other output device res */
@@ -851,6 +855,10 @@
 			width = 1680;
 			height = 1050;
 			break;
+		case DISP_TV_MOD_2048_1536P:
+			width = 2048;
+			height = 1536;
+			break;
 		}
 	}
 	/* FIXME: add other output device res */
diff -ur a/drivers/video/sunxi/disp2/hdmi/aw/hdmi_bsp_sun8iw7.c b/drivers/video/sunxi/disp2/hdmi/aw/hdmi_bsp_sun8iw7.c
--- a/drivers/video/sunxi/disp2/hdmi/aw/hdmi_bsp_sun8iw7.c	2017-01-10 00:58:30.855762503 +0800
+++ b/drivers/video/sunxi/disp2/hdmi/aw/hdmi_bsp_sun8iw7.c	2017-01-16 18:32:23.269582210 +0800
@@ -70,7 +70,8 @@
 	{{37        , 4     , 0,    96,     5,   3,     4,  1,   1, 0,  0,  0,      152,    48,    112,       0,    42, 1,  1	}},
 	{{38        , 4     , 0,    96,     5,   7,     3,  1,   3, 0,  0,  80,     176,    64,    112,       0,    32, 1,  1	}},
 	{{39        , 4     , 0,    96,     5,   9,     3,  1,  18, 0,  1,  160,    216,    32,    152,     132,    37, 1,  1	}},
-	{{40        , 2     , 0,    96,     6,   3,     4,  2,   1, 0,  0,  144,     64,   104,    184,      26,    44, 1,  1	}},	
+	{{40        , 2     , 0,    96,     6,   3,     4,  2,   1, 0,  0,  144,     64,   104,    184,      26,    44, 1,  1	}},
+	{{41	    , 1     , 0,    96,     8,   3,     6,  0,   1, 0,  0,  128,    208,   128,    216,       0,    45, 1,  1}},
 };
 
 static unsigned char ca_table[64]=
[... 51 lines cut ...]
 #define HDMI1680_1050P          40
+#define HDMI2048_1536P          41
 #define HDMI1080P_24_3D_FP  (HDMI1080P_24 +0x80)
 #define HDMI720P_50_3D_FP   (HDMI720P_50  +0x80)
 #define HDMI720P_60_3D_FP   (HDMI720P_60  +0x80)
diff -ur a/include/video/sunxi_display2.h b/include/video/sunxi_display2.h
--- a/include/video/sunxi_display2.h	2017-01-10 00:58:30.939762501 +0800
+++ b/include/video/sunxi_display2.h	2017-01-10 01:06:19.711749570 +0800
@@ -136,7 +136,8 @@
 	DISP_TV_MOD_1360_768P           = 0x22,
 	DISP_TV_MOD_1440_900P           = 0x23,
 	DISP_TV_MOD_1680_1050P          = 0x24,
-	DISP_TV_MODE_NUM                = 0x25,
+	DISP_TV_MOD_2048_1536P          = 0x25,
+	DISP_TV_MODE_NUM                = 0x26,
 }disp_tv_mode;
 ========================

you can check it , it works for me , but you have to edit the h3disp file for adding this:
--------------
            37|2048x1536) # res
                HDMIMode=37
                pll_video=688
                ;; 
--------------

ThomasKaiser, 2017-05-09:
Please look at your comment above. If you want us to add support for this resolution please use exactly the patch you used and submit as a pull request.

simoncheen, 2017-05-09:
I'm a new user of github , can you please tell me how to make a  pull request ?

simoncheen, 2017-05-09:
this is the file

[z-004-add-addonal-2048x1536.txt](https://github.com/armbian/build/files/987331/z-004-add-addonal-2048x1536.txt)

igorpecovnik, 2017-05-09:
Check this: https://docs.armbian.com/Process_Contribute/#collaborate-on-the-project

ThomasKaiser, 2017-05-09:
Since we already have a patch to add all those additional resolutions I tried to integrate it there to keep number of patches low: https://github.com/armbian/build/commit/d2d85a09648c4878f46176edacd3d1f71009408d

Please test immediately and report back :)

simoncheen, 2017-05-09:
I have tried my patch file , and works , but haven't tried your file , because I have no compile platform currently. but I'm 100% sure my patch file is working, and I'm on using it. but please do not forget to add this in the h3disp.
=======================
            37|2048x1536) # res
                HDMIMode=37
                pll_video=688
                ;; 
========================

closed 2017-05-09 by igorpecovnik

maximilliangreat, 2019-03-10:
[u-boot-h3-video-helper.bin.zip](https://github.com/armbian/build/files/2949990/u-boot-h3-video-helper.bin.zip)

> @simoncheen
> I just managed to modify a u-boot which reads out preffered values and calculates array for you. You can build it by yourself from https://github.com/jernejsk/u-boot/tree/h3_video_helper [...]

As the link to "u-boot-h3-video-helper.bin" is dead, I'm attaching new prebuild version.