armbian/build #598: mpv builds

zerkalica, 2017-01-12:
In latest debian jessie desktop i saw mpv version 0.22, but in debian repository very old 0.6 version.
It's you own build of mpv? I can't find it in you armbian repository.

Do you think move you armbian packages to launchpad ppa, arm bulds propably supported: https://dev.launchpad.net/CommunityARMBuilds, http://askubuntu.com/questions/371339/how-to-make-packages-for-arm-in-launchpad

igorpecovnik, 2017-01-12:
IIRC we used to build it, now mpv is stock and we focus on Xenial desktop which is little newer than Jessie ATM.

> Do you think move you armbian packages to launchpad ppa, arm bulds propably supported: 

No need.

zerkalica, 2017-01-12:
xenial and yakkety mpv "stock" versions are too old too - 0.18
For some applications i need a ppa with fresh builds, like this https://launchpad.net/~mc3man/+archive/ubuntu/mpv-tests.
I wrote to mc3man about arm builds too.

ppa better than custom repository. Integration with launchpad services and with ubuntu (add-apt-repository is common way for installing ppas), simple search, multi-target builds, community, free resources. There is some ppa for raspberry pi, like [ubuntu-raspi2](https://launchpad.net/~ubuntu-raspi2)

PS: I simply update armbian to yakkety and all works fine on my banana pro, including mpv with vdpau.

zador-blood-stained, 2017-01-12:
> xenial and yakkety mpv "stock" versions are too old too - 0.18
For some applications i need a ppa with fresh builds, like this https://launchpad.net/~mc3man/+archive/ubuntu/mpv-tests.
I wrote to mc3man about arm builds too.

We tried providing updated versions of some packages and it didn't work perfectly, so IMO it's better to leave it as is, and end users can add custom compatible PPAs if they want.

> ppa better than custom repository. Integration with launchpad services and with ubuntu (add-apt-repository is common way for installing ppas), simple search, multi-target builds, community, free resources. There is some ppa for raspberry pi, like ubuntu-raspi2

"better" is a matter of opinion. For this project a custom repository works fine and I don't see any reasons to change the current setup. Launchpad is a third party service and add-apt-repository is Ubuntu specific while we are providing Debian packages too (yes, add-apt-repository may work on Debian, but with packages like mpv there will be a dependencies problem - already been there, don't see any reasons to start it again).

closed 2017-02-02 by zador-blood-stained