armbian/build #600: catch 22

trzetrzebinski, 2017-01-14:
Hi,

I've downloaded and installed both of your'd systems for OrangePi zero. Because of its infrastructure I use it headless and will it so use further. But I have catch 22 problem. Your gorggious system is asking me for new root password. After I write password system automaticly reboots and my Pytty sesion is broken. I have to enter new session and then Your system ask me to change password for root. After I change it your gorgious system automaticly reboots and my Pytty sesion is broken. I have to enter new session and then Your system ask me to change password for root.After I change it your gorgious system automaticly reboots and my Pytty sesion is broken. .. Enought???? Congratulation !!!! Your's system is realy great !!!

Trzetrzebinski

trzetrzebinski, 2017-01-14:
Hi,

I've downloaded and installed both of your'd systems for OrangePi zero. Because of its infrastructure I use it headless and will it so use further. But I have catch 22 problem. Your gorggious system is asking me for new root password. After I write password system automaticly reboots and my Pytty sesion is broken. I have to enter new session and then Your system ask me to change password for root. After I change it your gorgious system automaticly reboots and my Pytty sesion is broken. I have to enter new session and then Your system ask me to change password for root.After I change it your gorgious system automaticly reboots and my Pytty sesion is broken. .. Enought???? Congratulation !!!! Your's system is realy great !!!

Trzetrzebinski

jadr2ddude, 2017-01-14:
Are you using serial or ssh?

zador-blood-stained, 2017-01-14:
@trzetrzebinski 
To the forum please. https://forum.armbian.com/ This tracker is only for development related issues, as you should have noticed before opening an issue.

closed 2017-01-14 by zador-blood-stained

trzetrzebinski, 2017-01-15:
shh

trzetrzebinski, 2017-01-15:
Dear Misha, I thing the Grrrrrrrreeeeeeeeeaaaaat bug in system logic IS an
Developmeny issue. Am I right???? If not pls define development.

Regards

W.Matuszczak

zador-blood-stained, 2017-01-15:
> Dear Misha, I thing the Grrrrrrrreeeeeeeeeaaaaat bug in system logic IS an
Developmeny issue. Am I right???? If not pls define development.

As per https://github.com/igorpecovnik/lib/blob/master/.github/ISSUE_TEMPLATE.md, development is something related to scripts or configuration files from **this repository**, and there are no scripts or configuration files that would cause described behavior (either forced reboot after failed root password change attempt or second forced root password change if previous one finished successfully).

trzetrzebinski, 2017-01-15:
> github/ISSUE_TEMPLATE.md, development is something related to scripts or

It is YOUR choice,

I DO NOT completly agree with your explanation. Look, I have NOT to have to
use your system, there is a lot better choices.

Regards

igorpecovnik, 2017-01-15:
You made a mistake and you were politely advised to put your question into right place (since you failed to read instructions at first place). This is not end user support channel, but forum: https://forum.armbian.com Yes, there are other choices, but there you probably won't get any answer at all.