armbian/build #604: OPi Zero with 4.9 kernel | No sound cards available (expansion board)

Fourdee, 2017-01-18:
> https://github.com/Fourdee/DietPi/issues/679#issuecomment-273484449

Is this is a simple case of loading a specific sunxx audio/codec module, or fex job?

zador-blood-stained, 2017-01-18:
It's a simple matter of waiting until we switch to the 4.10 branch or grabbing the patches from upstream and adding DT patches to activate the codec on different boards
https://linux-sunxi.org/Linux_mainlining_effort#Status_Matrix

Fourdee, 2017-01-18:
> It's a simple matter of waiting until we switch to the 4.10 branch

Most likely, you guys will have 4.10 support, before I get a chance to do the other 😉 .

> https://linux-sunxi.org/Linux_mainlining_effort#Status_Matrix

Assuming ```H3``` here applies to ```H2+``` on OPi Zero?

zador-blood-stained, 2017-01-18:
> Assuming H3 here applies to H2+ on OPi Zero?

Yes

closed 2017-02-04 by Fourdee