armbian/build #605: apt.armbian.com does not work on non-legacy hosts

synnack, 2017-01-23:
$ sudo apt-get update
Err http://apt.armbian.com jessie InRelease
  
Err http://apt.armbian.com jessie Release.gpg                                           
  Could not resolve 'apt.armbian.com'

$ host apt.armbian.com
apt.armbian.com has address 185.94.112.80

Our ARM devices do not get legacy IP, because they do not need them (and we don't have enough legacy IP space). 

Please consider making apt.armbian.com reachable on the current internet (IPv6).

simonvetter, 2017-06-17:
+1, would love this!

jadr2ddude, 2017-07-09:
I can setup a squid proxy to ipv6 on my server if you want

igorpecovnik, 2017-07-09:
I contacted my hosting provider a month ago and his words were: "it will be arranged, but it's very likely not to happen soon"

jadr2ddude, 2017-07-13:
@igorpecovnik What hosting provider do you have?

igorpecovnik, 2017-07-13:
Hosting is provided by @laurivosandi

edolnx, 2017-08-01:
As a suggestion, turning on Cloudflare free tier for the apt.armbian.com domain will give it an IPv6 endpoint. It may, however, mess with metrics related to usage if you are using a traditional log scraper for that sort of thing.

igorpecovnik, 2017-08-01:
Thank you for tip. Well, there is another problem, not exposed here - we already create more traffic, that is acceptable for free versions. Adding IPv6 and creating additional costs (just) because of that, doesn't sound reasonable. For future proof we seek geo load balancing with good access in china and we are at enterprise level in no time :) If we can get that tailored for non-profit, that perhaps yes.

laurivosandi, 2017-08-01:
It sounds you're abusing Armbian infrastructure @synnack, for any larger installation it is necessary to set up your own mirror or caching web proxy. So until you're not a paying customer for Igor or something there isn't much that can be complained about :D

ghmer, 2017-10-06:
+1
It's 2017 - No excuse for not having ipv6 support

igorpecovnik, 2017-10-06:
Tomorrow :) https://www.armbian.com -> 

> 6. October 2017. We are upgrading main download and package repository server during this week. There will be 20x download speed upgrade. Server migration will complete during this weekend. In case of download troubles use this backup torrent download option.

jadr2ddude, 2017-10-06:
Hi! I set up a temporary mirror (IPv6 only) here: https://armbian.projectpanux.com/

mcr, 2017-12-11:
If jadr2ddude's mirror is acceptable, then just add the AAAA RR.  How much space do we need for the mirror?  I have spare bandwidth in my linodes; just tell me how much disk space I'll need for rsync.

jadr2ddude, 2017-12-11:
@mcr i had to shut down my mirror temporarily. I will set it back up at a later time.
@igorpecovnik how much bandwidth do you think we are talking about? if there is not that much IPv6 traffic possible, I can set up one of my older machines and give you SSH access.

mcr, 2017-12-11:
I think that the IPv6 usage will be moderate, it's just a question of how big the repo is....

jadr2ddude, 2017-12-12:
@mcr My copy from the mirror was 5.4 GB

mcr, 2017-12-12:
Jaden Weiss <notifications@github.com> wrote:
    > @mcr My copy from the mirror was 5.4 GB

Not that big.
Is there an rsync source?

igorpecovnik, 2018-08-18:
ipv6 is up. Thanks to https://k-space.ee/

closed 2018-08-18 by igorpecovnik