armbian/build #606: Can't rebuild orange pi zero dev kernel.

y-salnikov, 2017-01-23:
I'm trying to build kernel with this command:
"./compile.sh BRANCH=dev BOARD=orangepizero PROGRESS_DISPLAY=plain  KERNEL_ONLY=y
es  RELEASE=jessie PROGRESS_LOG_TO_FILE=yes FORCE_CHECKOUT=no"
And get this error message:
...
Device Tree Source is not correctly specified.
Please define 'CONFIG_DEFAULT_DEVICE_TREE'
or build with 'DEVICE_TREE=<device_tree>' argument

make[1]: *** [arch/arm/dts/sun8i-h2plus-orangepi-zero.dtb] Error 1
make: *** [dts/dt.dtb] Error 2
[ error ] ERROR in function compile_uboot [ common.sh:118 ]
[ error ] U-boot file not found [ u-boot-sunxi-with-spl.bin ]
[ o.k. ] Process terminated

But it's compile fine with FORCE_CHECKOUT=yes.

zador-blood-stained, 2017-01-23:
`FORCE_CHECKOUT=no` disables all additional patches including the patch that contains OPi Zero config.

y-salnikov, 2017-01-24:
So, how should i modify kernel before compilation?

igorpecovnik, 2017-01-24:
Check create_patch switch, which will wait for your uboot and kernel changes and will also create patch on your changes.

closed 2017-01-24 by y-salnikov