armbian/build #611: NanoPi M1 |  rtl8192 causes system reboot (kernel bug)

Fourdee, 2017-01-28:
Serial log, occurs when bringing up the adapter, during DHCP it seems:
https://github.com/Fourdee/DietPi/issues/567#issuecomment-275846502

Fourdee, 2017-01-28:
When using static, wlan index changes:
```
root@DietPi:~# ifup wlan0
wlan0: ERROR while getting interface flags: No such device
Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlan0 ; No such device.
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; No such device.
Error for wireless request "Set Encode" (8B2A) :
    SET failed on device wlan0 ; No such device.
Error for wireless request "Set ESSID" (8B1A) :
    SET failed on device wlan0 ; No such device.
wpa_supplicant: /sbin/wpa_supplicant daemon failed to start
run-parts: /etc/network/if-pre-up.d/wpasupplicant exited with return code 1
Failed to bring up wlan0.

root@DietPi:~# dmesg | grep wlan
[   11.410640] systemd-udevd[197]: renamed network interface wlan0 to wlan2
```

ThomasKaiser, 2017-01-28:
Interesting, it's new to me that M1 has onboard Wi-Fi? Or do you plan to support random 3rd party USB Wi-Fi dongles with DietPi? Why not reading through [documentation](https://github.com/igorpecovnik/lib.docs/blob/master/docs/User-Guide_Advanced-Features.md#how-to-build-a-wireless-driver) how to do this first when you want to do this (Armbian refrains from that, if it's not onboard it's up to the user to build the appropriate kernel module with legacy kernels -- that's one of the advantages of switching to mainline, a lot more 3rd party hardware works out of the box since appropriate drivers included in kernel)

> When using static, wlan index changes

This is an issue you have to report *here*? That the config you chose for whatever reasons with DietPi somehow depends on how you defined udev rules in DietPi?

Fourdee, 2017-01-28:
@ThomasKaiser 
As always Thomas, an endless joy of helpfulness.

Lets try and clear this up

> Armbian refrains from that, if it's not onboard it's up to the user to build the appropriate kernel module with legacy kernels

Are you saying ARMbian does NOT support common USB WiFi adapters with 3.x? I failed to see any mention of this in your documentation: 
https://github.com/igorpecovnik/lib.docs/blob/master/docs/User-Guide_Advanced-Features.md#how-to-build-a-wireless-driver

> The config you chose for whatever reasons with DietPi somehow depends on how you defined udev rules in DietPi?

udev rules are unchanged from the ARMbian build. 

For DietPi, we prefer to use bare-bones and simplisitc ```/etc/network/interfaces``` rather than the bulky and somewhat unnecessary ```network-manager```.

martinayotte, 2017-01-28:
If you've tried several WiFi dongles, it is pretty normal that index is changing.
Just look at the content of /etc/udev/rules.d/70-persistent-net.rules files, you will see all the MAC you've tried in past.

zador-blood-stained, 2017-01-28:
> Are you saying ARMbian does NOT support common USB WiFi adapters with 3.x?

This depends on your definition of "common". For example rtl8192cu module from 4.4 mainline kernel supports 72 different VID:PID pairs which means different hardware variants. Can you say that you provide support for all of them for the Raspberry Pi DietPi images?

Look, it's not a secret that 3.x kernels and especially 3.x kernels from Allwinner may contain bugs and outdated drivers and the only reason we keep them around for A10/A20 boards are multimedia features. Do you suggest that anybody should backport the whole wireless stack to the kernel code base that was released 4.5 years ago?

> `[   11.410640] systemd-udevd[197]: renamed network interface wlan0 to wlan2`

I'm not sure if this is the result of "persistent naming", but wireless device indexes and names are not absolute and are expected to change, so you should be ready to handle that (i.e. by checking `iw dev` for the list of wireless devices)

Fourdee, 2017-01-28:
@zador-blood-stained 

> Can you say that you provide support for all of them for the Raspberry Pi DietPi images?

If its an official rtl8192cu inside the device, and used as-per Realtek's manufactures specs, then yes.

> Do you suggest that anybody should backport the whole wireless stack to the kernel code base that was released 4.5 years ago?

Nope. I'am asking for a simple yes/no answer:
- Does ARMbian currently provide support USB WiFi adapters on 3.x?

If the answer is no, then please close the ticket.

zador-blood-stained, 2017-01-28:
> Nope. I'am asking for a simple yes/no answer:
> Does ARMbian currently provide support for USB WiFi adapters on 3.x?

Unfortunately it's not that simple.

- We try to provide as much support as possible for the onboard wireless adapters (yes, these ase not USB but SDIO)
- We try to provide as much support as possible for the wireless adapters that we actually own and can test
- We provide support for compiling out of tree kernel modules (including wireless drivers) by providing the kernel headers. Finding and compiling necessary drivers is up to the end users.

I looked through the linked issue and I don't see a VID:PID values (from `lsusb`), so I can;t check if there are other drivers or fixes for the existing drivers that apply to this particular Wi-Fi adapter.

ThomasKaiser, 2017-01-29:
> For DietPi, we prefer to use bare-bones and simplisitc /etc/network/interfaces rather than the bulky and somewhat unnecessary network-manager.

Sweet. And you still don't get why you (and DietPi users) are affected by device name changes (`udev`) and feel the urgent need to report this nonsense here?

closed 2017-02-04 by ThomasKaiser