armbian/build #616: Banan Pro no SATA power

Redwid, 2017-02-01:
Starting from release 5.01 there are no power supply for SATA on Banana Pro boards.
```
[    3.659655] ahci-sunxi 1c18000.sata: PHY power up failed.
[    3.659733] ahci-sunxi: probe of 1c18000.sata failed with error -5
```
Recent failures:
https://forum.armbian.com/index.php/topic/3385-banana-pro-sata/

Thread with some workarounds:
https://forum.armbian.com/index.php/topic/691-banana-pro-testers-wanted-sata-drive-not-working-on-some-boards/page-5#entry12416

My version is ARMBIAN Debian GNU/Linux 7 (wheezy) 4.8.4-sunxi:
```
root@bananapro ~ # dpkg -l | grep next
ii  linux-dtb-next-sunxi            5.23                              armhf        Linux DTB, version 4.8.4-sunxi
ii  linux-firmware-image-next-sunxi 5.23                              armhf        Linux kernel firmware, version 4.8.4-sunxi
ii  linux-headers-next-sunxi        5.23                              armhf        Linux kernel headers for 4.8.4-sunxi on armhf
ii  linux-image-next-sunxi          5.23                              armhf        Linux kernel, version 4.8.4-sunxi
ii  linux-u-boot-bananapi-next      5.23                              armhf        Uboot loader 2016.09
ii  linux-wheezy-root-next-bananapi 5.23                              armhf        Armbian tweaks for wheezy on bananapi (next branch)
```
I've do not made any changes in my hardware configuration, just made upgrade from 5.01 to 5.23 and my SATA disk is gone. There are no way to come back to 5.01 version right now:
```
root@bananapro ~ # apt-get install linux-u-boot-bananapi-next=5.01
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Version '5.01' for 'linux-u-boot-bananapi-next' was not found
```

I've tested on the latest Mainline version 4.9.5 - the same error with power up failed.

Just for information, [Bananian 16.04](https://www.bananian.org/download) works fine with SATA. Power supply is correct for SATA.

igorpecovnik, 2017-02-01:
You are using wrong image and it's normal that it don't work. Remove boot loader and install boot loader for Banana PRO and it should work.

```
dpkg -r linux-u-boot-bananapi-next
apt-get install linux-u-boot-bananapipro-next
```

igorpecovnik, 2017-02-01:
Selfbuild image SATA works so if you need it now, you need to build it on your own or wait that we push out update / fix.

zador-blood-stained, 2017-02-02:
Should be fixed in new Banana Pro specific images. Download page and documentation should be updated to reflect the changes.

closed 2017-02-02 by zador-blood-stained