armbian/build #617: nand-sata-install not properly creating symlinks?

danry25, 2017-02-02:
Using nand-sata-install on the Armbian_5.20_Orangepiplus2e_Debian_jessie_3.4.112_desktop.7z image, I noticed that /etc/resolv.conf is copied as a file instead of as a symlink, so when attached to a new network, network-manager can't change the DNS server. I'm guessing other symlinks are affected, not sure (but happy to check).

igorpecovnik, 2017-02-02:
We do rsync with switch "copy symlinks as symlinks". This is the exact command: https://github.com/igorpecovnik/lib/blob/master/scripts/nand-sata-install/usr/lib/nand-sata-install/nand-sata-install.sh#L106

zador-blood-stained, 2017-02-02:
`/etc/resolv.conf` is provided as a file by default in Armbian images (this solves some build time problems), so nand-sata-install doesn't change it.

zador-blood-stained, 2017-02-02:
https://github.com/igorpecovnik/lib/blob/master/distributions.sh#L93-L94

closed 2017-02-02 by zador-blood-stained

danry25, 2017-02-02:
Is that why it isn't getting updated by network manager when placed on a new network? Couldn't we just move it and create the proper symlink on first boot? Eg: mv /etc/resolv.conf /run/resolvconf/resolv.conf && ln -s ../run/resolvconf/resolv.conf /etc/resolv.conf