armbian/build #618: LVM is not working in linux-mvebu-default

iavael, 2017-02-07:
LVM is not working in linux-mvebu-default because device mapper subsystem is disabled
```
# CONFIG_BLK_DEV_DM is not set
```
I had my /var on lvm, so this change broke the boot.

iavael, 2017-02-07:
Why not to make kernel closer to ubuntu's one in term on features? E.g. there's also lack of CONFIG_ZSWAP, CONFIG_KEXEC, CONFIG_DEFAULT_SECURITY_APPARMOR, CONFIG_CIFS_FSCACHE, CONFIG_9P_FS, CONFIG_F2FS_FS_ENCRYPTION, CONFIG_XEN, that can be useful on this board for some users.

igorpecovnik, 2017-02-07:
You mean close to Ubuntu X86?

zador-blood-stained, 2017-02-07:
> CONFIG_BLK_DEV_DM, CONFIG_CIFS_FSCACHE, CONFIG_9P_FS, CONFIG_F2FS_FS_ENCRYPTION

Added with some other related options: 

> CONFIG_ZSWAP, CONFIG_KEXEC

Marked as "experimental" in 4.4, and I'm not sure kexec is useful outside of testing/debugging scenarios.

> CONFIG_XEN

I'll add it with related options if you need it, but not just to get it "closer to Ubuntu"

> CONFIG_DEFAULT_SECURITY_APPARMOR

And somedbody else will ask for another default, i.e.SELinux. You can enable AppArmor by using kernel command line "security" parameter.

> Why not to make kernel closer to ubuntu's one in term on features?

Because we have 22 differnet kernel configs for different kernel versions, so some features will be missing here and there.

iavael, 2017-02-07:
> You mean close to Ubuntu X86?
close to ubuntu arch-independent configuration
http://kernel.ubuntu.com/git/ubuntu/ubuntu-xenial.git/tree/debian.master/config/config.common.ubuntu
and ubuntu common armhf configuration
http://kernel.ubuntu.com/git/ubuntu/ubuntu-xenial.git/tree/debian.master/config/armhf/config.common.armhf

> Marked as "experimental" in 4.4
you mean zswap? it's disabled by default even if compiled in and is enabled via kernel cmdline

> I'm not sure kexec is useful outside of testing/debugging scenarios
systemd has it's support

> And somedbody else will ask for another default, i.e.SELinux.
Ubuntu supports apparmor and doesn't support selinux, that's why I talked about apparmor :)

closed 2017-05-28 by zador-blood-stained