armbian/build #623: Bug info collection script

jadr2ddude, 2017-02-17:
I think having a script to automatically pull together debugging info could make it easier to deal with bug reports. I started writing it when I got bored today.

So far I have implemented collection of the following info:

- Included by default:
  - dmesg output
  - lsmod
  - uname
  - /boot/armbianEnv.txt
- Can be added (but not default):
  - List of installed packages
  - IP Addresses
  - User-specified note/description

The script will upload the output to transfer.sh if a network connection is available and print a link in markdown, or otherwise save it to the SD/eMMC.
Also, do you think that the output should be in .tar.gz or .zip format (these are the 2 archive types supported by GitHub issue attachments)?

What other info do you think is important?

igorpecovnik, 2017-02-17:
Check this - it's not automated but it does similar:
https://github.com/igorpecovnik/lib/tree/master/scripts/armbianmonitor

ThomasKaiser, 2017-02-17:
@jadr2ddude please look at the `CollectSupportInfo` function in `armbianmonitor` that gets called when the script's argument is `-u`. Most information is gathered from `/etc/init.d/armhwinfo` which collects debug data on every startup (also `dmesg` output on shutdown) and writes them to rotated logfiles which then are simply submitted to an online pasteboard service together with some recent information (I just added `lsmod` output and `armbianEnv.txt` contents -- thanks for the inspiration)

jadr2ddude, 2017-02-17:
It appears that armbianmonitor only works with a network connection. It might be a good idea to put the output in the filesystem if we are debugging network issues (especially with boards with wifi). I can probably add this tomorrow - it should be pretty easy to do.

ThomasKaiser, 2017-02-17:
Well, please look at it in detail: https://github.com/igorpecovnik/lib/blob/11174f235b6bd8e47629cdc468859e9cc457d990/scripts/armbianmonitor/armbianmonitor#L685-L694

In fact, most information is already 'in the filesystem' in form of `armhwinfo.log`. I also thought a bit about it the last hours and came to the conclusion that installed packages, lsmod and armbianEnv.txt collection should also be moved to `armhwinfo` script since this allows us to analyze problems that are related to upgrades/changes the user made.

But I'm not clear what's the best way (`lsmod` output collected in both `start` and `stop` case of `armhwinfo` script might be useful) and want to think a bit about. IMO there's no need to rush, better let us collect useful items to add to `armhwinfo.log` and please don't overestimate this whole thing since often enough users are scared by the amount of information and fear providing any info at all.

zador-blood-stained, 2017-11-25:
Since we made some useful additions to `armbianmonitor -u` we should have a good enough bug info collection tool for our purposes, so let's close this.

closed 2017-11-25 by zador-blood-stained