armbian/build #624: Multiplie pll_video string

2345lug, 2017-02-24:
I'm use h3disp -m33 and see that script create multiplie pll_video in script.bin. And last pll_video is wrong.
I replace pll_video to correct and delete others and everything is work perfectly.

zador-blood-stained, 2017-02-25:
What version of the board support package do you have? Please provide output of
```
dpkg -l | grep linux
```

2345lug, 2017-02-27:
root@orangepipc:~# dpkg -l | grep linux
ii  bluez-tools                           0.2.0~20140808-3                     armhf        Set of tools to manage Bluetooth devices for linux
ii  console-setup-linux                   1.123                                all          Linux specific part of console-setup
ii  libselinux1:armhf                     2.3-2                                armhf        SELinux runtime shared libraries
ii  libv4l-0:armhf                        1.6.0-2                              armhf        Collection of video4linux support libraries
ii  libv4lconvert0:armhf                  1.6.0-2                              armhf        Video4linux frame format conversion library
ii  linux-headers-sun8i                   5.26                                 armhf        Linux kernel headers for 3.4.113-sun8i on armhf
ii  linux-image-sun8i                     5.26                                 armhf        Linux kernel, version 3.4.113-sun8i
ii  linux-jessie-root-orangepipc          5.23                                 armhf        Armbian tweaks for jessie on orangepipc (default branch)
ii  linux-libc-dev:armhf                  3.16.39-1+deb8u1                     armhf        Linux support headers for userspace development
ii  linux-u-boot-orangepipc-default       5.25                                 armhf        Uboot loader 2017.01
ii  util-linux                            2.25.2-6                             armhf        Miscellaneous system utilities
root@orangepipc:~#

zador-blood-stained, 2017-02-27:
> ii linux-jessie-root-orangepipc 5.23

Please make a system backup, update `linux-jessie-root-orangepipc` package (`apt-get install linux-jessie-root-orangepipc` or `apt-get dist-upgrade`) and check again

2345lug, 2017-02-27:
root@orangepipc:/home/lug# dpkg -l | grep linux
ii  bluez-tools                           0.2.0~20140808-3                     armhf        Set of tools to manage Bluetooth devices for linux
ii  console-setup-linux                   1.123                                all          Linux specific part of console-setup
ii  libselinux1:armhf                     2.3-2                                armhf        SELinux runtime shared libraries
ii  libv4l-0:armhf                        1.6.0-2                              armhf        Collection of video4linux support libraries
ii  libv4lconvert0:armhf                  1.6.0-2                              armhf        Video4linux frame format conversion library
ii  linux-base                            3.5                                  all          Linux image base package
ii  linux-headers-sun8i                   5.26                                 armhf        Linux kernel headers for 3.4.113-sun8i on armhf
ii  linux-image-sun8i                     5.26                                 armhf        Linux kernel, version 3.4.113-sun8i
ii  linux-jessie-root-orangepipc          5.25                                 armhf        Armbian tweaks for jessie on orangepipc (default branch)
ii  linux-libc-dev:armhf                  3.16.39-1+deb8u1                     armhf        Linux support headers for userspace development
ii  linux-u-boot-orangepipc-default       5.25                                 armhf        Uboot loader 2017.01
ii  util-linux                            2.25.2-6                             armhf        Miscellaneous system utilities

How i can place my bin file?

zador-blood-stained, 2017-06-05:
Tested with the latest version of h3disp and can't reproduce this, so this must be already fixed and any issues may be caused by improper upgrade from earlier versions.

closed 2017-06-05 by zador-blood-stained