armbian/build #628: Orange Pi Zero dev kernel, only one CPU core

ssinyagin, 2017-03-02:
I upgraded to the latest dev kernel and bootloader, according to your instructions:

```
aptitude remove ~nlinux-dtb ~nlinux-u-boot ~nlinux-image ~nlinux-headers 
aptitude remove ~nlinux-firmware ~narmbian-firmware ~nlinux-$(lsb_release -cs)-root
apt-get install linux-image-dev-sun8i linux-dtb-dev-sun8i linux-headers-dev-sun8i 
apt-get install linux-u-boot-orangepizero-dev linux-$(lsb_release -cs)-root-dev-orangepizero 
apt-get install armbian-firmware sunxi-tools swconfig a10disp
reboot
```

But the kernel sees only one CPU core instead of four:

```
root@orangepizero:~# uname -a
Linux orangepizero 4.10.1-sun8i #20 SMP Thu Mar 2 02:56:46 CET 2017 armv7l GNU/Linux
root@orangepizero:~# 

root@orangepizero:~# cat /proc/cpuinfo 
processor	: 0
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 100.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

Hardware	: Allwinner sun8i Family
Revision	: 0000
Serial		: 02c00042af6763fe
root@orangepizero:~# 

root@orangepizero:~# ls -l /sys/devices/system/cpu
total 0
drwxr-xr-x 5 root root    0 Mar  2 21:51 cpu0
drwxr-xr-x 4 root root    0 Mar  2 21:51 cpu1
drwxr-xr-x 4 root root    0 Mar  2 21:51 cpu2
drwxr-xr-x 4 root root    0 Mar  2 21:51 cpu3
drwxr-xr-x 4 root root    0 Mar  2 21:51 cpufreq
drwxr-xr-x 2 root root    0 Mar  2 21:51 cpuidle
drwxr-xr-x 2 root root    0 Mar  2 21:51 hotplug
-r--r--r-- 1 root root 4096 Mar  2 21:51 isolated
-r--r--r-- 1 root root 4096 Mar  2 21:51 kernel_max
-r--r--r-- 1 root root 4096 Mar  2 21:51 offline
-r--r--r-- 1 root root 4096 Mar  2 21:51 online
-r--r--r-- 1 root root 4096 Mar  2 21:51 possible
drwxr-xr-x 2 root root    0 Mar  2 21:51 power
-r--r--r-- 1 root root 4096 Mar  2 21:51 present
-rw-r--r-- 1 root root 4096 Mar  2 21:51 uevent

root@orangepizero:~# cpufreq-info 
cpufrequtils 008: cpufreq-info (C) Dominik Brodowski 2004-2009
Report errors and bugs to cpufreq@vger.kernel.org, please.
analyzing CPU 0:
  driver: cpufreq-dt
  CPUs which run at the same hardware frequency: 0 1 2 3
  CPUs which need to have their frequency coordinated by software: 0
  maximum transition latency: 4.24 ms.
  hardware limits: 120 MHz - 1.01 GHz
  available frequency steps: 120 MHz, 240 MHz, 312 MHz, 480 MHz, 624 MHz, 816 MHz, 1.01 GHz
  available cpufreq governors: conservative, ondemand, userspace, powersave, performance, schedutil
  current policy: frequency should be within 240 MHz and 1.01 GHz.
                  The governor "schedutil" may decide which speed to use
                  within this range.
  current CPU frequency is 1.01 GHz (asserted by call to hardware).
  cpufreq stats: 120 MHz:0.00%, 240 MHz:0.00%, 312 MHz:0.00%, 480 MHz:0.00%, 624 MHz:0.00%, 816 MHz:64.24%, 1.01 GHz:35.76%  (11070)
root@orangepizero:~# 

```

igorpecovnik, 2017-03-02:
We don't provide end user support for DEVelopment kernels for obvious reasons. This question was brought up many times on forum: https://forum.armbian.com/index.php?/search

closed 2017-03-02 by igorpecovnik

igorpecovnik, 2018-06-16:
> Well. It seems that as it was in dev , it is also in production.
> Only one cpu.

This is the build script engine issue tracker. Please report the issue to correct part of the forum and provide logs with armbianmonitor -u ... otherwise, we don't know much about the problem. I suspect you upgrade a kernel and forget to upgrade an u-boot as well.

jannoke, 2018-06-21:
will do. Just some info here - did not update uboot. Did not know i had to or it was not done automatically.

But it works perfectly well with all 4 cpus enabled on Armbian Bionic nightly mainline kernel master branch 4.17.y and all cpu's enabled, no uboot update. Just wrote image to sdcard.

Also for some reason the 4.14 v also heated the cpus so it was always running on 80 degree end. I would assume it could be be because only one cpu is available and must work all the time at highest frequency to keep up, but seemed to be still high on quite idle.

igorpecovnik, 2018-06-21:
> Did not know i had to or it was not done automatically

Automated updating works on the unchanged system. If you were changing things on your own, you might break it. Hard to say.

> Also for some reason the 4.14 v also heated the cpus so it was always running on 80 degree end

Put "overheating zero" or similar into our search engine:
https://www.armbian.com/search/

jannoke, 2018-06-21:
It's my 3d printer "bridge".

Now if I started checking ..I have to confess now that it's actually orangepi lite, not zero. So H2+ vs H3
I do have Zero on other printer but it's running dietpi so it does not apply here. But no heatsink on same setup and all cores enabled. So it's actually not that heavy workload.

I know about overheating. I have clocked down ddr and disabled cores if needed. I mostly run with 2 cores only. And with v4.14 and one core I got overheating problems and also printer started to pause if there was too fast printing or I tried to access web interface. Also I run it with heatsink. No active cooling however.
It would seem that cores would still run, but not available to OS to use. So if single core was running at full speed, then all of them would run at it and was just generating heat.

I also run Orangepi I96 which has only single core and can do same job with no interruptions.

I used this thread: https://forum.armbian.com/topic/1614-running-h3-boards-with-minimal-consumption/

Also it seems that these paths are not valid anymore with new kernel.

***

No hacking. Just made the image and installed it. I had older version which did not support neither my 5g wifi or my sjcam so I decided to just copy my few files to usb stick to restore them back later. Nothing OS related.Just wrote the img to sdcard and apt upgraded to latest and installed repetier. Thats it. Single core.

Repeated steps with v4.17 and all good.