armbian/build #632: Suggestion: use e2cp for generating images instead of mounting

jadr2ddude, 2017-03-10:
This would:
1. Allow images to be built in unprivileged Docker containers
2. Eliminate issues caused by killing compile.sh when compiling

I can do this later today if you think this will be helpful

igorpecovnik, 2017-03-11:
If this would eliminate the need to run under root, it's helpful.

jadr2ddude, 2017-03-11:
Ok. I am going to try to do that now then.

jadr2ddude, 2017-03-11:
Well, actually we still need root to install packages, but we could theoretically install them once then run it as a non-root user.

igorpecovnik, 2017-03-11:
Dependencies install sure ... but if regular run wouldn't require root, that would be nice.

zador-blood-stained, 2017-03-11:
I believe this (e2cp) won't work perfectly with different filesystems, and not sure if it works with ext4. For dropping superuser access requirement predefined entries in /etc/fstab may work (apart from parallel builds), but this will also require manually changing ownership of all newly created files, so it will be hard to completely drop root requirements for full image builds.

jadr2ddude, 2017-03-11:
@zador-blood-stained it is EXT4/3 only unfortunately

igorpecovnik, 2017-03-12:
Than we have to close this :( Also the development of e2cp seems to be dead.

closed 2017-03-12 by igorpecovnik