armbian/build #638: How can I control the image size under 1000MB

xsjqqq123, 2017-03-22:
Hello, I use orange pi zero to develop some tools, I find the system img is too big, So I comple it myself. Some packet in system may not my need.How can I exclude them and make the output img smaller? The document seems not mention it.

ThomasKaiser, 2017-03-22:
Just out of curiousity: What's the use case to strip the installed size to below 1GB?

ThomasKaiser, 2017-03-22:
Anyway, there are two opportunities that leave you both with an unsupported system later:

- choose dev kernel and also `btrfs` for the `rootfs`. This will [reduce the needed disk space to below 60%](https://github.com/igorpecovnik/lib/commit/b14da27a4181e8e232bd8f526e71d2a931a8252f#commitcomment-19484855). Please understand that we don't support dev kernel builds currently and that you should understand the pros and cons of btrfs as filesystem.
- diagnose the amount of space our default packages need and walk into dependency hell trying to deselect packages in `configuration.sh`. There are [several ways to do this](http://unix.stackexchange.com/questions/40442/which-installed-software-packages-use-the-most-disk-space-on-debian) eg. `aptitude -O installsize -F'%p %I' search '~i'` running in the final image.

There exist some urban myths claiming the more packages are installed the slower the system gets (BS) and it's a really great idea to choose an installation media of appropriate size (that's at least 8GB these days since SD cards with less capacity show both horrible performance and price/performance ratio)

ThomasKaiser, 2017-03-22:
Hmm... somewhat interesting (sun8i Xenial dev image):

    root@orangepiplus2e:~# wajig large
    Package                           Size (KB)        Status
    =================================-==========-============
    cpp-5                               11,899      installed
    g++-5                               12,598      installed
    systemd                             13,312      installed
    libperl5.22                         13,410      installed
    gcc-5                               13,611      installed
    locales                             13,684      installed
    libstdc++-5-dev                     13,840      installed
    git                                 15,840      installed
    iso-codes                           16,035      installed
    perl-modules-5.22                   17,260      installed
    vim-runtime                         26,870      installed
    libicu55                            28,460      installed
    unicode-data                        31,681      installed
    linux-image-dev-sun8i               44,757      installed
    linux-headers-dev-sun8i             71,385      installed
    linux-firmware                     152,061      installed

Vs. 'freed space' after potential removal:

    root@orangepiplus2e:~# wajig large | awk -F" " '/installed$/ {print $1}' | while read ; do
    > echo -e "${REPLY}: \c"
    > apt-get --assume-no --purge remove "${REPLY}" | grep "be freed" | awk '{print $4, $5}'
    > done
    cpp-5: 39.2 MB
    g++-5: 12.9 MB
    systemd: 18.6 MB
    libperl5.22: 39.5 MB
    gcc-5: 26.9 MB
    locales: 14.1 MB
    libstdc++-5-dev: 27.1 MB
    git: 16.2 MB
    iso-codes: 16.7 MB
    perl-modules-5.22: 57.1 MB
    vim-runtime: 29.1 MB
    libicu55: 42.1 MB
    unicode-data: 32.4 MB
    linux-image-dev-sun8i: 45.8 MB
    linux-headers-dev-sun8i: 73.1 MB
    linux-firmware: 156 MB

zador-blood-stained, 2017-03-22:
> How can I exclude them and make the output img smaller?

You can change package list variables using bash syntax and common Linux tools (grep, sed, ...) in `userpatches/lib.config`

Current package list is defined [here](https://github.com/igorpecovnik/lib/blob/master/configuration.sh#L120-L139) and I would recommend to start by removing packages defined in `PACKAGE_LIST_ADDITIONAL` variable

xsjqqq123, 2017-03-26:
I try to build the firmware myself, always fail to start the system. Really don't know why.
so, I try another method. I mount img from you, and rm all the files , copy others from runable system, It works. ^_^. files in image had limit below 400M, 
    But , how to resize the img to make it suitable for mini sd card(512M)？
    I tried "resize2fs Armbian_5.25_Orangepipc_Debian_jessie_default_3.4.113.img 480M" but get error string:"resize2fs: Bad magic number in super-block while trying to open..."
    thank you for your help.

xsjqqq123, 2017-03-26:
another question, how to remove welcome string when open ssh. sometimes cost a lot time..

zador-blood-stained, 2017-04-12:
> But , how to resize the img to make it suitable for mini sd card(512M)？
I tried "resize2fs Armbian_5.25_Orangepipc_Debian_jessie_default_3.4.113.img 480M" but get error string:"resize2fs: Bad magic number in super-block while trying to open..."

You can't resize the image file directly.

Assuming the filesystem has enough free space, you need to:
1. Mount the image to a loop device and scan the loop device for partitions (losetup, partprobe)
2. Resize the filesystem on the root partition (i.e. loop0p1)
3. Resize the root partition (i.e. with fdisk or parted)
4. Unmount the image file from the loop device
5. Truncate the image file

> another question, how to remove welcome string when open ssh. sometimes cost a lot time..

This was already changed so login process should not take that much time.
Also different MOTD components can now be disabled in /etc/default/armbian-motd

closed 2017-05-11 by zador-blood-stained