armbian/build #640: Merge Configuration File with Ubuntu Default Configuration?

cztian, 2017-03-23:
Some features in Debian/Ubuntu are not functioning due to the kernel configuration. Is it possible( At least in the mainline kernels) to merge the configuration file with Debian/Ubuntu configuration?
http://kernel.ubuntu.com/git/ubuntu/ubuntu-xenial.git/tree/debian.master/config/config.common.ubuntu (Arch unspecific)
http://kernel.ubuntu.com/git/ubuntu/ubuntu-xenial.git/tree/debian.master/config/arm64/config.common.arm64 (Arch arm64)

zador-blood-stained, 2017-03-23:
Please provide a list of kernel options that you need and kernel branches that those options should be added to (sunxi-next? mvebu-default/mvebu-next? etc.). Some options are hardware specific or arch specific and I would prefer to not bloat the kernel with drivers that can't be used on supported hardware.

zador-blood-stained, 2017-05-28:
No feedback in more than 2 mounth, closing

closed 2017-05-28 by zador-blood-stained