armbian/build #642: IR on OPI Zero

kutysam, 2017-03-23:
Hi, I'm using the nightlies. I'm not sure if this is an issue or is it a user problem.

I am trying to use infrared receiver on the OPI Zero without the expansion board. Basically I connected my 3 pin IR Receiver on 5v gnd and the last pin (IR-RX)
I did a modprobe sunxi-cir too. Unfortantely I am unable to get it working.
I'm unable to view the IR in "cat /proc/bus/input/devices"

lsmod shows me this
sunxi_cir               3825  0

dmesg | grep cir --> Nothing comes out.

Any ideas? I'll be glad to test out

zador-blood-stained, 2017-03-23:
If you are using fresh enough nightlies you need to use DT overlays. Please refer to the documentation: https://docs.armbian.com/Hardware_Allwinner_overlays/

kutysam, 2017-03-24:
Thank you for pointing me to the right direction. I'll read up! Will close it now and comment if it works!

closed 2017-03-24 by kutysam