armbian/build #643: GDB not working for orange pi zero

justin-romano, 2017-03-25:
Reproduction steps
```
#include <stdio.h>

int main (void)
    {
        printf("hello, world!\n");
        return 0;
    }

```
compiled with g++ and -g for symbols.
#./hello
hello, world!

all good

but
#gdb ./hello
(gdb) r
Starting program: /root/src/WiringOP-Zero/examples/hello

```
Program received signal SIGSEGV, Segmentation fault.
0xb6fd9822 in ?? () from /lib/ld-linux-armhf.so.3
(gdb)
```
Please refer to https://github.com/raspberrypi/linux/issues/766
for a fix
Im sure if i can get this working then my remote debugging will work also.

Any help would be appreciated.
Cheers.

zador-blood-stained, 2017-03-25:
Did you try installing `libc6-dbg`? I'm guessing that you are using the legacy kernel, but on mainline it seems to be helping.
```
root@orangepiplus2e:~# gdb ./test
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.04) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./test...done.
(gdb) r
Starting program: /root/test

Program received signal SIGSEGV, Segmentation fault.
0xb6fd9822 in ?? () from /lib/ld-linux-armhf.so.3
(gdb) q
A debugging session is active.

        Inferior 1 [process 2149] will be killed.

Quit anyway? (y or n) y
root@orangepiplus2e:~# apt-get install libc6-dbg
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  libc6-dbg
0 upgraded, 1 newly installed, 0 to remove and 6 not upgraded.
Need to get 6,042 kB of archives.
After this operation, 40.0 MB of additional disk space will be used.
Get:1 http://ports.ubuntu.com xenial-security/main armhf libc6-dbg armhf 2.23-0ubuntu7 [6,042 kB]
Fetched 6,042 kB in 2s (2,064 kB/s)
Selecting previously unselected package libc6-dbg:armhf.
(Reading database ... 30914 files and directories currently installed.)
Preparing to unpack .../libc6-dbg_2.23-0ubuntu7_armhf.deb ...
Unpacking libc6-dbg:armhf (2.23-0ubuntu7) ...
Setting up libc6-dbg:armhf (2.23-0ubuntu7) ...
root@orangepiplus2e:~# gdb ./test
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.04) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./test...done.
(gdb) r
Starting program: /root/test
Cannot parse expression `.L966 4@r4'.
warning: Probes-based dynamic linker interface failed.
Reverting to original interface.

Hello, world
[Inferior 1 (process 2518) exited normally]
(gdb) q
root@orangepiplus2e:~#
```

vargavik, 2017-04-12:
Hi,
Situation is the same on the Orange Pi PC using legacy kernel.
Installing debug symbols does not help in getting closer.

Dump in my case:
GNU gdb (Linaro_GDB-2017.02) 7.12.1.20170216-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-w64-mingw32 --target=arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
Reading /lib/ld-linux-armhf.so.3 from remote target...
warning: File transfers from remote targets can be slow. Use "set sysroot" to access files locally instead.
Reading /lib/ld-linux-armhf.so.3 from remote target...
Reading /lib/ld-2.23.so from remote target...
Reading /lib/.debug/ld-2.23.so from remote target...
**0xb6fd7a40 in ?? () from target:/lib/ld-linux-armhf.so.3**

Program received signal SIGSEGV, Segmentation fault.
**0xb6fd9822 in ?? () from target:/lib/ld-linux-armhf.so.3**

Program terminated with signal SIGSEGV, Segmentation fault.
The program no longer exists.

zador-blood-stained, 2017-04-12:
Just tested a fresh Debian Jessie image with the legacy kernel.
Local debugging works for me:
```
root@orangepiplus2e:~# gdb ./test
GNU gdb (Debian 7.7.1+dfsg-5) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./test...(no debugging symbols found)...done.
(gdb) r
Starting program: /root/test
hello, world!
[Inferior 1 (process 1337) exited normally]
(gdb) q
root@orangepiplus2e:~# g++ -g test.c -o test
root@orangepiplus2e:~# gdb ./test
GNU gdb (Debian 7.7.1+dfsg-5) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./test...done.
(gdb) r
Starting program: /root/test
hello, world!
[Inferior 1 (process 1347) exited normally]
(gdb) q
root@orangepiplus2e:~#
```

If remote debugging fails for you, then it's most likely your remote debugging setup problem and not Armbian problem.

vargavik, 2017-04-12:
Sorry, I forget to mention it was about remote debugging and I use Ubuntu Xenial.
Using gdb locally works just fine.

zador-blood-stained, 2017-04-12:
> Sorry, I forget to mention it was about remote debugging and I use Ubuntu Xenial.
Using gdb locally works just fine.

This issue was started as a problem with local debugging, and according to the tests it works fine (with some workarounds). Remote debugging issues are not related to the kernel configuration or Armbian build configuration, and unfortunately I have no idea what may be the cause here.

vargavik, 2017-04-12:
However it is off-topic, but let me post, the solution to remote debugging problem can be found here:
https://forum.armbian.com/index.php?/topic/1188-debugging-fails-with-illegal-instruction/#comment-9016

zador-blood-stained, 2017-04-12:
`CONFIG_OABI_COMPAT` may be only a workaround and not a proper solution since it affects the kernel size and performance. Especially since we are using `THUMB2_KERNEL` for several kernel configurations to optimize the kernel size, which is not compatible with `OABI_COMPAT`.
If your cross-compiler and/or GDB can't work without compatibility with the old ABI, then please upgrade them or recompile the kernel, it's not that hard.

Also according to this `OABI_COMPAT` is not a solution either: https://bugs.launchpad.net/gdb/+bug/1576432/comments/6

zador-blood-stained, 2017-04-26:
Closing since local debugging works; remore debugging problems are related to remote toolchains, so not Armbian related issue.

closed 2017-04-26 by zador-blood-stained

AlekseyShi, 2017-06-28:
Required packages: apt-get install libc6-dbg