armbian/build #644: Wrong path in /nand-sata-install.sh

Tanarri, 2017-03-28:
Hi,

I have read the program and want to understand it, and (I think) I found a bug in the file "/usr/lib/nand-sata-install/nand-sata-install.sh"

Line 236
`mkdir -p /mnt/rootfs/media/mmc/boot`

correct should be (without "/" between "mmc" + "boot")
`mkdir -p /mnt/rootfs/media/mmcboot`

in the next lines this newly generated path it used:
echo "$sduuid	/media/**mmcboot**	ext4    ${mountopts[ext4]}" >> /mnt/rootfs/etc/fstab
echo "/media/**mmcboot**/boot  

Greetings
TanarRi

igorpecovnik, 2017-03-29:
It's o.k. this way. Whole boot media is mounted under "mmcboot" and than we bind mount boot sub directory to /boot.

Tanarri, 2017-03-29:
Thanks for info.
I'm believe the line 236 should change from 
`	mkdir -p /mnt/rootfs/media/mmc/boot` 
to
`	mkdir -p /mnt/rootfs/media/mmcboot `

But you are the programmer - so you might be right

pfoo, 2017-04-11:
nand-sata-install.sh might do some cleanup after a successfull migration. In case of booting using mmc and root on sata :  
on sdcard : /mnt/bootfs/ and /mnt/rootfs/ can be removed
on sata drive : /media/mmcroot/ and /media/mmc/ can be removed if not using btrfs.

SATA /media/mmcroot/ is only used when using btrfs filesystem

I'm actually not understanding why creating /media/mmc/boot/ (and  therefore /media/mmc) (line 241) as those directories are not used anywhere else.

closed 2019-03-06 by lanefu