armbian/build #647: http://www.armbian.com is down

brianddk, 2017-04-02:
Wrong place for the issue I know, but seemed the closest I bucket I could find to notify the Admins.

For others trying to access the site... I checked the archive and it is fairly complete:

http://web.archive.org/web/2/https://www.armbian.com/

brianddk, 2017-04-02:
The Armbian download archive is up as well:
https://dl.armbian.com/

brianddk, 2017-04-02:
Docs site is up too:
https://docs.armbian.com/

brianddk, 2017-04-02:
https://www.armbian.com is back up

closed 2017-04-02 by brianddk

igorpecovnik, 2017-04-02:
Server related issues. Fixed now.

ffactory-ofcl, 2019-04-07:
Seems down again?

dengsn, 2019-04-07:
Yes, seems down to me as well. Cannot reach the site and my distro won't update,

ffactory-ofcl, 2019-04-07:
I have found many of the builds here, but **be warned**, they may not be original. Has files for amlogic s905 and s912 as well as others

igorpecovnik, 2019-04-07:
Switch failure in the network center. It should be back soon.

igorpecovnik, 2019-04-07:
One of the backups for download or upading: http://mirrors.netix.net/armbian/

ffactory-ofcl, 2019-04-07:
Please consider making this mirror more prominent as I couldn't find it. Maybe mention it in the Wikipedia article or have it on the front page so the people browsing with waybackmachine can find it.

frownbreaker, 2019-04-14:
When will this be fixed? 

I've a console only orange pi I want to update. But don't know how to use the mirror provided the apt upgrade looks for apt.armbian.com

hwacookie, 2019-04-14:
Same situation here. Could someone be so kind to give the server a kick?

JpegXguy, 2019-04-14:
check armbian twitter @frownbreaker @hwacookie

frownbreaker, 2019-04-15:
Thanks all fixed.

igorpecovnik, 2019-04-15:
We had to change some hardware https://twitter.com/armbian/status/1117446855808495619 but since we didn't build and can't afford (armbian is non-profit and free for you) redundancy on all levels, services were put offline.