armbian/build #648: [RfE] Ease migration from Bananian?

ThomasKaiser, 2017-04-03:
Nico just informed me that [Bananian went EOL](https://www.bananian.org/news#the_end_-_2017-04-02) unfortunately (security fixes will be provided for some time but users need to migrate next year).

I thought I open this issue here as a place for us and **experienced** Bananian users willing to contribute to discuss whether/how Armbian should provide some means of smooth migration.

IMO only Bananian releases already running with 4.4LTS should be supported and then also a distro upgrade to stretch should the be mandatory.

igorpecovnik, 2017-04-03:
We have this [upgrade.sh](https://github.com/igorpecovnik/lib/blob/master/upgrade.sh) tool but it's also deprecated. Last time, year(s) ago, it was successful in migrating from Bananian, but it need RFC to meet today's reality. I am also not familiar with current Bananian state.

Thank you Nico for all of his contribution!

ThomasKaiser, 2017-04-03:
> We have this upgrade.sh tool but it's also deprecated.

Yes, and it would also need some checks:

- check current Bananian installation. If it's not both Jessie and kernel 4.4 then stop and let the user upgrade before (this will also hopefully rule out the occurences of users running off from SD cards that got broken in the meantime)
- check what needs to be adjusted. This could be become PITA since crossgrading from Bananian needs a lot more love to implement perfectly compared to upgrading really old Armbian installations (where we currently recommend to backup user data and start from scratch)

A final result of this discussion might not be a tool but maybe just an article how to transfer data/settings to a new Armbian installation. Let's wait and see. Fortunately no one is in a hurry.

nisenbeck, 2017-04-03:
Hi Thomas, Hi Igor,
thank you for this open discussion!

Bananian ships "bananian-update" which was intended to do distribution updates. Maybe it can also be used to offer a migration/upgrade to Armbian?

If a migration from Bananian to Armbian is not possible, I would use bananian-update to show some EOL information in the future. Checking the kernel version and Bananian release is pretty simple. I can provide you those checks.

However, let me know if you need my help with this topic. I would really love to see a migration method from Bananian to Armbian and would also ship it via bananian-update.

ThomasKaiser, 2017-04-03:
@nisenbeck well, getting kernel or Bananian version is no problem at all. I'm more concerned about differences regarding partitioning, boot scripts (related are [DT overlays](https://docs.armbian.com/User-Guide_Allwinner_overlays/)), users who run Bananian off a SATA or USB disk already or those using SD cards that are pretty old and will then die after Armbian migration when doing the dist-upgrade to Stretch later.

As already said: No need to hurry, let's collect thoughts first.

thdankert, 2017-04-26:
Just to chime in: I took the plunge and tried the old upgrade.sh tool to go from Bananian 16.04 to Armbian.
TL;DR: It kind-of-worked, but required manual intervention, mostly because I have my rootfs on an external drive.

I've been using a 4.4 mainline kernel before and have my rootfs on a SATA disk (`/boot` is still on the SD card).
Here are my observations:
- did not remove `bananian-etc`, leading to problems during kernel installation with existing files 
- u-boot was successfully upgraded
- generated boot.cmd assumes all files are in directory `/boot` on the SD, but the bananian "copy to sata disk" script moves it directly to the root dir of the SD card

After fixing boot.cmd and (re-)installing the correct kernel and firmware packages my board was booting fine.

Here's my upgrade.log, in case you're interested:
```
No packages will be installed, upgraded, or removed.
0 packages upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Need to get 0 B of archives. After unpacking 0 B will be used.
No packages will be installed, upgraded, or removed.
0 packages upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Need to get 0 B of archives. After unpacking 0 B will be used.
Note: selecting "linux-image-dev-rockchip" instead of the
      virtual package "linux-image-4.10.0-rockchip"
Note: selecting "linux-image-sun8i" instead of the
      virtual package "linux-image-3.4.113-sun8i"
"linux-image-3.4.112-sun8i" exists in the package database, but it is not a
real package and no package provides it.
"linux-image-4.8.3-sun8i" exists in the package database, but it is not a
real package and no package provides it.
"linux-image-4.7.3-sun8i" exists in the package database, but it is not a
real package and no package provides it.
Note: selecting "linux-image-dev-sun8i" instead of the
      virtual package "linux-image-4.10.0-sun8i"
"linux-image-4.9.4-sun8i" exists in the package database, but it is not a
real package and no package provides it.
The following packages will be REMOVED:
  linux-image-4.4-bananian linux-image-4.4.34-bananian
0 packages upgraded, 0 newly installed, 2 to remove and 2 not upgraded.
Need to get 0 B of archives. After unpacking 45.3 MB will be freed.
(Reading database ... 51042 files and directories currently installed.)
Removing linux-image-4.4-bananian (16.04.06) ...
Removing linux-image-4.4.34-bananian (16.04.06) ...
Removing Bananian kernel 4.4.34-bananian (uImage-next)...
Note: selecting "linux-headers-sun8i" instead of the
      virtual package "linux-headers-3.4.113-sun8i"
"linux-headers-3.4.112-sun8i" exists in the package database, but it is not a
real package and no package provides it.
The following packages will be REMOVED:
  linux-headers-4.4.34-bananian
0 packages upgraded, 0 newly installed, 1 to remove and 2 not upgraded.
Need to get 0 B of archives. After unpacking 53.2 MB will be freed.
(Reading database ... 49028 files and directories currently installed.)
Removing linux-headers-4.4.34-bananian (16.04.06) ...
dpkg: warning: while removing linux-headers-4.4.34-bananian, directory '/lib/modules' not empty so not removed
The following packages will be REMOVED:
  linux-firmware-image linux-firmware-image-4.3.3-bananian
  linux-firmware-image-4.4.34-bananian
0 packages upgraded, 0 newly installed, 3 to remove and 2 not upgraded.
Need to get 0 B of archives. After unpacking 1,667 kB will be freed.
(Reading database ... 34032 files and directories currently installed.)
Removing linux-firmware-image (16.04.04) ...
Removing linux-firmware-image-4.3.3-bananian (15.08.04) ...
Removing linux-firmware-image-4.4.34-bananian (16.04.06) ...
No packages will be installed, upgraded, or removed.
0 packages upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Need to get 0 B of archives. After unpacking 0 B will be used.

```

ravermeister, 2017-04-27:
Hi,  Glad to know that a migration is planned, especially because over the time I did several customizations
(new custom daemons etc) one question, what will happen to custom compiled Kernel Modules?
especially this one: @http://forum.lemaker.org/forum.php?mod=viewthread&tid=23612&extra=page%3D2%26filter%3Dtypeid%26typeid%3D3
I'm using this module to encrypt my Home directories

inaciose, 2017-05-02:
Hi. Are you working in a script to do the migration automatically?

ThomasKaiser, 2017-05-02:
> Are you working in a script to do the migration automatically?

I don't know with whom you're talking but at least I'm not working on anything like that. The intention behind this thread here is to identify pitfalls and necessary steps that are needed to make a migration as flawless as possible so @nisenbeck could even provide it as [bananian-update](https://github.com/armbian/build/issues/648#issuecomment-291118070) variant.

So let's hope that some experienced Bananian users collect their experiences while migrating so that a collective work approach can start that will help inexperienced users later. But as already said there's no need to rush since Bananian still receives security updates for a longer time.

ravermeister, 2017-06-01:
@igorpecovnik the link to the upgrade.sh is dead. Does it still exist somewhere? are there any plans to get it up to date? if I can I'm willing to help, but my Shell Scripting knowledge is not the best :)

igorpecovnik, 2017-06-01:
We removed script because it's deprecated and make more damage than benefits, while we were unable to provide working alternative. I am afraid deep understanding of Armbian is needed, not just scripting power ... If you are willing to help, start with easy tasks.

PeterRies, 2017-06-14:
just a question I don't know where to ask else...

I'm running Bananian. What if I ...
* remove the /etc/apt/sources.list.d/bananian.list file
* change /etc/apt/sources.list - alle jessie to stretch
* run 'apt-get upgrade' and 'apt-get dist-upgrade' - (full procedure as of https://linuxconfig.org/how-to-upgrade-debian-8-jessie-to-debian-9-stretch)

Would that work, too?

igorpecovnik, 2017-06-14:
Such upgrades are always critical - do a backup, prepare for the worst and try. Nobody can tell you this will work without a problem.

nisenbeck, 2017-06-14:
Pay attention that you won't receive kernel and u-boot updates after removing the Bananian sources list.

I did not try to update a Bananian installation to Stretch so far, but why not do a backup/image and just try?

ndandanov, 2017-06-28:
@PeterRies, did you try to upgrade from Jessie to Stretch and if yes - what was the outcome?
Thanks!

igorpecovnik, 2017-06-28:
I did one upgrade, by the way, on a clean Armbian Debian Jessie and it went almost well except network-manager had some issues. I didn't investigate whether this is our or general problem.

AntonioTrindade, 2017-07-20:
I bought a BPI-R1 recently and am trying to build a home router with it. I tried using Armbian because Bananian was being discontinued, but could not, after many tries to get hostapd working with it. the BPI-R1 requires a "special" hostapd with rtl871xdrv driver support and I was not able to build it after many tries and GIThub repositories. I then installed Bananian and to my amazement, it all worked at first try.

Having said this, is there any option to bring rtl871xdrv support to hostapd in Armbian without having to rebuild the 8192cu module and hostapd?

ThomasKaiser, 2017-07-20:
> BPI-R1

is a fail and [unsupported by Armbian as well](https://github.com/armbian/build/issues/511#issuecomment-316667464). The WiFi on this board is just a joke and you'll only waste your time since you can't [fix crappy hardware with software](https://www.google.de/search?q=r1+wifi+site:armbian.com). Also this is a Github issue covering possible migration strategies from Bananian to Armbian and not a support forum. So please head over to the forums if you still want to waste your time with this device.

igorpecovnik, 2017-07-20:
BTW. We provide both and latest stable hostapd versions since "version 1.0" since there are many Realtek chips out there who need this approach. apt-get install hostapd-realtek

pietrondo, 2017-08-13:
Hi! I have bananian installed on my banana pi.
Now! How I can migrate to armian? thx

JackTheDipper, 2017-09-27:
Sooo... is there a clean upgrade path now from bananian on Banana Pi 1 (and 3.x kernel)? That would be awesome.

Phyllo, 2017-10-06:
Seems either nobody reads this thread any more or there is no smooth way for migrating from bananian to armian.
Could someone of those more experienced with armian please comment?

igorpecovnik, 2017-10-07:
My recommendation would be any way to start with a clean image and transfer your services on a freshly installed system. Differences are growing every day and not everything is packed. I manage to migrate this way - from clean latest image: 
```rm /etc/apt/sources.list.d/bananian.list
rm /usr/sbin/bananian-*
echo "deb http://apt.armbian.com jessie main utils jessie-desktop" >> /etc/apt/sources.list.d/armbian.list
wget -qO - http://apt.armbian.com/armbian.key | apt-key add -
apt update
chsh -s /bin/bash root
chage -d 0 root
aptitude remove ~nbananian
apt install --no-install-recommends autoconf autotools-dev binutils bzip2 cpp cpp-4.9 dh-python dpkg-dev g++ g++-4.9 gcc gcc-4.9 klibc-utils libasan1 libatomic1 libc-dev-bin libc6 libc6-dev libcaca0 libcloog-isl4 libcurl3 libdpkg-perl libexpat1 libgcc-4.9-dev libgomp1 libisl10 libklibc libmpc3 libmpdec2 libmpfr4 libpython-stdlib libpython2.7-minimal libpython2.7-stdlib libpython3-stdlib libpython3.4-minimal libpython3.4-stdlib libsigsegv2 libssl1.0.0 libstdc++-4.9-dev libsysfs2 libtcl8.6 libtimedate-perl libubsan0 libuuid-perl linux-libc-dev lsb-release m4 make mime-support openssh-client openssh-sftp-server patch perl perl-base perl-modules python python-apt-common python-minimal python2.7 python2.7-minimal python3 python3-minimal python3.4 python3.4-minimal tcl-expect toilet-fonts ucf zlib1g-dev linux-base sunxi-tools figlet
mount /dev/mmcblk0p1 /boot/
rm -r /boot/*
apt install linux-jessie-root-bananapi
apt install linux-image-sun7i linux-u-boot-bananapi-default
wget -O /boot/boot.cmd https://raw.githubusercontent.com/armbian/build/master/config/bootscripts/boot-sunxi.cmd
wget -O /boot/armbianEnv.txt https://raw.githubusercontent.com/armbian/build/master/config/bootenv/sunxi-default.txt
mkimage -C none -A arm -T script -d /boot/boot.cmd /boot/boot.scr

ROOT=$(blkid | grep mmcblk0p2 | awk '{print $3}' | tr -d '"')
echo "rootdev=$ROOT" >> /boot/armbianEnv.txt

echo "$ROOT / ext4 defaults,noatime,nodiratime,commit=600,errors=remount-ro 0 1" >> /etc/fstab
echo "/dev/mmcblk0p1 /boot vfat defaults 0 1" >> /etc/fstab
reboot
```

ThomasKaiser, 2017-10-07:
> from clean latest image

This should be highlithed a little bit more since at least I had to deal with a few Bananian users recently that for whatever stupid reasons followed the german [never change a running system](https://www.faq-o-matic.net/2008/02/20/never-change-a-running-system-bullshit/) BS approach lazily refusing to update their installations.

Igor's migration guide above relies on a patched up Bananian installation already using Jessie and Nico's latest 4.4 kernel. If you're sitting on Wheezy with a smelly 3.4.x kernel do your homework first please and upgrade. But in the meantime I also second Igor's suggestion to better start from scratch and migrate settings carefully instead of trying to transform a Bananian install into Armbian. 

The time/efforts you would have to invest in numerous backups when trying to migrate (please do not expect that everything works as expected especially if you tweaked your Bananian installation already) is better invested in buying a new and **good** SD card ([great idea anyway](https://forum.armbian.com/index.php?/topic/954-sd-card-performance/)) and then migrating your settings from the SD card still containing your intact Bananian installation while at the same time **documenting** settings/tweaks you've done over the years.

ThomasKaiser, 2017-10-07:
> is there a clean upgrade path now from bananian on Banana Pi 1 (and 3.x kernel)? That would be awesome.

Quite the opposite, that would be insane. It seems you did not read or maybe even ignored [what has been written above](https://github.com/armbian/build/issues/648#issuecomment-291112355)?

If you're still running with a 3.x kernel and/or Wheezy do your homework now and upgrade in any case first. A lot of Bananian users that will do that will then realize that 'something went wrong', most probably your SD card simply died since... they do this after being used for some time, especially if you happened to buy an average one that can't cope with random IO that good (that applies to almost all of them especially noname cards or crap brands like Intenso, Kingston, PNY and so on).

Upgrades and/or huge package installations mean an awful lot of random writes so it's always advisable to backup **now** your SD card before you think about especially if you're the kind of guy thinking 'Huh? It worked always flawlessly. Why care?'.

If Armbian would provide a migration tool all those occurences of SD cards dying since they do this after a certain amount of writes will be added as 'Armbian is unreliable crap and killed my SD card'. Seriously, if you run Bananian off the same SD card now for years the best you could do is to 'invest' in a [**new and good** SD card](https://forum.armbian.com/index.php?/topic/954-sd-card-performance/) now. If you moved the rootfs already to a connected HDD then this is not much of a concern but then I fear migration might be even more complex (and Igor's guide above does **not** apply).

ThomasKaiser, 2017-10-07:
Just for the record: https://github.com/armbian/documentation/blob/454f24ff43e37ed653859e05ae85881e5d1db021/docs/User-Guide_Migration_from_Bananian.md

Now busy writing the document explaining the 3 main reasons why a migration is no good idea, what's to be considered if users still want to and a few other things that need some attention. Discussion about the document may happen here later :)

Phyllo, 2017-10-07:
Thanks for the fast answers.
Luckily, I'm still new to it all since I just got my first Banana Pi this week and set it up with Bananian. Then I played around with it and now I wanted to update and found that Bananian will soon reach the end of support and that its developers recommend Armbian instead. That's how I ended up here and I'll setup Armbian on a second SD card and then setup again what I did during this week. Not much work lost and a good practice for getting used to the whole system.

ThomasKaiser, 2017-10-07:
Too much text, right? https://github.com/armbian/documentation/blob/master/docs/User-Guide_Migration_from_Bananian.md

zador-blood-stained, 2017-10-07:
> Too much text, right?

Personally not a fan of talking about upgrading hardware when you are asked about upgrading software.

Explaining differences between kernel branches and upstream Ubuntu/Debian distributions, build time tweaks that can't be easily replicated, etc. IMO is better to summarize why in-place upgrade is not a good idea and it's better to either not upgrade at all or start from a clean Armbian image.

ThomasKaiser, 2017-10-07:
> Personally not a fan of talking about upgrading hardware when you are asked about upgrading software.

Well, I tried to address user expectations and stuff the same users usually miss (eg. that toy grade hardware like a Banana Pi and SD cards are not to be expected to last forever or just 10 years but usually die way earlier). Then there's a refreshing tendency to deal with technical problems in a way that the last change that has been done intentionally (eg. switching the distro) will be used as an explanation for what will happen next (hardware dying since old --> 'Armbian killed my SD card', 'Armbian killed my Banana Pi').

We have a situation now that the majority of Bananian users are happily running their boards for +24 months on average, most of them with the same SD card. Some cards already failed, some boards already failed. Within the next 12 months and with every kind of upgrade activity an awful lot more cards and boards will fail. Who's to blame in a situation the last thing the affected users did prior to their hardware dying was switching to Armbian?

If it's important to you to spread such 'Armbian killed the hardware that worked flawlessly with Bananian all the time before!!11!!1!' news and spend more time on unnecessary support efforts in the forum, then feel free to ignore the involved hardware issues. I went through this bizarre discussion already two times within the last months and don't want to waste any more time on this.

I even tried out a 'true Debian 9' installation as a result of such a personal support event just to realize that upstream Debian maintainers still do not care about the stuff we do in `/etc/init.d/armhwinfo` or @nisenbeck does in `/etc/rc.local` -- performance sucks, people are happy, I feel I should start to ignore all the ignorance...

zador-blood-stained, 2017-10-07:
I have nothing against SD card related info, but I was talking about 2 paragraphs specifically - "Hardware up to the task?" and "Replacing the hardware". 
Somebody already bought the board and sometimes people won't be happy hearing that they made a mistake and now need to spend more time and money. IMO it's easier to say "not supported by Armbian, ask vendor for software support" (and also let people discover by themselves how "great" support can be provided by the vendor) than say "buy a different (and often more expensive) board"

And we already have reviews, benchmarks and comparisons for A20 SATA vs everything else, so IMO ~3 large paragraphs can be replaced by ~3 small links.

Phyllo, 2017-10-09:
Nope - I just thought he summed it up with "explaining the 3 main reasons why a migration is no good idea". Why should I not trust him that he's right about that and follow his suggestion to not do it?

ThomasKaiser, 2017-10-13:
> IMO it's easier to say "not supported by Armbian, ask vendor for software support"

Which 'vendor support'? Those people no one can help ''stil''l buying those more recent Banana boards aren't those who want to migrate from Bananian since Bananian supported only A20 devices anyway (and M2/A31s somewhat -- no idea whether there are really BPi M2 Bananian users existent).

And those who want to upgrade now from Bananian (especially those guys irresponsibly following the stupid [never change a running system](https://www.faq-o-matic.net/2008/02/20/never-change-a-running-system-bullshit/) BS should be aware that it's almost 2018, that the hardware they used without problems the last months/years will fail eventually anyway maybe even sooner than later (since toy grade and not server grade hardware) and that the whole situation is a great opportunity to review some decisions: reliability, performance, whatever.

We all fooled ourselves years ago when hearing about 'A20 supports native SATA' expecting normal SATA performance and a BananaPi being a x86 box replacement. What about admitting this mistake... now?

closed 2017-10-13 by ThomasKaiser

ravermeister, 2017-10-22:
...and what are with those people who have used the hardware WITHOUT any issues (not using wlan at all)
have backed up their system, never had problems with their wise chosen SD-Card, and still want to migrate to another system? (I for myself think up to now is the best thing: backup /etc the package list, all stuff you want save e.g. home directory, and start with a fresh install) because up to now nothing really helpful was posted here. I'm willing to help (with shell scripts Java if needed, but looks not really promising here. Sad :/

pedroetb, 2017-10-22:
I'm on that group without issues yet, but now I'm worried about a unexpected failure.
To minimize risks, I'm running [this](https://github.com/pedroetb/rsync-incremental-backup/blob/master/rsync-incremental-backup-system) once a month, so in case of failure I can recover and apply most of configs and scripts to a new platform.

zador-blood-stained, 2017-10-22:
> I for myself think up to now is the best thing: backup /etc the package list, all stuff you want save e.g. home directory, and start with a fresh install) because up to now nothing really helpful was posted here.

I don't understand what kind of "really helpful" things do you want to hear. I believe the most helpful advice was already posted - don't try to do an in-place upgrade unless you understand what you are doing exactly, start from a fresh image and transfer data and settings for applications that you use. It is almost impossible to migrate absolutely everything for everybody in a scripted way - that may include applications and libraries installed without using the package manager, 3rd party and locally built packages, system tweaks (especially done outside of /etc).

In addition to that - /etc contents, package list and "all stuff you want to save" cannot be easily migrated between distributions (i.e. Debian and Ubuntu) and different releases (i.e. Jessie and Stretch).
If you transfer all /etc contents in this scenario (between different releases or distributions) you are likely to break the system either right away or during the next package upgrade, and if you want to add white and/or black lists you'll have to consider hundreds or more packages that may need special treatment.

kecsap, 2017-11-02:
@ghost and all others: I upgraded a headless (no X or other fancy UI) Bananian Linux to Stretch without any problems. I think it is a good alternative to keep the Bananian kernel and update the userspace to the latest packages. I will run my Banana PI until it will die, I don't need any fixes which the new kernel would contain.

wepanx, 2018-01-04:
@kecsap: on a 3.x or 4.x kernel and just by changing debian's sources.list entries? (I'm still on 3.x as bananian warned users to upgrade to the 4.x kernel those days. But hey, it was also said that there will be a bananian->armbian upgrade path ^^)

kecsap, 2018-01-04:
@wepanx Yes, I kept the old kernel (3.4.113-bananian) and the user-space is upgraded to stretch by sources.list. I have no issues so far.

noodle-, 2018-01-05:
@kecsap @wepanx I upgraded the same way without any big problems. 
The only thing I had to do to get everything running again was to create new SSL(?) keys for some web services like syncthing/sabnzbd by simply deleting or renaming the current `server.key` file which can be found in the respective services' root folders.

rugk, 2018-02-17:
> I don't need any fixes which the new kernel would contain.

That's a strong thing to say, considering it also includes security fixes. AFAIK the [dirtycow](https://en.wikipedia.org/wiki/Dirty_COW) vulnerability has been fixed in Bananian, but what if there is a new such vulnerability… 

So at least the switch to the mainline Debian kernel or so would be an idea. Don't know if that's possible, but AFAIK Debian suports Banana Pis out of the box.

igorpecovnik, 2018-02-17:
Upstream Debian (www.debian.org) support for those boards is very basic and install is still on "for geeks only" level. Armbian goes deeper/wider for single board computers than generic Debian will ever go.

rugk, 2018-08-11:
BTW [on Armbian's website](https://www.armbian.com/download/?tx_maker=lemaker) only Banana Pi Pro is listed as supported, none of the other Banana Pi versions.

nisenbeck, 2018-08-11:
See https://www.armbian.com/download/?tx_maker=sinovoip as they are from Sinovoip.

igorpecovnik, 2018-08-16:
> Hello! Is there any support for the official Banana-Pi LVDS displays?

By using https://www.armbian.com/search you can find this:

https://forum.armbian.com/topic/7233-bananapi-pro-7-touchlcd-with-debian-stretch-and-mainline-kernel-4xxxx/

frauhottelmann, 2018-09-24:
Thank you for the quick answert. Does armbian-config do that automatically? To which u-boot do I have to switch? Is there a u-boot next? I can't search right now, because for the moment I am back on bananian :smiley:

igorpecovnik, 2018-09-24:
It's complicated. :) armbian-config way of switching kernel works on armbian ... not really sure if it will work with Bananian. Since I lost track on this upgrade I can't help with details ... you need those packages:

```
apt install linux-image-next-sunxi
apt install linux-dtb-next-sunxi
apt install linux-jessie-root-next-bananapi
apt install linux-u-boot-bananapi-next
```
Since not everything is packed and since we can't help much its recommended to start with a clean image. On most other SBC operating systems this is mandatory. They are one time made images which receive no update beside stock debian packages.

frauhottelmann, 2019-01-11:
Do yourself a favor and just backup your data and restore it with a clean armbian image on a new SD. I tried a lot of things. Something always breaks 🤷

wehkah, 2019-05-02:
I am very interested in this topic, because I am running a Banana Pi (Model 1!), and because Bananian is no longer maintained, I can't install newer version of PHP, which is required for the next version of NextCloud. Bummer!

Here is my thought of how a script for the migration from Bananian to Ambian might work:

- plug in a USB memory drive (of course, that has do be done by the user, a script can't do this. ;)
- install Ambian to that drive
- copy /etc from Bananian to the new drive and make adjustments where necessary 
- move /boot on the sdcard to /boot.banian
- install boot files for Ambian to /boot on sdcard and modify them to boot from USB drive

If that works without crashing, you can copy the system from the USB drive back to the sdcard, or the appropriate hard drive, that Bananian was installed on, re-modify the boot files and off you go with Ambian!

If the new system does not boot, the user can easily take the sdcard to another machine and rename /boot.banania back to /boot, and boot the old Bananian again.

What do you think about that idea? Possible? Or complete rubbish?