armbian/build #649: can interface not working on lime2-emmc

Timple, 2017-04-03:
Good day,

We seem to have lost the can interface on new builds for our lime2 boards. I tracked the removal of the patch to this commit: https://github.com/igorpecovnik/lib/commit/c7bbc3257e8dcbf75398301602150fe9f1666c86

Could there have been a error in restructuring the patches?

zador-blood-stained, 2017-04-03:
CAN activation (together with other Device Tree patches) was moved to the Device Tree overlays.
DT overlays documentation: https://docs.armbian.com/User-Guide_Allwinner_overlays/

Please note that it requires the newest version of the boot script and adding an extra line (`overlay_prefix=sun7i-a20`) to `/boot/atmbianEnv.txt` in case you only updated the kernel packages; ful image builds should include all necessary changes.

Timple, 2017-04-03:
I started off with a clean slate. So from an empty folder, cloning lib etc.
`overlay_prefix=sun7i-a20` is present in `/boot/atmbianEnv.txt`

Any files I can check it the patch worked out?

zador-blood-stained, 2017-04-03:
The main difference is that CAN is no longer activated by default on the new images. To activate you need to add
```
overlays=can
```
line to `/boot/armbianEnv.txt` and reboot. Then you may check `dmesg` and `lsmod` to see if `sun4i-can` module was loaded (you still may need to load other CAN modules)

Timple, 2017-04-03:
Your link to the overlays put me in the correct direction. I had to add `overlay=can` to `/boot/atmbianEnv.txt` and do a reboot.

I expected a bug because the files containing the can0 interface had changed. But putting them in such an overlay makes sense!

Can be closed

closed 2017-04-03 by zador-blood-stained