armbian/build #650: Enhancing `resize2fs` behaviour in fixed image size situations?

ThomasKaiser, 2017-04-05:
I'm currently thinking about enhancing `resize2fs` behaviour for two reasons:

- keeping the amount of ugly hacks needed to build OMV installers as low as possible
- providing more convenience in situations where `$FIXED_IMAGE_SIZE` is defined

I would propose defining new behaviour as follows:

If `rootfs` is either `btrfs` or `f2fs` and if unpartitioned space is at least 1 GiB then create a new partition using all remaining free space but following our rules (leaving 2% unpartitioned on 8GB, 1% at 16GB or above to ease 'device backups')

If `/root/.rootfs_resize` is detected then resize `rootfs` according to defined rule in `.rootfs_resize` and then create a 2nd behind if above rule applies.

Any thoughts/objections?

If no one is against let's move to 2nd request for discussion:

I would then also propose to change `build-all.sh` behaviour after implementing the above change. All nightlies using `dev` branch should then be created with `FIXED_IMAGE_SIZE=7800` and `ROOTFS_TYPE=btrfs`. Also `debootstrap-ng.sh` should be enabled to detect the 'nightly' case and then add also `compress=lzo` to `/etc/fstab` in this case. Maybe we get then some useful feedback from nightly users the first time (lack of negative feedback considered also/especially valuable)

zador-blood-stained, 2017-04-05:
> If rootfs is either btrfs or f2fs and if unpartitioned space is at least 1 GiB then create a new partition using all remaining free space but following our rules (leaving 2% unpartitioned on 8GB, 1% at 16GB or above to ease 'device backups')

I don't know if f2fs is useful at all in the build system, but for the btrfs we can do some experiments and enhancements: implement FS resize if possible, add LZO compression and test it in different scenarios (slow/fast storage, server/desktop subjective tests)

Main concern in this scenario - when not targeting a single purpose systems (i.e. OMV) we should ensure that we have enough space on the system partition - for the possible upgrades, all kinds of caches, installing extra software, etc. So we would need to resize the system partition first and create the 2nd/3rd partition after that.

> I would then also propose to change build-all.sh behaviour after implementing the above change. All nightlies using dev branch should then be created with FIXED_IMAGE_SIZE=7800 and ROOTFS_TYPE=btrfs.

Except for desktop ones where we could still test btrfs but without the 3rd partition (again, if online resizing works for btrfs).

And not sure about using `FIXED_IMAGE_SIZE` - we still can create a minimal image just with a more complex resizing procedure, and since we would target mainline or at least 4.4 LTS kernels, we shouldn't need a reboot to finish the resize procedure.

zador-blood-stained, 2017-04-06:
After thinking some more about it - what is the main purpose of these changes, or rather which scenarios are targeted?

Currently it's about scenarios when rootfs is on the large and fast storage which can be used i.e. as a NAS storage in OMV:
- Allwinner boards booting from SPI flash to USB stick/HDD/SSD
- Clearfog booting directly from SATA, SPI+USB or SPI+SATA
- Using large enough and fast enough SD or eMMC on any board.

In other cases we are not talking about first boot changes, so it's a job of nand-sata-install to create multiple partitions and alter the fstab parameters.

And second - what filesystem should be used for the second/third partition by default and how the FS type on this partition should be controlled?

ThomasKaiser, 2017-04-06:
To be honest: I never thought about resizing `btrfs` due to `btrfs-tools` being horribly outdated in the distros we use. But that was Wheezy (insanely outdated version 0.19), currently we're talking about 3.17-1.1 in Jessie and not so surprisingly 4.4-1 in Xenial (maybe 4.8 on recent x64 Xenial). 

Versions are more recent now but after going through some serious hassles years ago due to mismatch between kernel and userland features (though that was `ext4` back then and the ability to create fs larger than 16TB while userspace utilities simply failed on Debian since plagued with 32-bit limitations) I fear that's just too much efforts or we're not even able to test through all worst case scenarios (imagine kernel update to 4.13 with some nice new features and a freshly introduced incompatibility with btrfs-tools 3.19 or earlier).

Your comments led to me thinking again about this stuff.

But besides of that the use cases I thought of are simply these: I'm pretty happy with the rootfs running off a SD card and even prefer this over situations where only a HDD is connected (reason: I want the HDD to spin down when not in use). And given `btrfs` is not resizable (which is what I was thinking until yesterday for the reasons above) I already do it this way: limit rootfs to 4GB and partition the unused space later (eg when mounting /var/www there with compress-force=zlib while using compress=lzo on the rootfs).

I wouldn't put a filesystem on the partition automatically since from then on it's easy. A simple `mkfs.whatever` call is much easier than partitioning and in the OMV case it's just clicking around in the web UI then.

ThomasKaiser, 2017-04-16:
Just for the record: With https://github.com/igorpecovnik/lib/commit/af560222c7f08034a8dbe218249a989aab478e46 I added the 2nd or 3rd partition creation in the `/root/.rootfs_resize` case. So if anybody has the time to play around with btrfs resizing and calculating `$imagesize` and `$sdsize` in `debootstrap-ng.sh` and this works reliably I would follow your suggestion now and then simply lift the [limitation to ext4 only we currently use](https://github.com/igorpecovnik/lib/blob/af560222c7f08034a8dbe218249a989aab478e46/scripts/resize2fs#L133).

But at least I won't spend time on this the next months since for my use cases a separate partition (with different btrfs compression options) is better anyway.

closed 2018-04-18 by ThomasKaiser