armbian/build #651: How to check sunxi_ve_mem_reserve size

topilski, 2017-04-08:
Hi, i installed server image 3.4.113, and i need to know how much reserved memory for the video acceleration? according your boot script https://github.com/igorpecovnik/lib/blob/491fc18ae0cb5680e98e0e23ee1ed762e148d00b/config/bootscripts/boot-sunxi.cmd#L40 you set sunxi_ve_mem_reserve to 0, how can i update and apply new value when booting OS? Also according this: http://linux-sunxi.org/Kernel_arguments i see next:

> sunxi_ve_mem_reserve=0 -- This eliminates the reserved memory for the video acceleration engine, saving 80MB. You can use this if you don't run accelerated video with programs such as VLC, Kodi or libvdpau-sunxi. (This kernel parameter is ignored in recent 3.4 kernels, if CMA is enabled in kernel configuration. If CONFIG_CMA=y, ve_size is hardcoded to 80MB in sunxi_cedar kernel module.)  

and according your script https://github.com/igorpecovnik/lib/blob/5fc53d7e9b2ad643e1a698f0a01b7d626fdfdd40/config/kernel/linux-sun8i-default.config#L1214 CONFIG_CMA=y, and ve_size is hardcoded to 80MB. I need to know how many memory allocated for video acceleration? Without video acceleration this device can't play video properly, because CPU can't decode video frames faster.

igorpecovnik, 2017-04-08:
Have you try our desktop image? Video acceleration works out of the box.

topilski, 2017-04-08:
Desktop not interesting for me, i want rendering on xserver only.

topilski, 2017-04-08:
If you can help please answer, how can i update and apply new value when booting OS?

igorpecovnik, 2017-04-08:
Ahaa. You can pass kernel parameters in boot scripts. Check this, it's self explanatory:
https://github.com/igorpecovnik/lib/blob/master/config/bootscripts/boot-sunxi.cmd#L40 
So you can add disp_mem_reserves=on to /boot/armbianEnv.txt ... it looks like we have wrong logic here ... 

Other just edit boot.cmd, save and convert to boot.scr when done and reboot.

igorpecovnik, 2017-04-08:
When not set to zero, there are some defaults, which are sufficient for video playing. If you want to experiments, just set those numbers as boot parameter ...

topilski, 2017-04-09:
Hi, thank you for your help, i tried to use your desktop version, why you think that on it have video acceleration? I saw usage CPU more than 100%, i think you didn't have also GL optimization on your desktop versions.

topilski, 2017-04-09:
Also i installed dev version for orange pi one, and how i can see it doesn't contain mali modules, without mali talk about GPU not make sense, and i can't find /dev/cedar_dev .

zador-blood-stained, 2017-04-09:
> Hi, thank you for your help, i tried to use your desktop version, why you think that on it have video acceleration?

Because it has **limited** video acceleration support activated in the default player (mpv)

> i think you didn't have also GL optimization on your desktop versions.

And why do you think that there should be any GL optimizations? Mali with the available driver can provide only OpenGL ES acceleration and with the available driver version it can't be used by the web browsers and similar software.

> Also i installed dev version for orange pi one, and how i can see it doesn't contain mali modules, without mali talk about GPU not make sense, and i can't find /dev/cedar_dev .

I believe "Known issues" tab on every H3 board download page clearly states what should be expected from the dev kernel (or rather what functionality can't be expected there)

topilski, 2017-04-09:
Hi, please show me code in build scripts where you patched egl libraries? Where you build mali user space library from sources? https://github.com/linux-sunxi/sunxi-mali.git Where tests, how you test GPU decoding?

topilski, 2017-04-09:
All what i see, slideshow in video player, this is horrible OS for end users.

zador-blood-stained, 2017-04-09:
> Hi, please show me code in build scripts where you patched egl libraries? Where you build mali user space library from sources? https://github.com/linux-sunxi/sunxi-mali.git

`libMali.so` userspace driver comes from the repository you linked "as is". There is no source code for it (at least without an expensive contract and a restrictive NDA and available only to SoC vendors)

> Where tests, how you test GPU decoding?

GPU decoding of what? Mali is purely 3D rendering engine, HW video decoding comes from another block in the SoC.

> All what i see, slideshow in video player, this is horrible OS for end users.

Feel free to contribute to libvdpau-sunxi or sunxi-cedrus projects if you are unhappy with the current status: https://linux-sunxi.org/Cedrus

The issue in the "horrible OS for end users" bug tracker then can be closed.

topilski, 2017-04-09:
Hi @zador-blood-stained , why you close my issue? May be you answer on my question (How to check sunxi_ve_mem_reserve size)? I answered about GL because if you use mesa-egl like in server os, you need to build libmali from sources, in other case rendering will be executed on CPU side. I know that decoding happend via VDPAU i also created issue for that some days ago https://github.com/linux-sunxi/libvdpau-sunxi/issues/55 . If you build some OS image you should have properly support not only build and that all. If community ask something, please find time to answer, if i create issue i will close it. If your OS freezed when video stream played may be need to investigate this? this is not my fault that armbian freezed in good devices.

zador-blood-stained, 2017-04-09:
> May be you answer on my question (How to check sunxi_ve_mem_reserve size)?

It's even stated on the wiki page (https://linux-sunxi.org/Cedrus/libvdpau-sunxi):
NOTE: This kernel parameter is ignored in recent linux-sunxi 3.4 kernels, if CMA is enabled in kernel configuration. If CONFIG_CMA=y, ve_size is hardcoded to 80MB in sunxi_cedar kernel module.
For A33 or H3, if you use the kernel from official SDK, you should modify the memory reservation configuration of ion memory allocator. 32 MB seems to be enough for 720p H.264 decoding, and 64MB is enough for 1080p H.264 decoding.

> I answered about GL because if you use mesa-egl like in server os, you need to build libmali from sources, in other case rendering will be executed on CPU side.

Maybe you should read the wiki before making such assumptions: https://linux-sunxi.org/Mali_binary_driver

> If you build some OS image you should have properly support not only build and that all.

Free open source projects are usually provided "as is" and all support is provided volunteerly, so "should" and" proper" are not the correct words in this situation.

topilski, 2017-04-09:
I write all this in start of conversation, so if 

> You can use this if you don't run accelerated video with programs such as VLC, Kodi or libvdpau-sunxi.

According your words: 

> Because it has limited video acceleration support activated in the default player (mpv)

How in desktop image mpv can use "video acceleration", if we have hardcoded 80 MB VPU? I read all wiki which can find, and created special build script for orange pi: https://github.com/fastogt/fastotv/blob/master/build/build_env.py#L151 because of this i asked are you sore that used in desktop image patched version of GL library? The same question are you sure that vdpau decoding happens in mpv on GPU side? In mpv you also need to specify hwaccels args. About opensourse, we don't who work more with opensourse more you or i. Please respect me and other communicates which use your solution.

zador-blood-stained, 2017-04-09:
> How in desktop image mpv can use "video acceleration", if we have hardcoded 80 MB VPU?

Because

> This kernel parameter is ignored in recent 3.4 kernels,

and sun8i-default is exactly the case of a "recent 3.4" kernel. Memory reservation for cedar kernel module is done by ION (as stated on the Wiki)

From the kernel config:
```
CONFIG_ION_SUNXI_RESERVE_LIST="160M@0,256M@0,130M@1,200M@1"
```
So it reserves 160M with the provided kernel config.

> https://github.com/fastogt/fastotv/blob/master/build/build_env.py#L151 because of this i asked are you sure that used in desktop image patched version of GL library? The same question are you sure that vdpau decoding happens in mpv on GPU side, or CPU do all work?

I already answered about "patched version of GL library" and relation of MPV, GPU, vdpau and video decoding.

topilski, 2017-04-09:
@zador-blood-stained thank you for your answers.

closed 2017-04-09 by topilski

roderick1980, 2018-09-07:
Hello Igor,

please, may I know where to download the mentioned Desktop image that work out of the box for HW accel (Orange Pi Zero)?

Thanks and kind regards

EvilOlaf, 2018-09-07:
https://www.armbian.com/orange-pi-zero/
Scroll to the very bottom of the page to get these images.

igorpecovnik, 2018-09-07:
None of those are with a desktop and since Opi zero does not have HDMI connector its function is disabled by default. The closest hack would be Orangepi One with some adjustments to the FEX file.

roderick1980, 2018-09-07:
Thanks Igor

1f7, 2018-11-12:
@igorpecovnik  > Opi zero does not have HDMI connector its function is disabled  

I installed vnc and all works fine except WebGl on chromium.
How to enable HW acceleration and make it work with vnc on 3.4.113?

igorpecovnik, 2018-11-12:
> I installed vnc and all works fine except WebGl on chromium.
> How to enable HW acceleration and make it work with vnc on 3.4.113?

Via VNC? Not sure you can do that. Try asking here: https://forum.armbian.com/forum/25-peer-to-peer-technical-support/