armbian/build #652: Add note about additional needed packages.

cnlohr, 2017-04-09:
On my system, I needed to install ccache and device-tree-compile in order to build.  It gave me  cryptic errors in the failure due to the fancy stuff going on with the pretty build.  Consider adding this to the build script to automatically install or associated documentation?

zador-blood-stained, 2017-04-09:
device-tree-compiler and ccache should be automatically installed by default: https://github.com/igorpecovnik/lib/blob/master/general.sh#L480

What kind of setup did you use (OS release and any virtualization/containerization)?

cnlohr, 2017-04-09:
I'm on my host OS. 

```
Linux cnlohr-desktop 4.4.0-53-generic #74-Ubuntu SMP Fri Dec 2 15:59:10 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```

I have no intent for virtualization or containerization.

Do I have to call ```general.sh``` or should it be called by ```./compile.sh```?

zador-blood-stained, 2017-04-09:
From your other issue it looks like you are using "Serena" release, which doesn't report itself as Xenial even though it is based on Xenial. Still I have no idea why those packages were missing - they and all other ones should be installed automatically, you don't need to use any other commands, and since you installed missing packages there is nothing to check (though you may try deleting them and check if they are reinstalled automatically by the build sctipts).

cnlohr, 2017-04-10:
I suppose we should just close this?  I understand not wanting to support Mint as a specific environment instead of the base for Ubuntu despite its popularity?

zador-blood-stained, 2017-04-10:
> I suppose we should just close this? I understand not wanting to support Mint as a specific environment instead of the base for Ubuntu despite its popularity?

It's more about reducing efforts to support multiple environments - different releases have different default compiler versions, may not have some additional tools packageed, etc.

Previously we supported Trusty and Xenial with specific quirks for Xenial, but I suppose we can reverse this logic to differentiate between Trusty (for legacy/backwards compatibility) and everything else that will include Xenial, newer ubuntu releases (Yakkey and Zesty) and Ubuntu derivatives (including Mint/Serena). I hope this will improve compatibility with different OS releases.

cnlohr, 2017-04-10:
I suppose it can be a matter of ease.  If it's difficult for you guys, I will absolutely switch to a VM environment or the like considering I only need to work for one specific, hopefully very small change.   I guess I was just confused there would be mutually exclusive changes between versions of Linux that far down in the toolchain.

cnlohr, 2017-04-11:
Edited first post to include the additional package I found was missing (which will fail silently if it doesn't exist)

closed 2017-04-13 by zador-blood-stained