armbian/build #653: ERROR in function compile_kernel [ common.sh:249 ], Kernel was not built.

cnlohr, 2017-04-09:
I wish I could make the title of this error more specific but I can't seem to figure out why it's not building.

I do the normal process
```
git clone --depth 1 https://github.com/igorpecovnik/lib
cp lib/compile.sh .
```

Then, I compile with:
```
./compile.sh BOARD=orangepione BRANCH=dev
```

I specify that I only want kernel and uboot (as I am trying to get a custom resolution).

It goes about its business and reports 

```
[ error ] ERROR in function compile_kernel [ common.sh:249 ]
[ error ] Kernel was not built [ @host ]
```

output/debug/output.log
```
Displaying message: Command line: setting BOARD to orangepione info
Displaying message: Command line: setting BRANCH to dev info
Displaying message: Preparing host info
Displaying message: Build host OS release serena info
Displaying message: Host system support was not tested serena wrn
Displaying message: Please don't ask for support if anything doesn't work
## BUILD SCRIPT ENVIRONMENT

Version: dff240e1dee0e40224a96be4f8be1a84f2a031e3

## BUILD CONFIGURATION

Build target:
Board: orangepione
Branch: dev

Kernel configuration:
Repository: https://github.com/megous/linux/
Branch: branch:orange-pi-4.10
Config file: linux-sun8i-dev

U-boot configuration:
Repository: git://git.denx.de/u-boot.git
Branch: tag:v2017.03
Offset: 1
Size: 0

CPU configuration:
480000 - 1200000 with ondemand
Displaying message: Syncing clock host info
Displaying message: Downloading sources  info
Displaying message: Checking git sources u-boot v2017.03 info
Displaying message: Checking out
Displaying message: Checking git sources linux-sun8i-mainline orange-pi-4.10 info
Displaying message: Checking out
Displaying message: Checking git sources sunxi-tools master info
Displaying message: Up to date
Displaying message: Cleaning output/debs for orangepione dev info
Displaying message: Cleaning u-boot/v2017.03 info
Displaying message: Compiling u-boot 2017.03 info
Displaying message: Compiler version arm-linux-gnueabihf-gcc 6.3.1 info
Displaying message: Checking out sources
Displaying message: Cleaning u-boot/v2017.03 info
Displaying message: Started patching process for u-boot sun8i-orangepione-dev info
Displaying message: Looking for user patches in userpatches/u-boot/u-boot-sunxi info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] 4kfix-limit-screen-to-full-hd.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] Merrii_Hummingbird_A20.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] a10-a20-lower-dram-clk.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-awsom-defconfig.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-bpi-m2plus.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-defconfig-options-orangepizero.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-lime2-emmc.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-nanopi-air.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-nanopi-m1.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-nanopi-m1plus.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-orangepi-plus2.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] addjust-nanopi-neo-cpufreq.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] disable-usb-keyboards.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] enable-DT-overlays-support.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] fix-cubieboard1-defconfig.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] h3-Fix-PLL1-setup-to-never-use-dividers.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] h3-add-disp2-video-driver-configs.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] h3-add-disp2-video-driver-jernej.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] h3-adjust-dram-frequency.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] h3-enable-power-led.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] h3-set-safe-axi_apb-clock-dividers.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] sun8i-set-machid.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] sunxi-boot-splash.patch succeeded info
Displaying message: Building deb linux-u-boot-dev-orangepione_5.27_armhf.deb info
Displaying message: Cleaning linux-sun8i-mainline/orange-pi-4.10 info
Displaying message: Patching kernel for compiler support
Reversed (or previously applied) patch detected!  Assuming -R.
Displaying message: Started patching process for kernel sun8i-dev info
Displaying message: Looking for user patches in userpatches/kernel/sun8i-dev info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-BergMicro-SPI-flashes.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-ad9834-dt-bindings.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-beelink-x2.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-configfs-overlay-for-v4.10.x.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-emac-pwr-en-orangepi-plus2e.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-h3-aliases.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-h3-overlays.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-h3-simplefb.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-opi-zero-dts-with-wireless.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-overlay-compilation-support.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-realtek-8189fs-driver.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-thermal-otg-wireless-opi-lite.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-uart-rts-cts-pins.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-wifi-pwrseq-opi-pc-plus.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] add-xradio-wireless-driver.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] enable-codec-orange-pi-2.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] enable-codec-orange-pi-pc.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] nanopi-neo-air-wireless-and-dvfs.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] packaging-4.x-DEV-with-postinstall-scripts.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] scripts-dtc-Update-to-version-with-overlays.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] spi-sun6i-allow-large-transfers.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] spidev-remove-warnings.patch succeeded info
Displaying message: Compiling dev kernel 4.10.3 info
Displaying message: Compiler version arm-linux-gnueabihf-gcc 6.3.1 info
Displaying message: Using kernel config file lib/config/kernel/linux-sun8i-dev.config info
Displaying message: ERROR in function compile_kernel common.sh:249 err
Displaying message: Kernel was not built @host err
Displaying message: Process terminated  info
```

P.S. Is there any plans to add support to the Armbian kernel to allow for custom HDMI timings without building custom kernels?  If so, when can it be expected?

zador-blood-stained, 2017-04-09:
Please add `PROGRESS_LOG_TO_FILE=yes` to the build options and attach the resulting compilation.log

> P.S. Is there any plans to add support to the Armbian kernel to allow for custom HDMI timings without building custom kernels? If so, when can it be expected?

Mainline (`dev` branch) uses video mode provided by EDID from the u-boot and it can be configured without recompilation (in the u-boot environment), but AFAIK the driver has some minor issues. Can't give any estimates, but new supported u-boot video driver most likely will be merged to u-boot 2017.07 ([release schedule](http://www.denx.de/wiki/view/U-Boot/ReleaseCycle)) and in-kernel DRM driver is in early WIP state.

cnlohr, 2017-04-09:
```
 PROGRESS_LOG_TO_FILE=yes  ./compile.sh BOARD=orangepione BRANCH=dev PROGRESS_LOG_TO_FILE=yes
```
I wasn't sure where to put the =yes... so I added it both... Doesn't seem to be adding anything to the debug/output log?

Compilation.log contains a fair but, but just compile lines.  No errors or anything.  It ends with:

```
...
  CC [M]  drivers/net/wireless/realtek/rtl8189fs/hal/phydm/rtl8188f/halhwimg8188f_rf.o
  CC [M]  drivers/net/wireless/realtek/rtl8189fs/hal/phydm/rtl8188f/halhwimg8188f_bb.o
  CC [M]  drivers/net/wireless/realtek/rtl8189fs/hal/phydm/rtl8188f/halhwimg8188f_fw.o
  CC [M]  drivers/net/wireless/realtek/rtl8189fs/hal/phydm/rtl8188f/phydm_regconfig8188f.o
  CC [M]  drivers/net/wireless/realtek/rtl8189fs/hal/phydm/rtl8188f/halphyrf_8188f.o
  CC [M]  drivers/net/wireless/realtek/rtl8189fs/hal/phydm/rtl8188f/phydm_rtl8188f.o
  CC [M]  drivers/net/wireless/realtek/rtl8189fs/platform/platform_ops.o
  CC [M]  drivers/net/wireless/realtek/rtl8189fs/core/rtw_mp.o
  CC [M]  drivers/net/wireless/realtek/rtl8189fs/core/rtw_mp_ioctl.o
  LD [M]  drivers/net/wireless/realtek/rtl8189fs/8189fs.o
  LD      drivers/net/wireless/built-in.o
  LD      drivers/net/built-in.o
  LD      drivers/built-in.o
```

@zador-blood-stained are there any hints as to how to use these features in the dev branch (assuming I can get the kernel to compile)?  Once I get the vive working on the OrangePi, I think the rest should be pretty straightforward.  It seems a number of people are already trying to get OGLES working, and all I'm after is a bunch of GL_LINEs

zador-blood-stained, 2017-04-09:
> Compilation.log contains a fair but, but just compile lines. No errors or anything.

Please attach the whole file as I asked. Compilation is parallelized into multiple processes and if one process ends with an error others are not killed, so the error message will be much earlier in the log.

> It seems a number of people are already trying to get OGLES working, and all I'm after is a bunch of GL_LINEs

So are you looking at Mali OpenGL ES acceleration? You may be able to get framebuffer Mali working by combining [this](https://github.com/mripard/sunxi-mali) with some Device Tree patches and kernel configuration changes, but to get X11 Mali you'll need a DRM video driver - the existing one is very limited and has some problems so it is disabled in the kernel config and new one doesn't exist yet.

If you are about video mode changes - it should just work, no changes are needed.

cnlohr, 2017-04-10:
Will post full log later this evening.  Re: OGLES: I have no interest in X.  I just want a full-screen context.  Is X required for OGLES acceleration?  Or am I going to need to bring in Xext and Xinerama?

zador-blood-stained, 2017-04-10:
> Re: OGLES: I have no interest in X. I just want a full-screen context. Is X required for OGLES acceleration?

No if I understand it correctly. There are examples in Mali SDK that run on plain framebuffer, but you'll need to make sure that you can find a matching framebuffer userspace binary driver (libMali.so for the specific hardware and kernel driver version). So far we only have [this](https://github.com/NextThingCo/chip-mali-userspace) which is a X11 DMA-buf r6p0 variant: `VARIANT=mali400-gles-linux-x11-dma_buf-no_Werror REVISION=Linux-r6p0-01rel0`

cnlohr, 2017-04-10:
May I make this a new issue as to not hijack this thread, or do I need to switch over to the forum?

zador-blood-stained, 2017-04-10:
> May I make this a new issue as to not hijack this thread, or do I need to switch over to the forum?

I don't think it's really an issue, so better move this to the forum. Though I'm not sure how much more help/info can be provided for this.

cnlohr, 2017-04-11:
Attached is the full compilation log.
[compilation.txt](https://github.com/igorpecovnik/lib/files/911965/compilation.txt)

cnlohr, 2017-04-11:
Ugh... I was missing u-boot-tools at the least.  Thanks for suggesting I fish through that file.

cnlohr, 2017-04-11:
Even after adding that still no go.
[compilation.txt](https://github.com/igorpecovnik/lib/files/911970/compilation.txt)

cnlohr, 2017-04-11:
Apparently you just have to try again... again... It got past that step if there are any further issues, will open a ticket. 

P.S. Here's the link for my forum discussion start. http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=2654

closed 2017-04-11 by cnlohr

igorpecovnik, 2017-04-11:
https://forum.armbian.com

zador-blood-stained, 2017-04-11:
I don't see any new posts on the Armbian forum (https://forum.armbian.com) and obviously we can't do things on the Orange Pi forum (http://www.orangepi.org/orangepibbsen/forum.php)

cnlohr, 2017-04-11:
WHOOPS!  I misread that!  Thanks, will post there soon!

cnlohr, 2017-04-11:
Thanks, everyone.  It is posted. https://forum.armbian.com/index.php?/topic/4009-next-thread-custom-hdmi-timingsresolution-hints-in-dev/