armbian/build #654: Can't find compiler for building legacy/default

cnlohr, 2017-04-13:
I understand this is probably a xineal issue, I had some difficulty coming up with the name for this one, but, performing all the same operations I did for the dev branch, with the following line:

```
./compile.sh BOARD=orangepione BRANCH=default PROGRESS_LOG_TO_FILE=yes 
```

Snip from compilation.log:
```
  AR      spl/arch/arm/lib/lib.a
  LD      spl/drivers/mmc/built-in.o
  LD      spl/arch/arm/lib/built-in.o
  CC      spl/drivers/serial/serial_ns16550.o
  CC      spl/drivers/serial/ns16550.o
  LD      spl/drivers/serial/built-in.o
  LD      spl/drivers/built-in.o
  LD      spl/u-boot-spl
  OBJCOPY spl/u-boot-spl-nodtb.bin
  COPY    spl/u-boot-spl.bin
  MKSUNXI spl/sunxi-spl.bin
  BINMAN  u-boot-sunxi-with-spl.bin
  CFGCHK  u-boot.cfg
dpkg-deb: building package 'linux-u-boot-orangepione-default' in 'linux-u-boot-orangepione_5.27_armhf.deb'.
ccache: error: Could not find compiler "arm-linux-gnueabihf-gcc" in PATH
  CHK     include/linux/version.h
  CHK     include/generated/utsrelease.h
  HOSTCC  scripts/genksyms/genksyms.o
  HOSTCC  scripts/genksyms/lex.lex.o
  HOSTCC  scripts/genksyms/parse.tab.o
  HOSTLD  scripts/genksyms/genksyms
  HOSTCC  scripts/kallsyms
  CC      scripts/mod/empty.o
ccache: error: Could not find compiler "arm-linux-gnueabihf-gcc" in PATH
scripts/Makefile.build:307: recipe for target 'scripts/mod/empty.o' failed
make[2]: *** [scripts/mod/empty.o] Error 1
scripts/Makefile.build:443: recipe for target 'scripts/mod' failed
make[1]: *** [scripts/mod] Error 2
make[1]: *** Waiting for unfinished jobs....
  Generating include/generated/mach-types.h
  CC      kernel/bounds.s
ccache: error: Could not find compiler "arm-linux-gnueabihf-gcc" in PATH
/home/cnlohr/git/armbian/sources/linux-sun8i/sun8i/./Kbuild:35: recipe for target 'kernel/bounds.s' failed
make[1]: *** [kernel/bounds.s] Error 1
Makefile:988: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2
make: *** Waiting for unfinished jobs....
Makefile:507: recipe for target 'scripts' failed
make: *** [scripts] Error 2
```

[compilation.log.zip](https://github.com/igorpecovnik/lib/files/918355/compilation.log.zip)
[output.log.zip](https://github.com/igorpecovnik/lib/files/918356/output.log.zip)

Any ideas?

cnlohr, 2017-04-13:
Ha! I just noticed your "Please don't ask for support if anything doesn't work" note.

Well, it's okay if you close it.

cnlohr, 2017-04-13:
Ok, after trying this in xenial... I'm not sure how anyone survives with it.  It seems even resizing the terminal in xenial breaks the build process with tons of "E: write error - write (32: broken pipe)"  >_<;  Ubuntu, you're becoming a mess.

cnlohr, 2017-04-13:
It does totally work in xenial.  Not really sure why there is a difference.

zador-blood-stained, 2017-04-13:
```
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gcc-aarch64-linux-gnu : Depends: gcc-5-aarch64-linux-gnu (>= 5.3.1-3~)
 gcc-arm-linux-gnueabi : Depends: gcc-5-arm-linux-gnueabi (>= 5.3.1-3~)
 gcc-arm-linux-gnueabihf : Depends: gcc-5-arm-linux-gnueabihf (>= 5.3.1-3~)
```
^ In this case packages installation is not performed at all which explains missing packages in #652

And based on this
```
Ign:1 http://packages.linuxmint.com serena InRelease
Hit:2 http://repo.steampowered.com/steam precise InRelease
Hit:3 http://packages.linuxmint.com serena Release
Hit:5 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu xenial InRelease
Get:6 http://security.ubuntu.com/ubuntu xenial-security InRelease [102 kB]
Hit:7 http://archive.ubuntu.com/ubuntu xenial InRelease
Hit:8 http://archive.canonical.com/ubuntu xenial InRelease
Hit:9 http://ppa.launchpad.net/js-reynaud/kicad-4/ubuntu xenial InRelease
Get:10 http://archive.ubuntu.com/ubuntu xenial-updates InRelease [102 kB]
Hit:11 http://ppa.launchpad.net/kdenlive/kdenlive-stable/ubuntu xenial InRelease
Hit:12 http://ppa.launchpad.net/maarten-baert/simplescreenrecorder/ubuntu xenial InRelease
Hit:13 http://ppa.launchpad.net/obsproject/obs-studio/ubuntu xenial InRelease
Hit:14 http://ppa.launchpad.net/ubuntu-wine/ppa/ubuntu xenial InRelease
Get:15 http://archive.ubuntu.com/ubuntu xenial-backports InRelease [102 kB]
```
there is a difference between default repositories and repository components that are enabled in Mint and Ubuntu.

cnlohr, 2017-04-13:
fair enough.

closed 2017-04-13 by cnlohr