armbian/build #660: lib/configuration.sh | BOOTSIZE= has no effect

Fourdee, 2017-04-26:
Always generates 1x EXT4 partitioned image.

```
sed -i "/REVISION=/a BOOTSIZE=64" lib/configuration.sh
#sed -i "/REVISION=/a BOOTFS_TYPE=fat" lib/configuration.sh #tried both with and without this option
./compile.sh
```

```
[ o.k. ] Calling image customization script [ customize-image.sh ]
[ o.k. ] Preparing image file for rootfs [ orangepizeroplus2-h5 jessie ]
[ o.k. ] Current rootfs size [ 891 MiB ]
[ o.k. ] Creating blank image for rootfs [ 1378 MiB ]
1.35GiB [96.8MiB/s] [====================================================>] 100%
[ o.k. ] Creating partitions [ root: ext4 ]
[ .... ] Creating rootfs [ ext4 ]
[ o.k. ] Copying files to image [ sdcard-dev-orangepizeroplus2-h5-jessie-no.raw ]
        804.90M  99%   40.23MB/s    0:00:19 (xfr#43164, to-chk=0/53993)

sent 808.09M bytes  received 867.29K bytes  41.49M bytes/sec
total size is 809.47M  speedup is 1.00
[ o.k. ] Copying files to /boot partition [ sdcard-dev-orangepizeroplus2-h5-jessie-no.raw ]
         22.07M  99%  223.54MB/s    0:00:00 (xfr#30, to-chk=0/37)

sent 22.07M bytes  received 612 bytes  44.15M bytes/sec
total size is 22.07M  speedup is 1.00
[ o.k. ] Free space: [ SD card ]
tmpfs                 1.5G  891M  610M  60% /root/armbian/output/cache/sdcard-dev-orangepizeroplus2-h5-jessie-no
/dev/loop0p1          1.3G  919M  361M  72% /root/armbian/output/cache/mount-dev-orangepizeroplus2-h5-jessie-no
[ o.k. ] Writing U-boot bootloader [ /dev/loop0 ]
[ o.k. ] Done building [ /root/armbian/output/images/Armbian_5.27_Orangepizeroplus2-h5_Debian_jessie_dev_4.10.0.img ]
[ o.k. ] Runtime [ 27 min ]
```

ThomasKaiser, 2017-04-26:
Just half a year ago `WARNING: This option is deprecated` appeared to inform/prepare 3rd parties that Armbian moved away from FAT partitions on **all** supported boards for a couple of reasons that were already discussed in detail: https://github.com/armbian/build/commit/e3c0988d3e793bdf0218cd24394302a84eb28ee6

And 3 days ago this happened: https://github.com/armbian/build/commit/e2035642d214660798b779e7153de517d4f72641

zador-blood-stained, 2017-04-26:
@ThomasKaiser 
Using FAT /boot partition is still supposed to work (for most boards), just wasn't tested yet after the latest changes.

Setting boot partition type manually should be fixed in 4952d45 
Now `BOOTFS_TYPE=fat` should be used to achieve previous behavior and `BOOTSIZE` can be used to override the default boot partition size of 64MB.

Fourdee, 2017-04-26:
@zador-blood-stained 

> Setting boot partition type manually should be fixed

Works a treat, thanks for quick fix 👍

closed 2017-05-11 by zador-blood-stained