armbian/build #662: awk error on local/getty logon

muwlgr, 2017-04-29:
I use PL2303 adapter connected to ttyUSB0 on my PC and to ttyS0 on my Cubietruck
I enabled getty on ttyS0 and I run 'screen ttyUSB0 115200' on the PC
agetty is started by systemd as '/sbin/agetty --keep-baud 115200 38400 9600 ttyS0 vt102'
When I log in locally to ttyS0, I see 'Last login' message, then kernel version (as in 'uname -a'), then this error : 
awk: fatal: error reading input file `-': Invalid argument
then it proceeds by showing colorful Cubietruck/Armbian logo and motd with various system parameters, and gets to the shell prompt.

This awk error is not seen when I log in by SSH, but then, I see 'Last login:' message after motd, not before. I suppose that sshd processes logon scripts differently from login but I am not sure where to look for this difference. Can you please try to reproduce this in your setup ?

muwlgr, 2017-04-29:
In my /etc/pam.d/login I see lines
session    optional   pam_lastlog.so
session    optional   pam_exec.so type=open_session stdout /bin/uname -snrvm
session    optional   pam_motd.so
(@include common-session is somewhere below)

and in my /etc/pam.d/sshd : 
@include common-session
session    optional     pam_motd.so  motd=/run/motd.dynamic
session    optional     pam_motd.so noupdate
session    optional     pam_mail.so standard noenv # [1]
session    required     pam_limits.so

can't determine where is sshd configured to print 'Last login:'

zador-blood-stained, 2017-04-29:
Most likely it is related either to [MOTD scripts](https://github.com/armbian/build/tree/master/scripts/update-motd.d) or the [profile script](https://github.com/armbian/build/blob/master/scripts/check_first_login.sh).

But both were heavily reworked since the last update, so next Armbian update may solve this issue already.

zador-blood-stained, 2017-10-08:
Don't think this was ever reproduced, so I'm not sure what can be done without a way to reproduce this error message. In any case it's not a critical issue.

closed 2017-10-08 by zador-blood-stained