armbian/build #663: Orange Pi Zero Unable to get interfaces

kutysam, 2017-05-04:
Hi there,

I tried the latest nightly Armbian_5.27.170505_Orangepizero_Ubuntu_xenial_dev_4.11.0
and tried Armbian_5.27.170501_Orangepizero_Ubuntu_xenial_dev_4.10.11

It seems 0501 is able to get interfaces (ifconfig) while 0505 only shows loopback interface when ifconfig is done.

I somehow believe the issue lies with https://github.com/armbian/build/commit/744192b750015a5e045afe136b0432f019c26d68 or some other commit on 2nd May.

Do let me know on how I can help. Thanks!

ThomasKaiser, 2017-05-04:
https://github.com/armbian/build/commit/ee5b5b9a28acf4310f5ff3e88f2c106f787ced94

zador-blood-stained, 2017-05-04:
XRadio driver fell out of 4.11 too due to some compilation issues.
I have no intention of rebasing/fixing it, no in-tree driver currently may be a better option considering how much "valueable feedback" we get on issues with this chip.

ThomasKaiser, 2017-05-04:
For anyone interested in XR819 and mainline Armbian: surprisingly @fifteenhex is still not that annoyed by user feedback to totally stop working on the driver: https://github.com/fifteenhex/xradio/commits/master

So it's up to interested users to sumbit a PR/patch again but unless Allwinner releases some documentation I would prefer to simply forget about Wi-Fi on this device.

martinayotte, 2017-05-04:
I think I have a fix for this XRadio compile issue.
I will test the run-time in few minutes...

zador-blood-stained, 2017-05-04:
I think the compile issue should be fixed already in the master branch of the driver repo. It's just a question of whether no driver at all (out of the box) is better than partially broken one. If people assume that we are responsible for the driver quality if we "added" it to the kernel, then we should not provide this driver by default.

martinayotte, 2017-05-04:
Compile issue doesn't seem fixed yet ...
Ok ! I won't push my patch and keep it for myself :smile: 
Anyway, it seems that those error floodings are back...

kutysam, 2017-05-05:
However what puzzles me is why is eth0 also not working?

martinayotte, 2017-05-05:
The fix for eth0 has been mentioned above : https://github.com/armbian/build/commit/ee5b5b9a28acf4310f5ff3e88f2c106f787ced94
Did you use the most recent build ?

kutysam, 2017-05-05:
I'll wait for the new build to be built. After which I'll update. Thanks @martinayotte

kutysam, 2017-05-05:
It works! Thank you for your help.
About the wifi issue, personally, yes the speed is slow but it works fine in my current setup of using motioneye.
basically it will upload a video to my webserver after recording a video. and there has been no issues so far. 
I can't understand why do people expect a 1000% from a $8 equipment.

Hope you can add it back in. Just for normal usage. Thanks!

closed 2017-05-05 by kutysam